ev3dev/ev3dev #1273: Cannot install pip3/pip

leenahere, 2019-04-30:
I'm trying to install pip3 on the latest ev3dev. I've tried this multiple times. I more or less always made the following steps:

- Latest version of ev3 from [download page](https://www.ev3dev.org/downloads/)
- Flashed SD card (8GB, even tried two different cards) via macOS command line tools
- Booted ev3, connected to WIFI for SSH connection
- Entered `sudo apt update`
- Entered `sudo apt install python3-pip`

It will start installing pip3. However, at the very end it will just stop at:
"Processing triggers for libc-bin (2.24-11+deb9u4) ..."
Once or twice the ev3 even crashed at this point. But I can't find anything in the logs that would suggest an installation error.

Afterwards pip3 is installed as I can call `which pip3` (results in `/usr/bin/pip3`). Now, when trying any pip3 command (I've tried multiple: `pip3 --version`, `pip3 --help`, `pip3 install some_package`) nothing happens (I've waited up to 20 minutes..). When I press ctrl-C, I get the following messages:

```
pip3 --help
^CTraceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_vendor/__init__.py", line 33, in vendored
    __import__(vendored_name, globals(), locals(), level=0)
ImportError: No module named 'pip._vendor.pkg_resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    from pip import main
  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 21, in <module>
    from pip._vendor.requests.packages.urllib3.exceptions import DependencyWarning
  File "/usr/lib/python3/dist-packages/pip/_vendor/__init__.py", line 76, in <module>
    vendored("pkg_resources")
  File "/usr/lib/python3/dist-packages/pip/_vendor/__init__.py", line 36, in vendored
    __import__(modulename, globals(), locals(), level=0)
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 74, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/_vendor/packaging/requirements.py", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/extern/__init__.py", line 43, in load_module
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "<frozen importlib._bootstrap_external>", line 1323, in _fix_up_module
KeyboardInterrupt
```

I've tried a lot of things that were suggested on stackoverflow and Co with [related/similar issues](https://stackoverflow.com/questions/49478573/pip3-install-not-working-no-module-named-pip-vendor-pkg-resources), however, nothing has worked so far. I also tried to call pip with the specific python version (`python3 -m pip --version`), but this raises the same issues.

Somehow pip3 seems to not be installed properly. I also tried to reinstall it. I can reproduce this over and over again when re-flashing the sd card with the downloaded image.

List of installed packages in `/usr/lib/python3/dist-packages`:

```
Crypto                                                dbus                                      keyring-10.1.egg-info          ptvsd-4.2.0.egg-info             rpyc
PIL                                                   debconf.py                                keyrings                       pyasn1                           rpyc-3.3.0.egg-info
Pillow-4.0.0.egg-info                                 easy_install.py                           keyrings.alt-1.3-nspkg.pth     pyasn1-0.1.9.egg-info            secretstorage
PyBluez-0.22.egg-info                                 ev3dev                                    keyrings.alt-1.3.egg-info      pycrypto-2.6.1.egg-info          serial
SecretStorage-2.3.1.egg-info                          ev3dev2                                   openrobertalab-1.7.1.egg-info  pygobject-3.22.0.egg-info        setuptools
__pycache__                                           evdev                                     pip                            pygtkcompat                      setuptools-33.1.1.egg-info
_cffi_backend.cpython-35m-arm-linux-gnueabi.so        evdev-0.4.7.egg-info                      pip-9.0.1.egg-info             pyserial-3.2.1.egg-info          six-1.10.0.egg-info
_dbus_bindings.cpython-35m-arm-linux-gnueabi.so       gattlib-0.20150805.egg-info               pixy-0.3.egg-info              python_ev3dev-1.2.0.egg-info     six.py
_dbus_glib_bindings.cpython-35m-arm-linux-gnueabi.so  gattlib.cpython-35m-arm-linux-gnueabi.so  pixy.py                        python_ev3dev2-2.0.0b3.egg-info  smbus.cpython-35m-arm-linux-gnueabi.so
_pixy.cpython-35m-arm-linux-gnueabi.so                gi                                        pkg_resources                  pyudev                           wheel
bluetooth                                             idna                                      plumbum                        pyudev-0.21.0.egg-info           wheel-0.29.0.egg-info
cryptography                                          idna-2.2.egg-info                         plumbum-1.6.2.egg-info         pyxdg-0.25.egg-info              xdg
cryptography-1.7.1.egg-info                           keyring                                   ptvsd                          roberta
```

**Desktop:**
 - OS and version: macOS 10.14
 - Connection Type: Wi-Fi + SSH

**Robot:**
 - Device: EV3
 - Versions: Linux ev3dev 4.14.96-ev3dev-2.3.2-ev3 (SD Card Image)

Online nothing suggests that installing pip on the EV3 should be a problem. Also I'm very sure that last week I had already installed pip and was able to use it. However, installing something via pip raised SSL certificate errors (that were probably due to the universities network). Unfortunately, I didn't save the image with pip successfully installed.

Any help would be greatly appreciated.

leenahere, 2019-04-30:
**Update**

Somehow after reinstalling pip multiple times it is indeed working. `pip3 --version` gives me the following response:

`pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)`

`pip3 --help` is working as well.

However, when trying to install numpy as a test, nothing really happens. My guess is that it's just painfully slow. On top of that the SSH connection is lost in that process.

I guess my question now is: Is this normal? Should pip be that slow?

I want to work with scikit-learn which is why I wanted to install pip in the first place. Is there another way to achieve this?

Thanks in advance.

dlech, 2019-04-30:
We highly recommend installing the debian packaged versions of python packages rather than using pip.

Many packages (like numpy) require binary components that have to be compiled from source code. As you have seen, the EV3 is really slow, so compiling things on the EV3 takes a long time, not to mention you have to install all of the build dependencies first.

So `sudo apt install python3-numpy python3-scipy` will get numpy and scipy installed much easier than using pip.

leenahere, 2019-05-13:
Thank you for the swift reply. Installing sklearn via apt worked fine.

closed 2019-05-13 by leenahere