ev3dev/ev3dev #1274: set_device attribute not working

alexdewar, 2019-05-02:
I am trying to get my large Lego Mindstorms motors (``lego-ev3-l-motor``) to work with my BrickPi3 but I'm having some issues. I understand that by default ev3dev sets the motor ports to be NXT drivers, but I seem unable to change the motor type using the ``set_device`` file in the ``lego-ports/*`` folders:
```sh
echo lego-ev3-l-motor > /sys/class/lego-port/port4/set_device
```
and I get
```
-bash: echo: write error: Operation not supported
```

dlech, 2019-05-02:
I think that `set_device` only works for sensors and not motors on BrickPi3. It seems like there was another issue about that recently so if we can find it, it should tell us for sure.

alexdewar, 2019-05-03:
Thanks for getting back to me. I've just got a couple of questions:
1. Is there any other way of setting the motor type for the motor output ports (e.g. with a kernel parameter), or would it involve hacking the kernel drivers?
2. Is there any issue with using the NXT driver for the "large" motors? My motors still spin around, but I'm not sure if e.g. the parameters in the sysfs files will be wrong etc.

dlech, 2019-05-03:
1. Probably not, but I'd have to look at the code to be sure.
2. The NXT large motors are nearly identical to the EV3 large motors so you shouldn't notice a difference (the max speed might be slightly different).