ev3dev/ev3dev #1275: EV3 Brick Manager fails to start with No supported video driver plugins found

sanyi, 2019-05-10:
After installing `snapshot-ev3dev-stretch-ev3-generic-2019-05-09.img`, the EV3 boots OK, the green LED lits up but the **brickman** fails to start.

I went back to an earlier version from `2018-11-13` and everything was OK **brickman** worked.

I set up a network connection and went to upgrade by `apt update`, `apt upgrade`. During the upgrade the display of the EV3 changed to showing logs similar to boot time.

After the reboot **brickman** failed to start again, similarly to `2019-05-09` version.

Since I am still able to connect via SSH, here are some info:

```
 brickman.service - Brick Manager
   Loaded: loaded (/lib/systemd/system/brickman.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2019-05-10 19:06:33 UTC; 32min ago
  Process: 1770 ExecStart=/usr/sbin/brickman (code=exited, status=0/SUCCESS)
 Main PID: 1770 (code=exited, status=0/SUCCESS)
```
attempting to manually start the service by: `sudo systemctl start brickman.service`, `sudo journalctl -u brickman.service` returns:
```
May 10 19:40:41 ev3dev systemd[1]: Started Brick Manager.
May 10 19:40:42 ev3dev brickman[2052]: main.vala:127: No supported video driver plugins found
```
**Desktop:**
 - OS: XUbuntu 18.04
 - Connection Type: USB

**Robot:**
 - Device: EV3
 - Versions: 2019-05-09 and 2018-11-13 (after upgrade)

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-11-13
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.1
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165361A3E0
BOARD_INFO_TYPE=main
```

dlech, 2019-05-10:
Thanks for reporting the issue. As a workaround you can install the `libgrx-3.0-2-plugin-linuxfb` package.

    sudo apt update && sudo apt install libgrx-3.0-2-plugin-linuxfb

sanyi, 2019-05-10:
Thank you. I confirm that `libgrx-3.0-2-plugin-linuxfb` solves the problem.

dlech, 2019-05-10:
To fix this properly, we should add that package as a dependency of the ev3devkit package

ndward, 2019-05-11:
I confirm that the May 9 image won't boot to Brickman on EV3. I suggest that that image should be removed from https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/ before other people waste their time with it.

terbos90803, 2019-05-26:
I hit this same problem by doing an apt update; apt upgrade.  It broke the 2019-03-03 image in the same way described by the OP.
I also confirm that the suggested workaround fixes the problem.

timriker, 2019-05-26:
Hit the same issue. Is the package dependency going to get fixed?

dlech, 2019-05-26:
Should be fixed now in `2019-05-26` snapshot.

closed 2019-05-26 by dlech

ndward, 2019-05-29:
The image files in the 2019-05-26 snapshot are dated 05-09 in the filenames, which is very confusing.

dlech, 2019-05-30:
Thanks for noticing. It looks like an older image was rebuilt. There is a new 2019-05-29 build.

ndward, 2019-05-30:
I downloaded and installed **snapshot-ev3dev-stretch-ev3-generic-2019-05-29.img.xz** and when I tried to boot up the EV3 it got stuck while displaying 'MINDSTORMS starting...', with red LEDs. That means this is the third consecutive release that does not boot to Brickman for me.