ev3dev/ev3dev #1276: Add support for encoding and decoding RSF Files

JakubVanek, 2019-05-12:
**Is your feature request related to a problem? Please describe.**
I want to be able to encode sound files that will play back on the stock firmware.

**Describe the solution you'd like**
Ideally it would be a utility that would take a sound format and transcode it to RSF.

**Describe alternatives you've considered**
I'd like to write:
- a SoX format handler (or something like that) which would read and write RSF files directly, or
- a (Go?) program that would transcode the source file via sox or libsox to U8 or IMA ADPCM raw output and then pack it.

**Additional context**
I'd want to add more support for the lms2012 formats ([lmsformats](https://github.com/jakubvanek/lmsformats)).

dlech, 2019-05-13:
I think it would be really cool to have support for .rsf in SoX. I though about doing it years ago, but it was too easy to do it with a [shell script](https://github.com/ev3dev/lms2012-compat/blob/ev3dev-stretch/lmssrc/raw2rsf.sh). 😄

JakubVanek, 2019-05-15:
I have finished the first implementation: https://github.com/JakubVanek/sox-rsf
It also found somewhere that actually RSF is the same format as RSO, so ffmpeg can be used too:
```sh
ffmpeg -i [input.sth] -ac 1 -ar 8000 -f rso [output.rsf]
```

JakubVanek, 2019-05-15:
However I've discovered something strange: I tried playing back a IMA ADPCM RSF on the brick and the result sounded quite bad. I have no idea why is it so, but U8 works, so that should suffice.

The support for ADPCM is implemented, but it is disabled at the end of `rso.c`.

dlech, 2019-05-17:
Only the U8 format is implemented in the EV3 firmware, so it is not surprising to me that ADPCM sounds terrible - it is probably being treated as U8.