ev3dev/ev3dev #1277: cannot work with Power Functions Servo Motor(88004)

chitacan, 2019-05-17:
**What are you trying to do?**

Hi, I am trying to work with Power Functions Servo Motor([88004](https://shop.lego.com/en-US/product/Power-Functions-Servo-Motor-88004)). But It's not working at all :(
I connected 88004 with my ev3 via [8528](https://www.bricklink.com/v2/catalog/catalogitem.page?S=8528-1#T=S&O={%22iconly%22:0}) and [8886](https://www.bricklink.com/v2/catalog/catalogitem.page?S=8886-1&name=Power%20Functions%20Extension%20Wire%20(20cm)&category=%5BPower%20Functions%5D#T=S&O={%22iconly%22:0}) cables. With same cables [88003](https://shop.lego.com/en-US/product/Power-Functions-L-Motor-88003) works well. What am I missing??

**Source Code**

```bash
$ cd /sys/class/lego-port/port4
$ echo dc-motor > mode
$ cd /sys/class/dc-motor/motor0
$ echo 50 > duty_cycle_sp
$ echo 1000 > time_sp
$ echo run-timed > command
```

**Desktop (please complete the following information):**
 - OS and version: macOS 10.14
 - Connection Type: Wi-Fi
 - Software: Terminal.app

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533F94DB
BOARD_INFO_TYPE=main
```

dlech, 2019-05-17:
> What am I missing??

Wires. The 8528 cable only has 2 wires and the 88004 servo motor requires 4 wires. Other Power Functions motors only used 2 wires.

So you will need to create your own cable in order to make it work. Also, I _think_ that in that case, the motor will draw power from the 5V supply, which is shared with USB, input ports and output ports and limited to 1A.

Another option is to use the Power Functions IR Receiver ([8884](https://www.lego.com/en-us/themes/power-functions/products/ir-receiver-8884)) along with a [HiTechnic IR transmitter](https://modernroboticsinc.com/product/hitechnic-nxt-irlink-sensor/).

chitacan, 2019-05-18:
@dlech Oh... Thanx for the detailed reply. I found this [video](https://www.youtube.com/watch?v=K-iEnWZ5Kew) and it worked for me.

closed 2019-05-18 by chitacan