ev3dev/ev3dev #1279: Cannot connect second EV3 via Bluetooth

terbos90803, 2019-05-27:
**What are you trying to do?**

I am trying to connect multiple EV3 to the same laptop via Bluetooth through the Brickman menu

**What did you expect to happen**

The program should have let me do this without any trouble

**What actually happened**

The program did let me connect one EV3.  When I try to connect the second EV3, it tries to connect after pairing, but always fails with GDBus.Error:net.connman.Error.Failed Input/output error

**Source Code**

Haven't made it that far yet.

**Screenshots**
none

**Desktop (please complete the following information):**
 - OS and version: Windows 10 1809
 - Connection Type: Bluetooth
 - Software: Just trying to get connected

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]
robot@ev3dev-BLUE:~$ ev3dev-sysinfo -m

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.1
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653491EDE
BOARD_INFO_TYPE=main
```

**Background Information**

I am trying to make a central Windows laptop control up to 4 EV3 robots at the same time.  All of the robots must connect via Bluetooth because they must be used away from WiFi infrastructure and there is no room in the model for a USB WiFi dongle.

dlech, 2019-05-27:
Windows 10 Bluetooth PAN is basically only designed to connect Windows 10 to a mobile phone or a Windows 10 Phone to a computer. It doesn't work well at all with ev3dev.

I would suggest using a Bluetooth socket or the `rfcomm` utility to set up a serial port over Bluetooth.

terbos90803, 2019-05-27:
Thanks.  I'll look into those options.

But to clarify the obvious question then, why does ev3dev use Bluetooth PAN if it doesn't work well?  Would it work better if my host was running Linux (e.g. a Pi)?

dlech, 2019-05-27:
There is more of a discussion about Windows 10 Bluetooth at #1154. It used to work better on Windows 7. The back story is that we made all connections to the EV3 network connections so that we didn't have to build any special tools to connect to ev3dev and so that the EV3 could access the Internet without Wi-Fi.

Linux will probably work better, but it is probably not trivial to setup.

terbos90803, 2019-05-27:
Thanks for the info.  Hopefully I can get something working.

closed 2019-05-27 by terbos90803

terbos90803, 2019-05-30:
After much hacking, I did eventually get it working.
The winning config is:

- pybluez on ev3dev2
- pybluez-win10 on Windows 10 (requires the installation of Visual Studio and the Windows Platform SDK, even though you never actually use them directly).
- pybluez sample code found online

Not only can I run multiple robots at the same time, but there's no lag in the controls like with RPyC.  So all around a win.

Audrielcz, 2020-03-04:
@terbos90803 I am trying to connect 4 EV3 bricks together. Mainly master/slave system. Although I tried both RPyC and PyBluEz. None of them seem to work for me though. Could you tell me how did you manage it with PyBluEz? 

1) somehow install that library on the bricks.. (magic)
2) connect them together
3) make it work? 🗡 
Thank you

terbos90803, 2020-03-12:
My code repo with step by step instructions on installing the bits is at: https://github.com/terbos90803/ev3-soccerbots

See the doc "How to setup new laptops and robots for EV3 Soccerbots.docx"