ev3dev/ev3dev #1280: GDbus Error

THEuuuBrickman, 2019-05-27:
I am trying to connect the EV3 to my Laptop.

the Wifi-Output:GDBus.Error..net.connman.Error.Operation.Aborted: Operation aborted
the Bluetooth-Output: GDBus.Error:org.bluez.Error.Failed: Protocol not available
USB: Connecting the EV3 with normal firmware works (wich means the EV3 is shown in the file explorer), but if i boot the EV3 with the ev3dev-linux-Operating-System the EV3 is not shown in the file Explorer.

the laptop specs:
Window 10 1809, ThinkPad Lenovo Yoga 380

the EV3:
EV3, Linux-Kernel: 4.14.96.-ev3dev-2.3.2-ev3

THEuuuBrickman, 2019-05-27:
ps.: 
1. I am using the Edimax EW-7811Un WiFi Dongle
2. Bluetooth: sometimes it is connecting, but only for some seconds

dlech, 2019-05-27:
Windows 10 Bluetooth PAN is basically only designed to connect Windows 10 to a mobile phone or a Windows 10 Phone to a computer. It doesn't work well at all with ev3dev.

Since you have a Wi-Fi dongle, just use that instead.

THEuuuBrickman, 2019-05-27:
Ok, but what do I have to do to get the Wifi working?

As i said earlier: the Wifi isn't working.
the Wifi-Output:GDBus.Error..net.connman.Error.Operation.Aborted: Operation aborted

dlech, 2019-05-27:
For Wi-Fi, I usually connect with USB first, then setup Wi-Fi using `connmanctl` instead of trying to do it on the brick because I have a really long passphrase.

https://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/

Asdew95, 2019-06-05:
I have the same issue here with the same ev3dev version. Tried rebooting multiple times.

memarsh, 2019-10-08:
I'm having the same bluetooth issue with the same ev3dev version as well, though I'm on a Mac (Mojave 10.14.6)

dlech, 2019-10-09:
DBus error can be caused by many things. Please give the full steps on both the computer and the EV3 to reproduce the problem.

kristoph, 2019-10-12:
@dlech when attempting to connect (as in pressing the 'Connect' button) through the Wireless and Networks > Bluetooth > Devices screen, to a pre-existing bluetooth device that action will ALWAYS fail with GDBus.Error:org.bluex.Error.Failed: Protocol not available. If I remove the pre-existing device and reconnect to that same device as listed in the scanned devices I am then able to see this device in the All Network Connections screen and I can click on that and connect through that screen.

My machine is Mac OS X 10.15 but I have observed this same behavior with other version of Mac OS X.

vjandrei, 2019-11-08:
I have the same problem as @kristoph I try to connect my ev3 to Alexa echo.

BhavinFaldu, 2019-12-30:
@kristoph, @dlech You have to only first enable the Bluetooth in EV3 then in VS run the command of mission_01.py python file (you can find code in https://www.hackster.io/alexagadgets/lego-mindstorms-voice-challenge-mission-1-f31925 ) then it'll automatically generate one Gadget280 pair id for your another device/alexa then only select & connect this Gadget280. And within 30 second it'll connect with your ev3dev.

benwiley4000, 2021-01-29:
@kristoph sorry, could you be more specific? I'm not sure what you mean "remove the pre-existing device and reconnect to that same device as listed in the scanned devices." Every time I try to remove and re-pair from the bluetooth screen my connection fails. If I go to the all network connections screen, the only menu option I see is "wired." BTW I had a similar issue with the EV3 default OS. MacOS Mojave.

benwiley4000, 2021-01-31:
I found a working solution! Start the timidity server:

```
timidity --output-mono --sampling-freq=25000 -iA
```

And send messages using the mido python library: https://mido.readthedocs.io/en/latest/

For example this worked for me:

```python
import mido
msg = mido.Message('note_on', note=50, velocity=80)
port = mido.open_output('TiMidity port 0')
port.send(msg)
```

benwiley4000, 2021-01-31:
Apologies for the earlier comment I deleted, I had posted in the wrong thread.

alanwilter, 2021-02-11:
I'm using, Lego EV3, updated stretch 9.13 (kernel 4.14.117-ev3dev-2.3.5-ev3), brickman 0.10.3, macOS 11.2.1. I follow all steps in 
https://www.ev3dev.org/docs/tutorials/using-bluetooth-tethering/ 
or
https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

But just after I do "pair" and accept in both EV3 and Mac, I see a "connected" and a bold "BT" icon on Brickman screen and then, few seconds later "disconnected" and if I try to connect again, I will get the brickman screen showing 
`Bluetooth-Output: GDBus.Error:org.bluez.Error.Failed: Protocol not available`

I had a similar GDBus error when I tried to set up wifi using only Brickman. In the end I used the usb cable to connect (no issues) and set up wifi, which is working fine. So, is there something astray with brickman interface?

Moreover, I tried to connect the EV3 via bluetooth with my ipad. At the moment it asked me to enable "AVRCP", it failed and now I try to access "Bluetooth" page in Brickman it shows "Not available". Yet the "BT" icon is ON and BOLD at the first line in the Brickman screen, which I understand means: connected. I know it's not supposed to work with iPads, but I hope it can helps to debug.

After rebooting, I could access Bluetooth page and then I removed my mac and iPad devices.

benwiley4000, 2021-02-11:
@alanwilter good summary of my experience as well.

dlech, 2021-02-11:
It is not technically possible to use ev3dev with iPad. It requires use of the proprietary MFi chip that we don't have access to. Don't use tethering unless you are daisy-chaining EV3s - it is for sharing an Internet connection from the EV3 to another device. It is normal to disconnect after pairing - the computer is just seeing what services are there and then disconnects. Try removing the EV3 from the list of devices on your computer and removing the computer from the list of devices on the EV3, disable tethering and start over with the "connecting to the Internet" instructions.

benwiley4000, 2021-02-11:
I was experiencing issues without tethering or using and iPad - standard ev3 connecting to computer stuff.

> It is normal to disconnect after pairing - the computer is just seeing what services are there and then disconnects. 

Sorry, I'm not sure I follow - how am I supposed to share internet if the device always disconnects?

dlech, 2021-02-11:
Pairing just pairs the devices, it doesn't initiate the network connection. Connecting is a separate step.

benwiley4000, 2021-02-25:
Finally I was about to connect using @kristoph's instructions. I attempted this and failed before, I guess the routine is somewhat confusing. Here's exactly what I did which finally worked:

1. removed the bluetooth device
2. stopped and restarted the scan under the bluetooth screen
3. paired by hitting accept on my ev3 and connect on my mac. this does briefly connect but then you're immediately disconnected.
4. DO NOT hit connect again on that screen. exit out and go to the all network connections screen, where the bluetooth device should now be listed.
5. Hit connect there... the first time I got a message about a timeout, but then when I tried again it worked. I set it to connect automatically, which seems to be working after a reboot.

alanwilter, 2021-02-27:
@benwiley4000 Absolutely brilliant! Worked exactly as you said. For me one can close this ticket.

benwiley4000, 2021-02-27:
Unfortunately I can't get the same to work when trying to pair with my Linux PC, but Bluetooth works for mac. Go figure

s0kr8s, 2022-07-16:
On Ubuntu 21, solved this problem by:

apt-get install blueman
open blueman
click View->Local Services
add a Network Access Point (NAP settings: dnsmasq, PAN support: NetworkManager, DUN support: Network Manager)
apply
on brick: wireless and networks -> bluetooth -> powered, visible, start scan, select device to connect to -> **network connection** (_not_ connect) -> connect.

For whatever reason, brick seems to demand bluetooth connection AND network access point, otherwise brick just disconnects after pairing. Once network access point is available, and link is initiated for network access, bluetooth connects and stays connected.

HannaFrangi, 2026-08-25:
A little late to this thread, but managed to get connected to the internet via Bluetooth using Windows 10.

What worked for me: pair the EV3 to your laptop over Bluetooth first, then on Windows go to **Settings > Network & Internet > Mobile hotspot**, and set "Share my internet connection over" to **Bluetooth** (not Wi-Fi) and turn it on . Once the EV3 shows up under "Devices connected", hit *Network Connection*, and then *Connect* on that screen. 

[image]

State should flip to **Online** on the EV3 and you're good to SSH in. 

Laptop: Dell Precision 7520 running windows 10 22H2