ev3dev/ev3dev #1281: date -R gives the wrong date

staadecker, 2019-05-30:
**Describe the bug**
Running `date -R` in SSH gives the wrong time and date (`Thu, 03 Nov 2016 18:16:06 +0000`).

```
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653557B01
BOARD_INFO_TYPE=main
```

staadecker, 2019-05-30:
Also some research online shows that `ntp` is usually the package responsible for time sync. I tried to uninstall it however now I can't install it. 

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package ntp is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ntp' has no installation candidate
```

dwalton76, 2019-05-30:
Can you run `timedatectl status` and post the output.  You should see something like
```
robot@ev3dev:~/ev3dev-lang-python$ 
robot@ev3dev:~/ev3dev-lang-python$ timedatectl status
      Local time: Thu 2019-05-30 00:29:44 UTC
  Universal time: Thu 2019-05-30 00:29:44 UTC
        RTC time: n/a
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
robot@ev3dev:~/ev3dev-lang-python$ 
robot@ev3dev:~/ev3dev-lang-python$ 
```

`date -R` for me
```
robot@ev3dev:~/ev3dev-lang-python$ date -R
Thu, 30 May 2019 00:30:34 +0000
robot@ev3dev:~/ev3dev-lang-python$ 
```

staadecker, 2019-05-30:
I was able to manually set the time using `date --set ...`. So my issue is fixed. Now `timedatectl status` works.

```
     Local time: Wed 2019-05-29 20:43:31 UTC
  Universal time: Wed 2019-05-29 20:43:31 UTC
        RTC time: n/a
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
```

However, "NTP synchronized" is no

dlech, 2019-05-30:
> however now I can't install it.

> E: Package 'ntp' has no installation candidate

   sudo apt update

staadecker, 2019-05-30:
That fixed it! Thank you!

closed 2019-05-30 by staadecker