ev3dev/ev3dev #1283: ev3dev GUI not loading/opening after 'apt-get upgrade'

runningOutOfUsernameIdeas, 2019-06-05:
**What are you trying to do?**

Run command 'sudo apt-get upgrade' - this command was run on a fresh copy of ev3dev stretch

Note: 'sudo apt-get update' works fine

**What did you expect to happen**

The command should have run successfully. (without errors)

**What actually happened**

While said command was running, the brick suddenly and unexpectedly turned from gui to console mode (I don't know if that's the correct way of saying that) - it also seemed to give some errors. Restarting didn't work (I restarted through the ssh server). The ssh server starts however, so I can interface with the brick. However the GUI never starts (a problem because I need bluetooth). The last line on the ev3 console states (everything else on the screen seems okay) when it gets stuck after restarting:

'[TIME] Timed out waiting..c.1.auto.device.'

I reimaged the sd card and ran the command again. The problem persists.

**Robot:**
 - Device: EV3
 - Versions: latest ev3dev stretch release

lf you want the output of 'sudo apt-get upgrade' on a fresh copy of ev3dev stretch outputs, I can give that to you.

dlech, 2019-06-05:
> lf you want the output of 'sudo apt-get upgrade' on a fresh copy of ev3dev stretch outputs, I can give that to you.

Yes please.

runningOutOfUsernameIdeas, 2019-06-08:
Here it is:

robot@ev3dev:~$ sudo apt-get update
Get:1 http://security.debian.org stretch/updates InRelease [94.3 kB]
Hit:3 http://archive.ev3dev.org/debian stretch InRelease
Ign:2 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:4 http://cdn-fastly.deb.debian.org/debian stretch Release
Get:5 http://security.debian.org stretch/updates/main armel Packages [474 kB]
Fetched 568 kB in 1d 20h 47min 31s (3 B/s)
Reading package lists... Done

robot@ev3dev:~$ sudo apt-get upgrade
[sudo] password for robot:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  ev3devkit-data libev3devkit-0.5-0 linux-image-ev3dev-ev3
The following packages will be upgraded:
  base-files bind9-host brickman ca-certificates-java exim4 exim4-base
  exim4-config exim4-daemon-light gir1.2-ev3devkit-0.5 java-common jri-11-ev3
  libbind9-140 libdns-export162 libdns162 libisc-export160 libisc160
  libisccc140 libisccfg140 liblwres141 libnss-myhostname libnss-resolve
  libpam-systemd libpng16-16 libruby2.3 libssh2-1 libsystemd0 libudev1
  openrobertalab ruby2.3 systemd systemd-sysv tzdata udev unzip wget
  wpasupplicant
36 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Need to get 30.3 MB of archives.
After this operation, 580 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:11 http://security.debian.org stretch/updates/main armel bind9-host armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [229 kB]
Get:12 http://archive.ev3dev.org/debian stretch/main armel brickman armel 0.10.1 [203 kB]
Get:13 http://archive.ev3dev.org/debian stretch/main armel gir1.2-ev3devkit-0.5 armel 0.5.3.1 [17.5 kB]
Get:14 http://archive.ev3dev.org/debian stretch/main armel jri-11-ev3 armel 11.0.4~4-1 [14.9 MB]
Get:15 http://security.debian.org stretch/updates/main armel libisc160 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [367 kB]
Get:1 http://cdn-fastly.deb.debian.org/debian stretch/main armel base-files armel 9.9+deb9u9 [67.4 kB]
Get:23 http://security.debian.org stretch/updates/main armel libdns162 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [954 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian stretch/main armel systemd-sysv armel 232-25+deb9u11 [82.4 kB]
Get:3 http://cdn-fastly.deb.debian.org/debian stretch/main armel libpam-systemd armel 232-25+deb9u11 [176 kB]
Get:4 http://cdn-fastly.deb.debian.org/debian stretch/main armel libnss-resolve armel 232-25+deb9u11 [175 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian stretch/main armel libsystemd0 armel 232-25+deb9u11 [260 kB]
Get:6 http://cdn-fastly.deb.debian.org/debian stretch/main armel libnss-myhostname armel 232-25+deb9u11 [104 kB]
Get:24 http://security.debian.org stretch/updates/main armel libisccc140 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [197 kB]
Get:7 http://cdn-fastly.deb.debian.org/debian stretch/main armel systemd armel 232-25+deb9u11 [2219 kB]
Get:25 http://security.debian.org stretch/updates/main armel libisccfg140 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [216 kB]
Get:26 http://security.debian.org stretch/updates/main armel libbind9-140 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [204 kB]
Get:27 http://security.debian.org stretch/updates/main armel liblwres141 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [210 kB]
Get:28 http://security.debian.org stretch/updates/main armel exim4-config all 4.89-2+deb9u4 [377 kB]
Get:29 http://security.debian.org stretch/updates/main armel exim4 all 4.89-2+deb9u4 [7862 B]
Get:30 http://security.debian.org stretch/updates/main armel exim4-base armel 4.89-2+deb9u4 [1091 kB]
Get:8 http://cdn-fastly.deb.debian.org/debian stretch/main armel udev armel 232-25+deb9u11 [1072 kB]
Get:31 http://security.debian.org stretch/updates/main armel exim4-daemon-light armel 4.89-2+deb9u4 [496 kB]
Get:9 http://cdn-fastly.deb.debian.org/debian stretch/main armel libudev1 armel 232-25+deb9u11 [121 kB]
Get:10 http://cdn-fastly.deb.debian.org/debian stretch/main armel tzdata all 2019a-0+deb9u1 [273 kB]
Get:32 http://security.debian.org stretch/updates/main armel libpng16-16 armel 1.6.28-1+deb9u1 [266 kB]
Get:16 http://cdn-fastly.deb.debian.org/debian stretch/main armel wget armel 1.18-5+deb9u3 [779 kB]
Get:33 http://security.debian.org stretch/updates/main armel libisc-export160 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [316 kB]
Get:34 http://security.debian.org stretch/updates/main armel libdns-export162 armel 1:9.10.3.dfsg.P4-12.3+deb9u5 [765 kB]
Get:17 http://cdn-fastly.deb.debian.org/debian stretch/main armel java-common all 0.58+deb9u1 [13.6 kB]
Get:18 http://cdn-fastly.deb.debian.org/debian stretch/main armel ca-certificates-java all 20170929~deb9u3 [15.1 kB]
Get:19 http://cdn-fastly.deb.debian.org/debian stretch/main armel libssh2-1 armel 1.7.0-1+deb9u1 [127 kB]
Get:20 http://cdn-fastly.deb.debian.org/debian stretch/main armel unzip armel 6.0-21+deb9u1 [153 kB]
Get:21 http://cdn-fastly.deb.debian.org/debian stretch/main armel libruby2.3 armel 2.3.3-1+deb9u6 [2873 kB]
Get:35 http://security.debian.org stretch/updates/main armel wpasupplicant armel 2:2.4-1+deb9u4 [816 kB]
Get:22 http://cdn-fastly.deb.debian.org/debian stretch/main armel ruby2.3 armel 2.3.3-1+deb9u6 [188 kB]
Get:36 http://archive.ev3dev.org/debian stretch/main armel openrobertalab all 1.7.4+1.0.0 [30.6 kB]
Fetched 30.3 MB in 40s (749 kB/s)
(Reading database ... 42306 files and directories currently installed.)
Preparing to unpack .../base-files_9.9+deb9u9_armel.deb ...
[... 160 lines cut ...]
[... cut ...]
Installing new da850-lego-ev3-color.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3.dtb
Taking backup of da850-lego-ev3.dtb.
Installing new da850-lego-ev3.dtb.
flash-kernel: installing version 4.14.96-ev3dev-2.3.2-ev3
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
Processing triggers for libc-bin (2.24-11+deb9u4) ...
robot@ev3dev:~$

It took like 1 hour to run completely.

Thanks

runningOutOfUsernameIdeas, 2019-07-02:
I've solved the problem - some important packages were 'held back'. I just had to bypass that and ev3dev rebooted fine.

closed 2019-07-02 by runningOutOfUsernameIdeas

racheliurui, 2020-04-19:
> I've solved the problem - some important packages were 'held back'. I just had to bypass that and ev3dev rebooted fine.

Hi, @jonajia could you share more details about how to bypass the problem? I am having the same issue with booting my EV3

Thanks