ev3dev/ev3dev #1284: Not able to set mode of a sensor connected to the mindsensors.com EV3SensorMUX unless program it's run with 'sudo'

KevinRSiu, 2019-06-08:
Hello, everyone! I'd really appreciate if someone could assist me with this little issue I ran into.

**What are you trying to do?**
I have a mindsensors.com EV3SensorMUX connected to a robot with ev3dev. I wrote a code in C++ to set up an Ultrasonic Sensor connected to this Multiplexer. First, I declare a `lego_port` variable and set up the `mode` and `set_device` to `uart`and `lego-ev3-us` respectively. Then, I declare a  `sensor` variable, and I'm able to see its address, modes, values, etc. when running the program.

Finally, I decided I wanted to try to switch the sensor mode from `US-DIST-CM` to `US-DIST-IN` and check the readings of the sensor now. 

**What did you expect to happen**
The program should've switched the sensor mode and print the new values.  

**What actually happened**
The program crashes when it tries to switch the mode with the following error:
```
terminate called after throwing an instance of 'std::system_error'
  what():  /sys/class/lego-sensor/sensor10/mode: No such device
Aborted
```
I'm able to set the mode manually from a console using the following command: 
```
 robot@ev3dev:~$ echo US-DIST-CM > /sys/class/lego-sensor/sensor10/mode
```
But for some reason, the program won't let me do this unless I do one of the following:
1. I run the program with `sudo ./test_multiplexer`
2. I change the ownership of the program so it uses `root` permissions using the following commands.
```
robot@ev3dev:~$ sudo chown root:root ./test_multiplexer   
robot@ev3dev:~$ sudo chmod a+s ./test_multiplexer
```
I'm wondering if there's a way to run the program without using `sudo` or changing the ownership of the program, as I understand that this could lead to potential security issues. 

**Source Code**
Here is the code of the program I am trying to run.
```
#include <ev3dev.h>
#include <iostream>
#include <string.h>
#include <thread>
#include <chrono>

using namespace std;
using namespace ev3dev;

lego_port mult1("i2c-legoev35:i2c80:mux1");
lego_port mult2("i2c-legoev35:i2c81:mux2");
lego_port mult3("i2c-legoev35:i2c82:mux3");

int main(){

    cout << "***Hola, ( EV3 ) ! ***" << endl;
    cout << "Vamos a Probar el Multiplexor" << endl;

	cout << "Inicializando Sensor en el Multiplexor" << endl;

	mult1.set_mode("uart");
	mult1.set_set_device("lego-ev3-us");
	this_thread::sleep_for(chrono::milliseconds(500));

	sensor sensorUltrasonido(mult1.address());

	if(sensorUltrasonido.connected()){
		cout << "Direccion: " << sensorUltrasonido.address() << endl;
		cout << "Modo Actual: " << sensorUltrasonido.mode() << endl;
		cout << "Modos Disponibles: ";
		set<string> moditos = sensorUltrasonido.modes();
		for(set<string>::iterator it=moditos.begin(); it != moditos.end(); ++it){
			cout <<"["<<*it<<"] ";
		}
		cout << endl;
		sensorUltrasonido.set_mode("US-DIST-IN");
		//sensorUltrasonido.set_mode("US-DIST-CM");
		cout << "Sensor Inicializado. Leyendo distancias por 10 segundos:" << endl;
		for(int i=0; i<20; i++){
			cout << (i+1)*0.5  << "s : " << sensorUltrasonido.float_value() << endl;
			this_thread::sleep_for(chrono::milliseconds(500));
		}
	}else{
		cout << "No conectado. Multiplexor debe estar en Puerto 3. Ultrasonido en Puerto C1" << endl;
	}
    cout << "*** ( EV3 ) Bye! ***" << endl;
    return 0;
}
```
**Desktop (please complete the following information):**
 - OS and version: Windows 10.0.17134.765
 - Connection Type:  USB
 - Software: Used Ubuntu 16.10 vía Windows Subsystem for Linux to do cross-compiling. PuTTY and WinSCP to transfer the program and execute it. 

**Robot (please complete the following information):**
 - Device: EV3 with a mindsensors.com EV3SensorMUX
 
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165347B61D
BOARD_INFO_TYPE=main
```

**Background Information**

I am trying to make an example C++ program that demonstrates how to use the EV3SensorMUX  and, eventually, will use it for other applications in the same robot. Thanks for any help in advance! :)

dlech, 2019-06-08:
> I'm able to set the mode manually from a console using the following command:

If this is the case, then the problem is with the C++ library or in your program.

> this_thread::sleep_for(chrono::milliseconds(500));

Perhaps a longer delay here would help.
---

A few other tips:
* I notice you are using an old ev3dev image. I don't upgrading will fix this problem, but possibly other problems. When you do upgrade, you will probably have to fix the port names in your program.
* If you are running your program from a terminal, you should start it using `brickrun -- <my-program>`. This does nice things like stop the motors if your program crashes and lets you use the display on the EV3 if you want to.

RyanJoergensen, 2019-06-16:
Hi,
Just to comment - I have a somewhat similar issue, also the SensorMUX, running the program (Python) from Visual Studio Code as per the new guide. 
Can read values, but getting PermissionError: [Errno 13] Permission denied: '/sys/class/lego-sensor/sensor56/mode', when trying to set the mode.
However, when running the program via a SSH terminal (MobaXterm), then I have no issues.
I don't know if this is any relevance, just fell over this issue post when trying to search for a solution myself.
If not, please disregard.

dlech, 2019-06-16:
@RyanJoergensen that is a different error, but the same advice applies. If there is not a long enough delay after `set_device` is called, then the permissions might not be updated by the OS yet when you try to access the device.

RyanJoergensen, 2019-06-17:
Thanks a lot, a longer delay made it work (1 sec.).
0.5 sec was enough when running via SSH before.