ev3dev/ev3dev #1285: Support ethernet networking on the Raspberry Pi 3 Model B+

salamanders, 2019-06-09:
Networking (ethernet) does not seem to work on the RPI3B+.

I've confirmed that the exact same hardware ([rpi3b+](https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/), [microSD](https://smile.amazon.com/gp/product/B073K14CVB), cable, etc) works with Raspbian for Robots and default raspbian.

Yet when flashing either the latest [stable](https://github.com/ev3dev/ev3dev/releases/download/ev3dev-stretch-2019-03-03/ev3dev-stretch-rpi2-generic-2019-03-03.zip) or [snapshot](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2019-05-30/snapshot-ev3dev-stretch-rpi2-generic-2019-05-29.img.xz) Stretch release of ev3dev, the network never gets an IP address.

Which makes it hard to log in to further debug.

Sack Exchange [thinks it is because networking stuff changed in the "plus" model](https://raspberrypi.stackexchange.com/a/99394/10540).

I don't know if this also applies to wireless networking, as I couldn't log in to set up the wifi.

salamanders, 2019-06-09:
Whoa, network works after some blind file replacement.

1. On the advice of [this reply](https://raspberrypi.stackexchange.com/a/99394/10540) I started with snapshot-ev3dev-buster-rpi2-generic-2019-05-29
2. Then overlayed latest https://github.com/Hexxeh/rpi-firmware
3. Then overlayed https://github.com/raspberrypi/firmware/blob/master/boot/bcm2710-rpi-3-b-plus.dtb
4. Tried [wifi config via file](https://www.raspberrypi.org/forums/viewtopic.php?t=191252) (longshot)
5. GOT AN IP ADDRESS YAY.  `sudo apt-get update && sudo apt-get dist-upgrade -y`
6. [installed raspi-config](https://raspberrypi.stackexchange.com/a/85491/10540)  then `sudo raspi-config`
7. Updated raspi-config, expanded filesystem
7. No wifi found :(  BUT - VERY happy to be able to log into it using ev3dev!

salamanders, 2019-06-09:
I may be more far of than I thought, `/sys/class/power_supply/brickpi3-battery` doesn't exist.  (which is the [platform detection from the standard java installer script](https://github.com/ev3dev-lang-java/installer/blob/master/installer.sh#L60))

dlech, 2019-06-09:
You have to enable brickpi3 in `config.txt` for any of the BrickPi3 hardware to show up in `/sys/`

salamanders, 2019-06-09:
Cool! So that leaves wifi.

salamanders, 2019-06-09:
Cool! So that leaves wifi.
But... this seems dodgy. For another $35 I can get a rpi 3b NOTPLUS and sell the 3b+ off to a friend. 

Advice? Wait for 3b+ support, or do these things take a few months?

salamanders, 2019-06-11:
I swapped my RPI 3B+ for a 3B (not-plus) and everything works great!  So works-for-me, but other 3b+ be forewarned.

closed 2019-06-11 by salamanders