ev3dev/ev3dev #1287: Support for NXT sensors on micropython

LeeJenkins, 2019-06-19:
**Question**
Is it possible to use the NXT sensors in the micropython environment?

**Background Information**
I am trying to use NXT light sensors because I have several of them. We're building multiple instances of an FLL robot that uses the light/color sensors to find/follow/square-on a black line. We are super-excited about using python, except we cannot use our inventory of NXT light sensors.

I wrote a little HTTP server in python3 with an API to create and query NXT sensors. I ran the service in the background and I wrote a client in micropython. The setup works like a champ, except it's extremely slow. I timed the average read time for reflected light intensity on the sensors for comparison: for a ColorSensor object on micropython, the average read time was 1.7ms; for a LightSensor object in python3, the average read time was about 5-6ms; but using an HTTP request from micropython to get the sensor value from the python3 HTTP server took 69ms on average.

I could write a low-level socket interface, but that's a lot of work (for me, anyway) and I don't know how much real improvement it would afford. Where is the overhead? Is it all HTTP? How much is in the socket interface? How much is just the multiprocessing/task-switch limitation of the EV3?

So before I dive into that pool, I have questions:
* Is there a better way to do this?
* Is support for NXT sensors planned in the very near future?
* Is there something I can do to help make it happen?

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: Not on hand, but I can provide if necessary.

dwalton76, 2019-06-19:
> **Question**
> Is it possible to use the NXT sensors in the micropython environment?
> 

I haven't tried it myself and don't recall if anyone tested it when we added micropython support for EV3 sensors.  I would say it is definitely worth a try.

> I could write a low-level socket interface, but that's a lot of work (for me, anyway) and I don't know how much real improvement it would afford. Where is the overhead? Is it all HTTP? How much is in the socket interface? How much is just the multiprocessing/task-switch limitation of the EV3?

I doubt you would see a huge improvement.   If you do decide to try something else, before writing your own low-level socket interface you could give [MQTT](https://github.com/micropython/micropython-lib/tree/master/umqtt.simple) a try.

> So before I dive into that pool, I have questions:
> 
> * Is there a better way to do this?

I think what you are doing sounds reasonable.

> * Is support for NXT sensors planned in the very near future?
> * Is there something I can do to help make it happen?

Give the NXT sensors a try.  If they do not work you could add support for them to [ev3dev-lang-python](https://github.com/ev3dev/ev3dev-lang-python), just submit a pull request and we can take a look at the patch.  We are always looking for more folks to contribute :)

LeeJenkins, 2019-06-19:
@dwalton76 Thanks for the quick reply. Just to clarify, the NXT sensors are already supported in ev3dev-lang-python, which is the package with the bindings for python3 that I used in my HTTP server.

My question is about support for the NXT sensors in micropython, not python3.

dwalton76, 2019-06-19:
ah, I see where we are getting our wires crossed.  ev3dev-lang-python supports micropython too so my answer was based around that.  This is not the official micropython solution in the recent lego ev3dev image though (although ev3dev-lang-python is included in the image).  The official micropython in the lego image is the `pybricks` module, is that what you are using?  I haven't used pybricks much so not sure if the NXT sensors would work.

LeeJenkins, 2019-06-19:
Oh, I didn't realize there were two micropython solutions! You're right, we are using the lego ev3dev image and ev3dev-lang-python is included on that image. And yes, I am using pybricks. I tried importing a LightSensor class like this:

    from pybricks.ev3devices import (ColorSensor, LightSensor)

The ColorSensor class works just fine, but LightSensor generates an ImportError. I also tried importing from ev3dev (again in my micropython program) like this:

    from ev3dev.ev3 import LightSensor

This generates an `ImportError: no module named 'ev3dev'`

Are you saying that it should be possible to import ev3dev into the lego image's micropython? Could it be as simple as adding the ev3dev package directory to the micropython search path? I don't actually have any idea where either of those things are (the package and the path variable).

LeeJenkins, 2019-06-20:
Follow-up: I've hacked around a bit trying to get something working, but clearly I have no idea what I'm doing. :-) For example, I printed the sys.path from a python3 program and then copied & pasted to extend the sys.path of my micropython program. It imported a couple of packages, but it eventually failed with the message `ImportError: no module named '_ctypes'`. Other more involved experiments failed just as spectacularly.

Here's a question: are the pybricks and ev3dev packages compatible? Is it possible to import both of them into the same micropython program?

dlech, 2019-06-20:
Try this:

```python
from pybricks.ev3devio import Ev3devSensor

class LightSensor(Ev3devSensor):
    _ev3dev_driver_name = 'lego-nxt-light'

    def ambient(self):
        self._mode('AMBIENT')
        return self._value(0) / 10

    def reflection(self):
        self._mode('REFLECT')
        return self._value(0) / 10
```

Most of the information comes from here: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#lego-nxt-light

LeeJenkins, 2019-06-20:
@dlech It worked! That is really awesome. Thank you so much for sharing the code. It looks like other sensors, like the HiTechnic EOPD or microinfinity gyro can also be supported by using the same model (driver name, method, mode, and value), right?

dlech, 2019-06-20:
The microinfinity is a bit special since it can't be automatically detected, but for the most part, yes you can create a class for most sensors that way.

closed 2019-06-20 by LeeJenkins

LaiYanKai, 2019-07-03:
Sorry I've to reopen this:
Regarding the HiTechnic NXT color sensor V2:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-color-v2

I've been trying to create a class in **pybricks-micropython**:
```
class HTColor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    _MODE_COLOR = 0
    _MODE_RGBW = 1
    def __init__(self, val):
        Ev3devSensor.__init__(self, val)
        self.m = self._MODE_COLOR
        self._mode('RGB')
        self._number_of_values = 4 # I want to extract tuple of [R,G,B,W] values
    def rgbw(self):
        if self.m != self._MODE_RGBW:
            self._mode('RGB')
            self.m = self._MODE_RGBW
        self._number_of_values = 4
        return self._value(0) # how to extract tuple / list?
    def color(self):
        if self.m != self._MODE_COLOR:
            self._mode('COLOR')
            self._number_of_values = 1
            self.m = self._MODE_COLOR
        return self._value(0)
```
I've tried to use this for the HT color sensor v2. But I have two problems:

1) I can only access the _red_ value, but not the other _blue_, _green_ and _white_ values, after using the following:
```
ht4 = HTColor(Port.S4)
print(ht4.rgbw())
```
**How to access the other 3 values?** Is there support for multiple values? Changing the _number_of_values does not work, neither does passing values higher than zero in self._value().

2) The color value is located at register 66, the red at register 67, the green at 68, blue at 69 and white at 70. **It is probably much faster reading these five registers at one go instead of changing modes.** Is there a way to do this?