ev3dev/ev3dev #1288: What is real brand of chip gyro in LEGO 45505 EV3 Gyro Sensor?

xdsl, 2019-06-19:
What is real brand of chip gyro in LEGO 45505 EV3 Gyro Sensor? Lego EV3 HDK (Appendix 8) says: ISZ-655 (https://www.lego.com/r/www/r/mindstorms/-/media/franchises/mindstorms%202014/downloads/firmware%20and%20software/advanced/lego%20mindstorms%20ev3%20hardware%20developer%20kit.zip). 

ev3dev support (http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#lego-ev3-gyro) 2-axis gyro, but ISZ-655 - single-axis gyro! (https://store.invensense.com/datasheets/invensense/PS-ISZ-0655B.pdf). 

Photo of the chip on my gyro: http://rubirobot.ru/files/other/gyro.jpg

dlech, 2019-06-19:
There are multiple versions of the gyro sensor with different chips in them. http://ev3lessons.com/en/ProgrammingLessons/advanced/GyroRevisited.pdf

I am pretty sure the one in your picture is IDG-2020: https://www.invensense.com/products/motion-tracking/2-axis/idg-2020/

xdsl, 2019-06-21:
Unlikely. 
IDG-2020 and IXZ-2020 have max full scale range 250 dps.
IDG-2020 also not use Z-axis.

dlech, 2019-06-21:
Must be a discontinued or unlisted part then. I also found this: http://www.invensense.com/wp-content/uploads/2018/06/PS-Ixx-2021A-00-v2-2.pdf

The product marking for IXZ-2021 is IZ21, so maybe there is such thing as a IXZ-2025 that has a higher range?

xdsl, 2019-06-24:
I checked two N5-gyro and one N6-gyro. Each has the same part number IZ25. Very similar to IXZ-20xx.
The problem is that I was getting gyro rate 300 dps and more, but IXZ-20xx max full scale range is 250 dps.