ev3dev/ev3dev #1290: UART communication over Sensor ports

runningOutOfUsernameIdeas, 2019-07-03:
Hello there,

I am currently trying to send bytes through the sensor ports of the ev3 brick using UART under tty. I am able to receive bytes from another device connected, however I am unable to send them.  It also appears that bytes are being sent to the other device (224 or 0xE0), however, I did not send them.

I have done some research and found something called the 'ev3-uart-sensor-ld driver'. It appears this may be interfering in what I want to do with the serial port (or that's what I think currently). 

This website states that 'The  ev3-uart-sensor-ld driver is implemented as a line discipline, which basically means that it **takes over your serial port** and makes it look like something else. '
https://lechnology.com/2014/09/using-uart-sensors-on-any-linux/

This website states that 'In order to detach the line discipline, kill(1) the ldattach process.' (Although I have no idea how to use kill to do that)
https://manpages.debian.org/stretch/util-linux/ldattach.8.en.html

If you think that the 'ev3-uart-sensor-ld driver' is the problem, how will I be able to disable this from only 1 of the sensor ports?

If you don't think that is the problem, do you have any idea what is causing these problems for me?

Thanks for the reply.

dlech, 2019-07-03:
You can put the port in `other-uart` mode.

http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/ev3.html#ev3-input-port-mode-info

runningOutOfUsernameIdeas, 2019-07-03:
I kind of understand; however, I don't know how to 'put the port into other uart mode'. I'm not familiar with linux or what sysfs is. 

From reading the article, the closest I have gotten is '/sys/devices/platform/ev3-ports/ev3-ports:in1'. I think I might have to edit something there, but I'm not sure (maybe the driver_override file??).

bmegli, 2019-07-03:
Try something along the line (from the command line):

```bash
echo other-uart > [path_to_port]/mode
```

Last time I used it (jessie, older ev3dev), after identifying if it was correct port, it would be:

```bash
echo other-uart > /sys/class/lego-port/port1/mode
```

You should be able to do the same from ev3dev bindings in a language of your choice (should you need)