ev3dev/ev3dev #1291: Stretch gets slower after multiple program runs

LaiYanKai, 2019-07-03:
Tried using compile / exec as in [here](https://github.com/ev3dev/ev3dev-lang-python/issues/204) to load py files. Works like a charm. 
But the python "menu" program I created to load these files takes about 0.3 seconds to respond to button changes, and it gets worse the more I terminate, re-download and test modifications using VS Code. It would take about 1 second to respond before I decided to reboot the EV3 to revert to its faster state. The brickman menu will be affected by these slowness.
This happens even after overclocking the CPU to 456 MHz as in [here](https://lechnology.com/2018/06/overclocking-lego-mindstorms-ev3-part-2/).
I also tried using the `process()` method for the buttons and assigning functions to the `on_...` properties but they behave similarly.
**However, it promptly registers every press event on the LCD screen.**
The while loop that runs the menu also faces the same deteriorating slowness, so I don't think it is just the buttons. Even just making the ev3 beep to respond to button presses (by removing the display to lcd, printing to vscode and parsing over the list containing the filenames) doesn't make it respond promptly (takes at least 0.3 seconds, I didn't time, but I can notice the lag), although the latency has improved.

The code is here:
```
#!/usr/bin/env python3
from sys import *
def console(string):
    print(string, file=stderr)
    # print(string)
console('[INFO] Loading Modules...')
import os
# from ev3dev2.motor import Motor, MediumMotor, OUTPUT_A, OUTPUT_B, OUTPUT_C, OUTPUT_D, MoveTank, MoveSteering
from ev3dev2 import *
from ev3dev2.sensor.lego import ColorSensor, GyroSensor
from ev3dev2.sensor import *
from ev3dev2.sound import Sound
from ev3dev2.display import Display
from ev3dev2.button import Button
from ev3dev2.led import Leds
import ev3dev2.fonts as fonts
sound = Sound()
btn = Button()
lcd = Display()
# leds = Leds()
import math
from threading import Thread
from time import sleep, time
console('[INFO] Finished Loading Modules!')
def _get_py_files():
    files_in_dir = os.listdir()
    py_files = []
    for filename in files_in_dir:
        if filename.endswith(".py") and not filename == "main.py":
            py_files.append(filename)
    return py_files
def run_py_file(filename):
    try:
        console('[INFO] Running ' + filename + '...')
        f = open(filename, 'r')
        code = f.read()
        code = compile(code, filename, 'exec')
        f.close()
        exec(code)
        beep_done()
        console('[INFO] ' + filename + ' terminated')
    except Exception as e:
        console('[ERROR] ' + repr(e))
        beep_err()
def beep_err():
    sound.tone([(880, 100, 30), (440, 100, 30)])
def beep_warn():
    sound.tone([(660, 100, 30), (660, 100, 30)])
def beep():
    sound.play_tone(1760, 0.05, play_type=1)
def beep_init():
    sound.tone([(440, 50, 30), (880, 50, 30), (1760, 50, 30)])
def beep_done():
    sound.tone([(1760, 50, 30), (880, 50, 30), (440, 50, 30)])
def _display_filename(filename):
    if filename is None:
        filename = '<No py files>'
    lcd.rectangle(False, 0, 50, 177, 74, 'white', 'white')
    lcd.text_pixels(filename, False, 0, 50, 'black', 'helvB24')
    lcd.update()
    console('[INFO] ' + _filename + ' selected')
_fileindex = -1
_files = _get_py_files()
if len(_files) > 0:
    _filename = _files[0]
else:
    _filename = None
_display_filename(_filename)
# --- Inits ---
# --- Program started ---
beep_init()
while True:
    # if return button pressed
    if btn.left:
        beep()
        console('[INFO] Terminating...')
        break
    elif btn.up:
        beep()
        if len(_files) > 0:
            _fileindex = (_fileindex + 1) % len(_files)
            _filename = _files[_fileindex]
        _display_filename(_filename)
    elif btn.down:
        beep()
        if len(_files) > 0:
            _fileindex = (_fileindex - 1) % len(_files)
            _filename = _files[_fileindex]
        _display_filename(_filename)
    elif btn.enter:
        beep()
        _files = _get_py_files()
        if len(_files) > 0:
            if _filename == _files[_fileindex]:
                run_py_file(_filename)
            else:
                beep_warn()
                console('[WARN] List of files updated, please reselect')
                _filename = _files[_fileindex]
        else:
            console('[WARN] No files found')
            _filename = None
            _fileindex = 0
            beep_warn()
        _display_filename(_filename)
```
**Desktop (please complete the following information):**
 - OS and version: Windows 10 Pro Build 17134
 - Connection Type: USB or Bluetooth
 - Software: VSCode

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2019-03-03.zip and
ev3micropythonv100sdcardimage-4b8c8333736fafa1977ee7accbd3338f.zip

The same happens for both the stretch and micropython images. I'm just gonna paste the micropython one here:
```
========== ev3dev-sysinfo ==========
Image file:         ev3-micropython-v1.0.0-sd-card-image
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           227.32
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534EC92D
BOARD_INFO_TYPE=main
```

dlech, 2019-07-03:
Can you post the code for your menu program?

My guess is that the slowness is because you are using all available RAM and getting into swap memory. It could be that the menu program is not releasing old copies of the program when they end.

LaiYanKai, 2019-07-03:
```
#!/usr/bin/env python3
from sys import *
def console(string):
    print(string, file=stderr)
    # print(string)
console('[INFO] Loading Modules...')
import os
# from ev3dev2.motor import Motor, MediumMotor, OUTPUT_A, OUTPUT_B, OUTPUT_C, OUTPUT_D, MoveTank, MoveSteering
from ev3dev2 import *
from ev3dev2.sensor.lego import ColorSensor, GyroSensor
from ev3dev2.sensor import *
from ev3dev2.sound import Sound
from ev3dev2.display import Display
from ev3dev2.button import Button
from ev3dev2.led import Leds
import ev3dev2.fonts as fonts
sound = Sound()
btn = Button()
lcd = Display()
# leds = Leds()
import math
from threading import Thread
from time import sleep, time
console('[INFO] Finished Loading Modules!')
def _get_py_files():
    files_in_dir = os.listdir()
    py_files = []
    for filename in files_in_dir:
        if filename.endswith(".py") and not filename == "main.py":
            py_files.append(filename)
    return py_files
def run_py_file(filename):
    try:
        console('[INFO] Running ' + filename + '...')
        f = open(filename, 'r')
        code = f.read()
        code = compile(code, filename, 'exec')
        f.close()
        exec(code)
        beep_done()
        console('[INFO] ' + filename + ' terminated')
    except Exception as e:
        console('[ERROR] ' + repr(e))
        beep_err()
def beep_err():
    sound.tone([(880, 100, 30), (440, 100, 30)])
def beep_warn():
    sound.tone([(660, 100, 30), (660, 100, 30)])
def beep():
    sound.play_tone(1760, 0.05, play_type=1)
def beep_init():
    sound.tone([(440, 50, 30), (880, 50, 30), (1760, 50, 30)])
def beep_done():
    sound.tone([(1760, 50, 30), (880, 50, 30), (440, 50, 30)])
def _display_filename(filename):
    if filename is None:
        filename = '<No py files>'
    lcd.rectangle(False, 0, 50, 177, 74, 'white', 'white')
    lcd.text_pixels(filename, False, 0, 50, 'black', 'helvB24')
    lcd.update()
    console('[INFO] ' + _filename + ' selected')
_fileindex = -1
_files = _get_py_files()
if len(_files) > 0:
    _filename = _files[0]
else:
    _filename = None
_display_filename(_filename)
# --- Inits ---
# --- Program started ---
beep_init()
while True:
    # if return button pressed
    if btn.left:
        beep()
        console('[INFO] Terminating...')
        break
    elif btn.up:
        beep()
        if len(_files) > 0:
            _fileindex = (_fileindex + 1) % len(_files)
            _filename = _files[_fileindex]
        _display_filename(_filename)
    elif btn.down:
        beep()
        if len(_files) > 0:
            _fileindex = (_fileindex - 1) % len(_files)
            _filename = _files[_fileindex]
        _display_filename(_filename)
    elif btn.enter:
        beep()
        _files = _get_py_files()
        if len(_files) > 0:
            if _filename == _files[_fileindex]:
                run_py_file(_filename)
            else:
                beep_warn()
                console('[WARN] List of files updated, please reselect')
                _filename = _files[_fileindex]
        else:
            console('[WARN] No files found')
            _filename = None
            _fileindex = 0
            beep_warn()
        _display_filename(_filename)
```
The code is in the issue above... up and down buttons navigate the menu. the enter/center button runs the compile/exec code. The up and down buttons do not reload the pyfiles in the directory, they simply iterate over an older copy of the menu of py files, stored in a list (_files)

But even if I do not use the exec and compile (by simply not pressing the center/enter button) and just playing the beep when the up or down buttons are pressed, the lag still exists, even after a fresh reboot.

You could comment out 
```
        if len(_files) > 0:
            _fileindex = (_fileindex + 1) % len(_files)
            _filename = _files[_fileindex]
        _display_filename(_filename)
```
for both up and down buttons and try it...

I suspect the problem lies in the python3 environment. A similar logical structure in the python-micropython environment (of course using the micropython methods) works really fast.