ev3dev/ev3dev #1292: EV3 Micropython - Issues

LaiYanKai, 2019-07-03:
This is for **pybricks-micropython**

**(1) Appropriate & Fast way to retrieve Color, RGB and White values from HT NXT Color V2?**
This is on top of an issue I have with the HiTechnic Color Sensor V2 in:
[https://github.com/ev3dev/ev3dev/issues/1287#issuecomment-507960364](https://github.com/ev3dev/ev3dev/issues/1287#issuecomment-507960364)

**(2) How to erase row of text on LCD Screen?**
Another issue is to display text onto the LCD screen. There is no way to erase current text other than clearing the screen, which takes quite a bit of time and makes the values flash on and off.

Usually, we would want to create a "port-view" program that displays sensor values on the screen, next to titles that need not be overwritten. This allows us to calibrate or troubleshoot the robot easily. These titles are written once at the start of the program, and the sensor values are written next to them as the program runs. **To update the screen quickly, we do it generally by padding trailing spaces to the values so that only the previous values will be cleared without touching the entire screen**.

This is simply not the case for micropython. Overwriting the same location with text and spaces simply **jumbles up the past and current text together**, so the values become unreadable. Even using a n x 177 pixel (n corresponds to row height, i tried different values) png image with white background to clear the row does not work (a png image with black background does overwrite the row with a black rectangle)

**(3) Need multiple reboots for Bluetooth to show as available **
I have briefly created steps to connect the Micropython image / Brickman to Windows 10 using Bluetooth: [Instructions for Bluetooth Connection.pdf](https://github.com/ev3dev/ev3dev/files/3353782/Instructions.for.Bluetooth.Connection.pdf)

However, even with the latest stretch (March 2019) or micropython image, the Bluetooth still doesn't show up occasionally after booting. I know that there are one or two issues related to this. Is there currently a solution? It's going to be a no go for competitions if this persists...

dlech, 2019-07-03:
> Sorry I've to reopen this:
> Regarding the HiTechnic NXT color sensor V2:
> http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-color-v2
> 
> I've been trying to create a class in **pybricks-micropython**:
> 
> ```python
> class HTColor(Ev3devSensor):
>     _ev3dev_driver_name = 'ht-nxt-color-v2'
>     _MODE_COLOR = 0
>     _MODE_RGBW = 1
>     def __init__(self, val):
>         Ev3devSensor.__init__(self, val)
>         self.m = self._MODE_COLOR
>         self._mode('RGB')
>         self._number_of_values = 4 # I want to extract tuple of [R,G,B,W] values
>     def rgbw(self):
>         if self.m != self._MODE_RGBW:
>             self._mode('RGB')
>             self.m = self._MODE_RGBW
>         self._number_of_values = 4
>         return self._value(0) # how to extract tuple / list?
>     def color(self):
>         if self.m != self._MODE_COLOR:
>             self._mode('COLOR')
>             self._number_of_values = 1
>             self.m = self._MODE_COLOR
>         return self._value(0)
> ```
> 
> I've tried to use this for the HT color sensor v2. But I have two problems:
> 
>     1. I can only access the _red_ value, but not the other _blue_, _green_ and _white_ values, after using the following:
> 
> 
> ```python
> ht4 = HTColor(Port.S4)
> print(ht4.rgbw())
> ```
> 
> **How to access the other 3 values?** Is there support for multiple values? Changing the _number_of_values does not work, neither does passing values higher than zero in self._value().
> 
>     1. The color value is located at register 66, the red at register 67, the green at 68, blue at 69 and white at 70. **It is probably much faster reading these five registers at one go instead of changing modes.** Is there a way to do this?

dlech, 2019-07-03:
Try this:

```python
class HTColor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    _MODE_COLOR = 0
    _MODE_RGBW = 1
    def rgbw(self):
        if self.m != HTColor._MODE_RGBW:
            self._mode('RGB')
            self.m = HTColor._MODE_RGBW
        return self._value(0), self._value(1), self._value(2), self._value(3)
    def color(self):
        if self.m != HTColor._MODE_COLOR:
            self._mode('COLOR')
            self._number_of_values = 1
            self.m = HTColor._MODE_COLOR
        return self._value(0)
```

Then you can write this to get the values:

```python
red, green, blue, white = sensor.rgbw()
```

dlech, 2019-07-03:
> Need multiple reboots for Bluetooth to show as available

Known issue. See https://github.com/ev3dev/ev3dev/issues/925

LaiYanKai, 2019-07-04:
**Regarding the HT Sensor**
Nope the code doesn't work. I also tried the following:

```python
class HTColorSensor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    def rgbw(self):
        self._mode('RGB')
        self._number_of_values = 4
        r = self._value(0)
        g = self._value(1)
        b = self._value(2)
        w = self._value(3)
        return (r,g,b,w)
    def color(self):
        self._mode('COLOR')
        self._number_of_values = 1
        return self._value(0)
```
Gives me an identical error
```Python traceback
Traceback (most recent call last):
  File "/home/robot/Test/main.py", line 68, in <module>
  File "/home/robot/Test/main.py", line 26, in rgbw
  File "pybricks/ev3devio.py", line 89, in _value
IndexError: list index out of range
```
Line 26 is
```Python
g = self._value(1)
```
Is there a source code for this? I've been using the `inspect` module but i could only used `getmembers()`...

**How to erase row of text on LCD Screen?**
Are there updates on the matter?

As far as I can see Micropython is really fast and good, of course, with the 456 MHz overclocking. The timing for each loop iteration seems to be on par with EV3G. I don't currently suffer from lag issues like ev3dev2, or perhaps the footprint is just too small to notice.
We need a lot of patience to load comprehensive programs / utility / troubleshooting programs on EV3G on the PC. And really... textual languages are the next evolutionary step for LEGO to stay competitive in robotics.
The bluetooth is an issue, but if we can get the third party I2C sensors to work, and find a way to clean up text without clearing the screen (we need this to do debugging, for example, on the algorithm we use to ID colors are working, or simply, finding the mode of white and black values on the entire map), we will most definitely use this.

dlech, 2019-07-04:
`_number_of_values` is a class attribute rather than an instance attribute, so it has to be set like this:

```python
class HTColorSensor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    _number_of_values = 4
    _default_mode = 'RGB'

    def rgbw(self):
        self._mode('RGB')
        r = self._value(0)
        g = self._value(1)
        b = self._value(2)
        w = self._value(3)
        return (r,g,b,w)

    def color(self):
        self._mode('COLOR')
        return self._value(0)
```

dlech, 2019-07-04:
> How to erase row of text on LCD Screen?

I don't have a good solution for this. 

As a workaround, you can probably get a [MicroPython framebuffer](https://docs.micropython.org/en/v1.9.4/pyboard/library/framebuf.html) via `pybricks.ev3brick.display._display._fb`. Then you can modify that framebuffer and then copy it to the screen with `pybricks.ev3brick.display._display._screen.update(pybricks.ev3brick.display._display._data)`.

LaiYanKai, 2019-07-04:
Nice, the following works:
```Python
class HTColorSensor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    _number_of_values = 4
    def rgbw(self):
        self._mode('RGB')
        while True:
            try:
                return self._value(0), self._value(1), self._value(2), self._value(3)
            except:
                errors += 1 # may give ENODEV error
                if errors > 1:
                    return 0, 0, 0, 0
    def color(self):
        self._mode('COLOR')
        errors = 0
        while True:
            try:
                return self._value(0)
            except:
                errors += 1 # may give ENODEV error
                if errors > 1:
                    return 0
```
The while loop is to avoid errors. There is probably a better way to code this.

I'll work on the framebuffer  thing tmr and the ev3dev2 menu thing later and keep this updated.

LaiYanKai, 2019-07-04:
I forgot, is there a way to send the command `50HZ` and `60Hz` to the HT color sensor?
[http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-color-v2](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-color-v2)

dlech, 2019-07-04:
> I forgot, is there a way to send the command `50HZ` and `60Hz` to the HT color sensor?

There isn't a built-in way to do this, but this might work:

```python

class HTColorSensor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    _number_of_values = 4
    
    def _open_files(self):
        self._command_file = open(self.path + 'command', 'w')
        super()._open_files()

    def _close_files(self):
        self._command_file.close()
        super()._close_files()

    def set_mains(self, hz):
        self._command_file.write(hz)
        self._command_file.flush()
```

LaiYanKai, 2019-07-05:
Thanks, it works!
```Python
class HTColorSensor(Ev3devSensor):
    _ev3dev_driver_name = 'ht-nxt-color-v2'
    _number_of_values = 4
    def _open_files(self):
        super()._open_files()
        self._command_file = open(self.path + 'command', 'w')
    def _close_files(self):
        super()._close_files()
        self._command_file.close()
    # configure the sensor for 50Hz power mains
    def set_mains_50hz(self):
        self._command_file.write('50HZ')
        self._command_file.flush()
    # configure the sensor for 60Hz power mains
    def set_mains_60hz(self):
        self._command_file.write('60HZ')
        self._command_file.flush()
    # returns the tuple of color values: (red, blue, green white). 0 to 255 for each
    def rgbw(self):
        self._mode('RGB')
        errors = 0
        while True:
            try:
                return self._value(0), self._value(1), self._value(2), self._value(3)
            except:
                errors += 1 # may give ENODEV error
                # returns tuple of zeros if there are two consecutive errors
                if errors > 1:
                    return 0, 0, 0, 0 
    # returns the color index. 0 to 17
    def color(self):
        self._mode('COLOR')
        errors = 0
        while True:
            try:
                return self._value(0)
            except:
                errors += 1 # may give ENODEV error
                # returns zero if there are two consecutive errors
                if errors > 1:
                    return 0
```
My country's mains is 50hz. i did a test by pointing the sensor onto a brown table top for about a minute, 8cm from the surface. The rgbw values for 50hz varies by about one throughout the entire time (white is mostly 29, sometimes 30), and is quite steady. Setting it to 60hz causes them to fluctuate quite obviously (white fluctuates from 27 to 31).

**Note for anyone looking at this**: while it may be able to return values for every 10 milliseconds or so, it usually takes about 30 milliseconds or so for the values to change. HT sensor is usually slower than the EV3 color sensor, but offers a better range.

LaiYanKai, 2019-07-05:
----
(1) Micropython bug: coordinates are the bottom-left, not top-left, of text
----
There seems to be a problem with the coordinates. They are not top-left as in the manual
The following code yields this on the screen:
```Python
brick.display.text('a1234567890', (0, 0))
brick.display.text('b1234567890', (0, 8))
brick.display.text('c1234567890', (0, 16))
```
[image]
While the following:
```python
brick.display.text('a1234567890', (0, 8))
brick.display.text('b1234567890', (0, 16))
brick.display.text('c1234567890', (0, 24))
```
[image]

----
(2) Display2 Class
----
Anyway to solve the text problem I've implemented a class Display2, which is based off your code:
```python
class Display2():
    def __init__(self, brick):
        self._disp = brick.display
        self._dv = self._disp._device
        self._fb = self._dv._fb
        self._font_height = 7
        self.set_line_height(10)
        self._font_width = 8
        self._auto_update = True
        self._prev_text_location = (0, -self._line_height)
    def set_auto_update(self, auto_update):
        # auto_update to display on the screen once the methods are called
        # for much faster implementation, it is better to use set_auto_update(False),
        # and call update() after making the necessary changes to screen
        self._auto_update = auto_update
    def set_line_height(self, line_height):
        # the method to set the line height, useful for scrolling only
        self._line_height = int(line_height)
        self._padding_top = int((self._line_height - self._font_height) / 2)
    def _update(self, update):
        if (self._auto_update and update != False) or update:
            self.update()
    def update(self):
        # writes changes to screen
        self._dv._screen.update(self._dv._data)
    def text(self, text, coordinate=None, grayscale=0, update=None):
        # get color
        grayscale = Display2._gray_to_hex(grayscale)
        # decide if scrolling is needed, otherwise, use the supplied tuple
        if coordinate is None:
            # auto positioning required
            new_coordinate = (self._prev_text_location[0], self._prev_text_location[1] + self._line_height)
            if new_coordinate[1] <= -self._padding_top:
                new_coordinate = (new_coordinate[0], 0)
            if new_coordinate[1] <= 128 - self._font_height - self._padding_top:
                # the next line is still visible on the screen
                coordinate = new_coordinate
                self._prev_text_location = coordinate
            else:
                coordinate = self._prev_text_location
                # scroll up by one line height
                self._fb.scroll(0, -self._line_height)
        # clear only the space where the font is written
        self._fb.fill_rect(coordinate[0], coordinate[1], len(text) * self._font_width, self._line_height, 0xffffff)
        # write the text
        self._fb.text(text, coordinate[0], coordinate[1] + self._padding_top, grayscale)
[... 69 lines cut ...]
```
[image]
____
Playing with the grayscale:
```python
disp = Display2(brick)
disp.set_auto_update(False)
disp.set_line_height(4)
for i in range(16):
    disp.text('hi' + str(i), None, float(i)/4)
disp.update()
```
[image]
____
Combining it all:
```python
disp = Display2(brick)
disp.set_auto_update(False)
disp.set_line_height(8)
for i in range(20):
    disp.text('hi' + str(i), None)
disp.text('Hello World', (0, 45), 1)
disp.update()
```
[image]

dlech, 2019-07-05:
Nice work!

LaiYanKai, 2019-07-16:
This whole comment discusses the Display2 class, error-handling and debugging, as well as a menu program that can make our lives a lot easier during debugging.

# (1) Updated Display2 Class + New Grayscale Class
- This new class gives the user the advantage of **wrapping** the text around the screen. This is particularly useful for **debugging**, such as calling `display.text(traceback.format_exc(), wrap=True)` to display long exception messages, of course after importing `traceback`, and after instantiating `display = Display2()`. The code has been "cleaned" for this application.
- The new **Grayscale** Class is similar to the parameter classes, used for coloring the fonts and shapes
- The image drawing, triangle drawing, etc. is left to the user to design. Otherwise, the original display class can be used.
- The commented out `capture()` was intended to capture the framebuffer before the service method `_update()` is called, which transfers the framebuffer in memory to the display device, something like a screenshot. The commented out `print()` can then be used to then print the captured framebuffer to screen again, using the `FrameBuffer` class `blit` method. These two methods are thus useful for reverting the display to a previous image. ***However***, these methods were abandoned because of the high memory overhead. If you want to use it, please adjust the `capture()` method because that does not currently create a copy of the `bytearray` when calling `self._fb`. You can then simply use a `list()` method to change the `bytearray`, 4 bytes per pixel, adjust the bytes, and convert it back to a `bytearray` using `bytearray()`. Also, note that the fourth byte is useless, and the methods in the `FrameBuffer` class is from the 3rd byte to the 1st byte. You will know what I mean if you examine the `FrameBuffer` class.

```Python
class Grayscale():
    BLACK = 0
    DARK_GRAY = 0x666666
    LIGHT_GRAY = 0xaaaaaa
    WHITE = 0xffffff

class Display2():
    def __init__(self, brick):
        self._disp = brick.display
        self._dv = self._disp._device
        self._fb = self._dv._fb
        self._font_height = 7
        self.set_line_height(10)
        self._font_width = 8
        self._auto_update = False
        self._prev_row = -1
        self._max_chars_in_row = 178 // self._font_width
    def _update(self, update):
        if (self._auto_update and update != False) or update:
            self.update()
    # def capture(self):
    #     # must be called before update is called. update clears the buffer.
    #     return self._fb
    # def print(self, captured_data, update=None):
    #     self._fb.blit(captured_data, 0, 0, 0xffffff)
    #     self._update(update)
    def set_auto_update(self, auto_update):
        # auto_update to display on the screen once the methods are called
        # for much faster implementation, it is better to use set_auto_update(False),
        # and call update() after making the necessary changes to screen
        self._auto_update = auto_update
    def set_line_height(self, line_height):
        # the method to set the line height, useful for scrolling only
        self._line_height = int(line_height)
        self._padding_top = int((self._line_height - self._font_height) / 2)
        self._max_num_rows = 128 // self._line_height
    def reset_prev_row(self, row=0):
        # to reset the previous text location back to the first line,
        # useful for scrolling text only
        self._prev_row = row - 1
    def update(self):
        # writes changes to screen
        self._dv._screen.update(self._dv._data)
    def text(self, text, coordinate=None, grayscale=Grayscale.BLACK, background=Grayscale.WHITE, wrap=False, update=None):
        if wrap:
            # text needs to be wrapped, get the number of rows
            rows = textwrap.wrap(text, self._max_chars_in_row, expand_tabs=False, replace_whitespace=False)
            num_rows = len(rows)
        else:
            # text does not need to be wrapped
            rows = [text]
            num_rows = 1
        
        if coordinate is None:
            # text needs to be auto-positioned
            # find the new _prev_row caused by this wrapping
            new_prev_row = self._prev_row + num_rows
            if new_prev_row >= self._max_num_rows:
                # if the new _prev_row is below the screen, 
                # find the rows to scroll up
                rows_to_scroll = new_prev_row - self._max_num_rows + 1
[... 339 lines cut ...]
            except Exception as e:
                Beeps.warn()
                # print error message to VScode
                print('[ERROR->' + _m_file + ']',  repr(e))
                traceback.print_exc(file=sys.stderr)
                print('[INFO] Killed <' + _m_file + '>')
                print('[INFO] Press any EV3 button to continue...')
                
                # print error to EV3
                display.text('[ERROR->' + _m_file + ']', (0, 0))
                display.text(traceback.format_exc(), wrap=True)
                display.text('Press any button to continue...', wrap=True, update=True)

                # wait for button bump
                while not any(brick.buttons()):
                    pass # wait for press
                while any(brick.buttons()):
                    pass # wait for release
                Beeps.btn()
            print('---------------------------------------')
        _roll_menu(None)

# terminating sound
Beeps.stop(True)
```

LaiYanKai, 2019-07-16:
# (1) Mover class
This is a speed synchroniser / controller that mimics the EV3-G Move Tank, Move Steer and Move On.
It dynamically switches between master / slave synchronisation so that if you stop one wheel, the other will also stop, in a manner that is quite like that in EV3-G. I have tried the cross-synchroniser, but it is impossible to replicate the performance of the LEGO software with that. 
### Methods
First you need to instantiate the class as follows:
```python
left_motor = Motor(Port.B) # use Motor(Port.B, Direction.COUNTERCLOCKWISE) if you want to invert the direction
right_motor = Motor(Port.C)
m = Mover(ml, mr) # use Mover(ml, mr, False) for medium motor bases
```
Then you can use the methods:
```python
end()
```
Used for properly terminating the class instance
```python
angle(motor='B')
```
Gets the angles travelled by the motors since the last reset using `reset_angle()`.
`motor`: 'L' for left, 'R' for right, 'B' for both which is equals to (left + right angles) / 2
```python
reset_angle()
```
Resets the angle without reseting the angles in the motor objects. Used primarily to monitor how many degrees have moved within each Mover method.
```python
tank_on(left_speed, right_speed)
```
"Instantly" tells the motors to move at these speeds. Can be used for line-following or user-designed acceleration / ramping. For large motors, the speeds are preferably between -860 to 860. For medium motors, the speeds are preferably between -1200 to 1200. Please see **Maximum Speeds** for why this is the case.
```python
steer_on(steering, speed)
```
"Instantly" tells the motors to move at these speeds and steering ratio.
`steering`: -100 to spin left, -50 to pivot left, 0 for straight, 50 to pivot right, 100 to spin right
```python
tank_time(left_speed, right_speed, duration, stop_type=Stop.HOLD)
```
`duration` is in milliseconds
`stop_type` can be Stop.HOLD, Stop.BRAKE, Stop.COAST and None. See **Stopping Conditions** below.
Other parameters are the same as above.
```python
steer_time(steering, speed, duration, stop_type=Stop.HOLD)
```
Parameters are same as above.
```python
tank_degrees(left_speed, right_speed, degrees, stop_type=Stop.HOLD)
```
Self-explanatory.
```python
steer_degrees(steering, speed, degrees, stop_type=Stop.HOLD)
```
Self-explanatory.
```python
stop(stop_type=Stop.HOLD)
```
Equivalent to Off block in EV3-G. Also accepts None.
### Dynamic Master/Slave vs Cross Synchronisation
This dynamic master/slave configuration is more reliable than the `DriveBase` class when the robot has to start moving very quickly, or when tracing a line. If the `DriveBase` class cannot be used at high speeds when following a line.
### Speed and Position Controller
There are two controllers that are run when needed, as threads. One is the **speed** controller, which PIDs with speeds as the setpoints, caps the maximum duty-cycle change from the PID controller, and synchronises the motors using dynamic master/slave configurations with hysteresis. 
The other is the **position** controller, which is **only** used for Stop.HOLD, in other words, when motor holding is necessary.
The speed controller is threaded every time `tank_on`, `tank_degrees`, `tank_time`, `steer_on`, `steer_degrees`, `steer_time` are called. If a speed controller is already running, no new speed controllers will be started. The only time the speed controller ends is when holding, coasting or braking is necessary, or when `end()` is called. 
If holding or braking is required, the speed controller's setpoints are set close to zero to synchronise the slowing down of the motors, so the robot is less likely to steer when the robot is suddenly slowing down. Once the speeds are slow enough, the robot will then brake or hold.
The position controller is called only when holding is required. It will end when the speed controller needs to be called.
### Stopping conditions
1. Stop.HOLD
2. Stop.BRAKE
3. Stop.COAST
4. None

The first three are the same as the ones used by the Motor and DriveBase class. **None** does nothing, which is useful if you want to run multiple move methods consecutively without stopping.
[... 592 lines cut ...]
                    pass
            # restore the duty cycle change cap
            self._rc = self._rc_acc
        # final stop action
        if stop_type == Stop.HOLD:
            self.reset_angle()
            # stop speed controller
            self._stop_speed_controller()
            # start position controller to hold motors
            self._angles_sp = (0, 0)
            self._start_position_controller()        
        elif stop_type is not None:
            # stop speed controller
            self._stop_speed_controller()
            # stop position controller
            self._stop_position_controller()
            if stop_type == Stop.BRAKE:
                # set duty cycle to zero
                self.left_motor.dc(0)
                self.right_motor.dc(0)
            else:
                # defaults for stop_type is not none: coast
                self.right_motor.stop(Stop.COAST)
                self.left_motor.stop(Stop.COAST)
```

dlech, 2019-07-16:
FYI, if you just want to write text to the screen, you can just write to stdout. It is a virtual console. It will take care of wrapping and scrolling and whatnot. See #1065 for tips on changing the font.

>     2\. threads are not terminated correctly at the end of the scripts. @dlech is there a way to solve this?

Not that I know of. I don't think MicroPython has the `daemon` argument like Python 3.

LaiYanKai, 2019-07-18:
Sure I'll try that haha. I'll close this for now. Thanks for your input! :)

LaiYanKai, 2019-07-19:
For those interested, the stdout method is as such:
```python
import sys
sys.stdout.write("\x1b[2J\x1b[H") # clear the console
sys.stdout.write('[ERROR->' + filename + ']\n')
traceback.print_exc(file=sys.stdout) # solves the problem of not printing the full traceback
sys.stdout.write('Press any button to continue...\n')
```

LaiYanKai, 2019-07-20:
Hi, is there a way to properly format the docstring on intellisense, I can't seem to find the answer anywhere else. What I mean is to at least add the proper indentations so it is easier to read... do you know of any proper style guide? For example, the following docstring 
```python 
class Display2():
    """An enhanced display class to replace the EV3 Micropython Display class.
    ----------
    Parameters
    ----------
    \u2022 brick : ev3brick
        Imported from pybricks. The ev3brick instance.
    """
```
yields the intellisense help:
[image]
Which doesn't have proper indentations.

The only way I can format is as such, but it is the wrong indentation:
```python
class Display2():
        """An enhanced display class to replace the EV3 Micropython Display class.
        ----------
        Parameters
        ----------
        \u2022 brick : ev3brick
            Imported from pybricks. The ev3brick instance.
        """
```
[image]

dlech, 2019-07-20:
It depends on what provides the intellesense. For example, if you are using VS Code with the official Python extension from Microsoft, it could be Jedi or Microsoft's new language server depending on your settings.

Personally, I prefer numpy style. Is is quite legible and seems to be widely supported by automated tools.
* https://sphinxcontrib-napoleon.readthedocs.io/en/latest/example_numpy.html
* https://numpydoc.readthedocs.io/en/latest/format.html

LaiYanKai, 2019-07-22:
It is Jedi, by default.

It's quite strange that there are two kinds of intellisense popups. One occurs when the user types `(` after a method call, and the other one occurs when we hover our cursor over the method. It's probably just my settings.
______________
For Jedi, we need to do quite a bit to the  to format both properly. If I didn't do that, there will be no indentations on the first kind of popup. Also methods with many parameters will mess up the first popup:
```python
    def text(self, text:str, coordinate=None, grayscale=Grayscale.BLACK, background=Grayscale.WHITE, wrap=False, update=None):
        """\177\r
        Writes text to screen.
        \177\r
        Parameters
        ----------
        \u2022 text : str
            The text to print.
        \u2022 coordinate : tuple, optional
         \177  Coordinate of the top-left corner of the text. Tuple 
         \177  of int (x, y) specifying the x and y coordinates 
         \177  from the top-left corner of the screen. x is 
         \177  positive and in the horizontal direction, from 0 to 
         \177  177 inclusive. y is positive and in the vertical 
         \177  direction, from 0 to 127 inclusive. If None, the 
         \177  text prints on the next line, by default None.
        \u2022 grayscale : Grayscale, optional
         \177  The shade of gray for the text. Use Grayscale 
         \177  attributes. By default Grayscale.BLACK.
        \u2022 background : Grayscale, optional
         \177  The shade of gray for the background. Use Grayscale 
         \177  attributes. By default Grayscale.WHITE.
        \u2022 wrap : bool, optional
         \177  Specfiy True to wrap text around the screen. Works 
         \177  only if coordinate is None. By default False.
        \u2022 update : bool, optional
         \177  Specify True to update the screen. If False, the 
         \177  screen will not update. If None, the screen defaults
         \177  to self._auto_update. Default is None.
        """
```
This is the first kind of popup, which appears after typing `(` to call a method.
[image]
[image]
This is the second kind:
[image]
_________________
Switching to the Microsoft Python Analysis Engine, the first and second popups are now formatted identically. For the following code, a nice format is achieved, though it doesn't follow the numpy docstring formatting or sphinx or anything else.
```python
    def text(self, text:str, coordinate=None, grayscale=Grayscale.BLACK, background=Grayscale.WHITE, wrap=False, update=None):
        """Writes text to screen.
        
        Parameters
        --
            text : str
        > The text to print.

            coordinate : tuple, optional
        > Coordinate of the top-left corner of the text. Tuple of int (x, y) specifying the x and y coordinates from the top-left corner of the screen. x is positive and in the horizontal direction, from 0 to 177 inclusive. y is positive and in the vertical direction, from 0 to 127 inclusive. If None, the text prints on the next line, by default None.
        
            grayscale : Grayscale, optional
        > The shade of gray for the text. Use Grayscale attributes. By default Grayscale.BLACK.
        
            background : Grayscale, optional
        > The shade of gray for the background. Use Grayscale attributes. By default Grayscale.WHITE.
        
            wrap : bool, optional
        > Specify True to wrap text around the screen. Works only if coordinate is None. By default False.

            update : bool, optional
        > Specify True to update the screen. If False, the screen will not update. If None, the screen defaults to self._auto_update. Default is None.

        """
```
We don't really need to wrap the docstring, because the engine will take care of it. Go to View > Toggle Word Wrap to wrap the docstring in the editor for easier reading.
[image]
[image]

You can use `# `, `## `, `### `, `#### `, `##### ` etc. before a phrase to print headings of decreasing sizes.
You can use `- ` or `+ ` for bullet points before a phrase.
`--` on the next line to print a large heading.
`> ` and `>> ` for indentations.
The indentation of the docstring must be correct
____
You can change the engine by unchecking the Python:Jedi Enabled box in the Settings (File>Preferences>Settings, then search for "Intellisense" or "Jedi"):
[image]

LaiYanKai, 2019-07-22:
Is there a way to interrupt threads? I looked into ctypes, sys, _thread  etc. modules to try and interrupt the threads but I can't find anything.
I'm looking into exiting a program run by the menu script by pressing the left button, but the only way to do it now is to check some flags during the program, which is slow and can be problematic in many ways.
I know you said that daemon doesn't work... because there isn't any, so I'm wondering if you know of any other way, low level or not?
By the way, is it possible to just load the environment at startup so we don't have to keep calling the same libraries? For now, only one brick is needed...

LaiYanKai, 2019-07-23:
> Is there a way to interrupt threads? I looked into ctypes, sys, _thread etc. modules to try and interrupt the threads but I can't find anything.
> I'm looking into exiting a program run by the menu [...]

I'm going to play with os.fork(). Had some code working, will update if successful

LaiYanKai, 2019-08-23:
Hi, low level examples and the source code are currently not available. @dlech is one of the writers of this? Most hints come from him, but you can try searching first what is available, particularly the more widely implemented ones, like threading, sys exceptions, math etc.

You can get some hints on what methods / properties are available by opening the ssh terminal on vscode, and using "help" once in the "pybricks-micropython" environment. However, it is not possible to get hints on the methods' parameters, what they do, or what they return.

To see what methods / properties are available:
1. Connect to your EV3 on VScode. 
2. Open the SSH terminal by right clicking the "ev3dev" (assuming u didn't change the brick host name) you see under "ev3dev device browser". 
3. Once the terminal has loaded, type "pybricks-micropython", without the double quotes, into the terminal. You have now started an environment (is that what they call it?) where u can test code used by EV3 micropython. For example, you can try "a=1+1", (then hit return/enter, won't repeat this again), and then type "a" to see what the variable a contains.
4. To see what low level methods are available in the module "threading", you must first import it: "import threading" in the terminal. Then, you use "help(threading)". You can then see a list comes out. threading relies on _thread, which you can see by using "help(threading._thread)". 
5. To see what is in display, you can use "help(brick.display)" or "help(pybricks.ev3brick.display)" 

I'm currently busy with other things, so I might not be able to help you further.

AHSamir, 2019-08-23:
Hello guys. I just downloaded ev3 micropython image on my SD card. BUt where do I find all these low level scripts, you are talking about. Thank you so much for your help

AHSamir, 2019-08-23:
I am working on windows 10. ALL I got is an example file (main.py) and extensions.json , launch.json and settingjson.   Where do I access all low level scripts for motors, sensors ...

alex309-duarte, 2020-01-11:
> Thanks, it works!
> 
> ```python
> class HTColorSensor(Ev3devSensor):
>     _ev3dev_driver_name = 'ht-nxt-color-v2'
>     _number_of_values = 4
>     def _open_files(self):
>         super()._open_files()
>         self._command_file = open(self.path + 'command', 'w')
>     def _close_files(self):
>         super()._close_files()
>         self._command_file.close()
>     # configure the sensor for 50Hz power mains
>     def set_mains_50hz(self):
>         self._command_file.write('50HZ')
>         self._command_file.flush()
>     # configure the sensor for 60Hz power mains
>     def set_mains_60hz(self):
>         self._command_file.write('60HZ')
>         self._command_file.flush()
>     # returns the tuple of color values: (red, blue, green white). 0 to 255 for each
>     def rgbw(self):
>         self._mode('RGB')
>         errors = 0
>         while True:
>             try:
>                 return self._value(0), self._value(1), self._value(2), self._value(3)
>             except:
>                 errors += 1 # may give ENODEV error
>                 # returns tuple of zeros if there are two consecutive errors
>                 if errors > 1:
>                     return 0, 0, 0, 0 
>     # returns the color index. 0 to 17
>     def color(self):
>         self._mode('COLOR')
>         errors = 0
>         while True:
>             try:
>                 return self._value(0)
>             except:
>                 errors += 1 # may give ENODEV error
>                 # returns zero if there are two consecutive errors
>                 if errors > 1:
>                     return 0
> ```
> 
> My country's mains is 50hz. i did a test by pointing the sensor onto a brown table top for about a minute, 8cm from the surface. The rgbw values for 50hz varies by about one throughout the entire time (white is mostly 29, sometimes 30), and is quite steady. Setting it to 60hz causes them to fluctuate quite obviously (white fluctuates from 27 to 31).
> 
> **On a different note for anyone looking at this**: while it may be able to return values for every 10 milliseconds or so, it usually takes about 30 milliseconds or so for the values to update. HT sensor is usually slower than the EV3 color sensor, but offers a better range.

hello, I need to use the sensor, but i don't know where i need to put this code, do you use visual studio code ? o which IDE do you use for programing in micropython ? i hope you can you help me could you send me screenshot with the steps? thanks

LaiYanKai, 2020-01-12:
@alex309-duarte Visual Studio Code. The steps are here:
https://le-www-live-s.legocdn.com/sc/media/files/ev3-micropython/ev3micropythonv100-71d3f28c59a1e766e92a59ff8500818e.pdf