ev3dev/ev3dev #1293: mindsensors.com 8-channel Servo Controller - define wider min and max limit position

cho934, 2019-07-05:
**What are you trying to do?**

I am trying to configure some servomotors which have wider min or max limit position 

**What did you expect to happen**

The program should have larger setting for set_min_pulse_sp and set_max_pulse_sp

**What actually happened**

The program uses default ranges that cannot be changed. 
// Min Pulse SP: read/write
  // Used to set the pulse size in milliseconds for the signal that tells the
  // servo to drive to the miniumum (counter-clockwise) position_sp. Default value
  // is 600. Valid values are 300 to 700.

// Used to set the pulse size in milliseconds for the signal that tells the
  // servo to drive to the maximum (clockwise) position_sp. Default value is 2400.
  // Valid values are 2300 to 2700. You must write to the position_sp attribute for
  // changes to this attribute to take effect.

```
servo_motors_[servo].set_min_pulse_sp(pulse);
```
with pulse defined depends on the servomotor.

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2019-03-03.zip

I am trying to use servomotors at there extreme limit position because some of them can reach the position 100 or 200.

dlech, 2019-07-05:
The kernel driver will have to be changed. e.g. https://github.com/ev3dev/lego-linux-drivers/blob/5f2f1c686d887c16ad52678d3796bb08a19aa55c/motors/servo_motor_class.c#L232

cho934, 2019-07-06:
Yes. In that case I will propose new wider ranges to be more flexible:

min:
value < 1000  & value > 0
medium:
value < 2000 & value > 1000
max:
value < 3000 & value > 2000

cho934, 2019-07-25:
Hi, 
Is it possible to integrate this fix in the next kernel release ?

dlech, 2019-07-25:
Probably. Would you like to make a pull request?

dlech, 2020-03-07:
Fix released in kernel ev3dev-2.3.5

closed 2020-03-07 by dlech