ev3dev/ev3dev #1294: loading 3rd party library

racheliurui, 2019-07-06:
I have tried to load 3rd party lib to implement MQTT message publishing while running the EV3 Scripts.

```python
import sys
sys.path.append('/usr/local/lib/python2.7/dist-packages/MyLib')
import MyLib

def main():
   print( sys.modules['MyLib'])

if __name__ == '__main__':
    main()
```

Above code works fine if I run from command line with 

```shell
python main.py
```

But when I try to run from brickrun, I got below exception,

```
Starting: brickrun --directory="/home/robot/vscode-hello-python" "/home/robot/vscode-hello-python/main.py"
Started.
----------
Traceback (most recent call last):
  File "/home/robot/vscode-hello-python/main.py", line 7, in <module>
    import MyLib
ImportError: No module named 'MyLib'
----------
Exited with error code 1.
```

Could you give some clue how to properly load 3rd party Python Lib while running the robot?

I have a feeling the brickrun will have its own specified env, but not sure where I can specify extra lib to implement some customized logic.

Thanks & Cheers

racheliurui, 2019-07-08:
I figured out a workaround by manually copy / install the lib in same folder with the code. This way the lib can be loaded locally by the script.
Hopefully this can help others to get a clue how to solve this. But still hopefully there can be a section in help document on how to work with 3rd party lib.

Thanks

dlech, 2019-07-08:
My guess is that the problem has to do with python2 vs python3. Also, we highly recommend using debian packages for python whenever possible. Ev3dev already ships with a python3 MQTT library. Could you use it instead?

racheliurui, 2019-07-10:
@dlech  Thanks for the reply. I found the document at,
https://www.ev3dev.org/docs/tutorials/sending-and-receiving-messages-with-mqtt/

I am using EV3 to play with AWS Greengrass device SDK. I am not sure if the build-in MQTT lib compatible with AWS Greengrass (although both support MQTT), as I see they added some security feature in it (public key and private key as parameters)

I will later try the build in MQTT library.

dlech, 2019-07-10:
If you use `python` as in:

https://docs.aws.amazon.com/greengrass/latest/developerguide/IoT-SDK.html

Then it will install the Python 2 version. But the ev3dev libraries are Python 3, so you should replace `python` with `python3` and hopefully it will work.