ev3dev/ev3dev #1295: Installing numpy on EV3 w/python thru pip3 BUT the process is very slow

uwieske, 2019-07-08:
I am trying to install NumPy on my EV3 but the process is VERY slow.
sudo pip3 install numpy

I had to install pip3 (pip for python3) first and the pip3 installation went well. I used the following lines:
sudo apt-get update
sudo apt-get install python3-pip 

But when want to install numpy it takes an eternity.
Expected this process to go much quicker. I have been waiting for 45 minutes.
Then I interrupted the pip3 process with Ctrl-C. (needed to repeat this action one or three times consecutively. (I am not sure now if my EV3DEV is in a well defined state (corrupted) at this point. I hope there was some rollback action by the script. I checked with pip3 list, but no signs of an numpy installation. 
I tried a second time (after having overclocked my EV3)... hoping I could gain some speed... but now I keep on getting a "killed" message from pip3 after having tried to "collecting numpy".

I have a EV3DEV streched 4.14.117-ev3dev-2.3.4-ev3
overclocked at=456

Need some help please.

dlech, 2019-07-08:
This is why we don't bother including `pip3`. There are debian packages for many popular python packages.

```
sudo apt update
sudo apt install python3-numpy
```

uwieske, 2019-07-08:
Hi David,

Awesome! Thanks a lot. Those two lines helped me move forward.

closed 2019-07-08 by uwieske