ev3dev/ev3dev #1296: send messages from ev3 to ev3 via Bluetooth

NikoNowak, 2019-07-11:
Has anyone here managed to make two EV3 bricks send messages to each other via Bluetooth (Java)? What I am trying to do is, when one EV3 car gets close to the other, it sends a message to the other. I have read #459, which gives no satisfactory answer, and it is from 2015.

I have also tried the [J4EV3](https://github.com/LLeddy/J4EV3) library, which failed because of bluecove libraries that are not found during the execution on the EV3. Since bluecove wasn't updated since I think 11 years, it might not be the best solution.

The messages MUST be sent over Bluetooth.

Any ideas or help would be very much appreciated!

**Robot (please complete the following information):**
 - Device: EV3 
 - Versions: Debian Stretch ev3dev-stretch-ev3-generic-2019-03-03 with ev3dev-lang-java
 - Kernel: 4.14.117-ev3dev-2.3.4-ev3
 - Bluetooth: 2.1
```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

wizard1066, 2019-07-11:
You could also use IR receivers/transmitters maybe.  Sure it isn't ideal, but put car A on channel 1 and car B on channel 2 and get their receivers to look out for each other. You get a distance and even an angle if their looking at each other!
Talking bluetooth Anton Vanhoucke has published a few excellent articles on getting a Sony PS3 and PS4 controller working. I am thinking out load, but there might be something on his Micropython solution on PS4 connectivity that you can use. Look at what he is doing here, directly interpreting bytes from a device file. https://antonsmindstorms.com/2019/06/16/how-to-use-a-ps3-gamepad-with-micropython-on-the-ev3-brick/

NikoNowak, 2019-07-12:
Thanks for the answer, wizard1066. The communication needs to happen via Bluetooth though, it is a requirement. By the way, can one even run Python code on ev3dev-lang-java?

dlech, 2019-07-12:
You could use the (deprecated) `rfcomm` command line tool to set up a Bluetooth serial connection between the EV3s, then just use it like a serial port in Java.