ev3dev/ev3dev #1297: Trying to connect ev3dev to internet via bluetooth with Ubuntu 18.04, but state is only "connected" instead of "online"

Gls-Facom, 2019-07-13:
**What are you trying to do?**
  Connect EV3 to internet via bluetooth on ubuntu 18.04
**What did you expect to happen?**
  Successfully connect to internet and get status: "online", and can access through SSH
**What actually happened?**
  No connection to internet, status: "connected"

**What hardware and software are you using (including version numbers)?**
  Kernel: 4.14.96-ev3dev-2.3.2-ev3
  Model: Lego mindstorms EV3
  Fresh install of ev3dev-stretch-ev3-generic-2019-03-03

  Host PC: Ubuntu 18.04.2 LTS 64bit

dlech, 2019-07-13:
Can you please provide the exact steps to reproduce the problem?

Gls-Facom, 2019-07-15:
I followed  all the steps on the tutorial on ev3dev page, that can be found here: https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/.

jxb4137, 2019-08-17:
I also met the same problem. How did you solve it?

dk88711, 2020-10-19:
Just encountered the problem and then solved it. Me on linux mint 20.
What I have done: check if status is "online" when connecting to an Android device bluetooth pan => Yes. so it is pc problem. Then I go Advanced Network Configuration -> pan1 edit -> IPv4 Settings -> Change Method to Shared to other computers (this step very similar to the connecting internet via usb tutorial) -> then viola! It sudden works! (able to ping 8.8.8.8!). It will be better if this step is also included in the via bluetooth tutorial I think.