ev3dev/ev3dev #1298: lego-linux-drivers-dkms on raspbian (raspberry pi 3)

JustACodingFox, 2019-07-23:
**What are you trying to do?**
I am trying to connect the EV3 Ultrasonic Sensor to the Raspberry Pi using the lego-linux-drivers-dkms package.

**What did you expect to happen**
The program should have made it possible to attach the "LEGOEV3 line discipline" to a serial port. So that I can use the lego-sensor class in sysfs to communicate with the sensor.

**What actually happened**
I tried to install the dkms tarball as described in the instructions for other linux and created the .rules file. But if try to use "sudo ldattach 29 /dev/ttyUSB0" I get following error:
sudo ldattach 29 /dev/ttyUSB0.
Following the instruction for installing on debian doesn't work either because I get following error:
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/stretch.

**Output Ldattach**
Here is the output of ldattach in debug mode:
ldattach: opened /dev/ttyS0
ldattach: set to raw 0 ---: cflag=0xcbb
ldattach: cannot set line discipline: Invalid argument

**Hardware**
 - Raspberry Pi 3 Model B Rev 1.2
 - EV3 UltraSonic Sensor
 - Connection Type: Serial Connection with GIPOs 14 and 15

**What I tried**
To fix my issue I tried building ev3dev rules from source and then installing the following package: lego-modules_7.0_all.deb. But I am unable to build ev3dev rules from source due to my limited knowledge and absence of a makefile.
To check if my installation was correct, I opened the following file, /proc/tty/ldiscs, to see if the line discipline was added there but it was not. I tried reinstalling but that did not help either.

dlech, 2019-07-23:
The dkms package is very out of date. I quit maintaining it quite a while ago because no one seemed to be using it. It probably won't build against a newer kernel.

JustACodingFox, 2019-07-23:
Thanks for the fast answer. Do you know an alternative solution for my problem or do I have to give up the approach of using the ev3dev line discipline to communicate with the sensor?

dlech, 2019-07-23:
It depends on how much time you want to spend on it. You could install the full ev3dev OS on your raspberry pi. Or you could try to update the dkms package to use the latest drivers. Or you could write your own program that implements the full protocol instead of relying on a kernel driver to do it for you.

JustACodingFox, 2019-07-24:
Thanks. I will try implementing the protocol then.Because I have no clue how to update the dkms package.

closed 2019-07-24 by JustACodingFox