ev3dev/ev3dev #1299: Best Practices for using IR Sensor and ev3dev2?

mwheinz, 2019-07-23:
**Question**

So, I've been experimenting with the ev3dev2 platform, I'm having a ball, it's great, but my attempt at making an IR-beacon-chasing robot has run into a serious issue - the data coming from the IR sensor seems very noisy and, after 20-30 seconds seems to switch to heading 0, distance 100, even if the beacon is turned off altogether or physically poking the sensor.

Any tips for getting better results or is this a basic limitation of the sensor?

**Desktop (please complete the following information):**
 - OS and version: MacOS Mojave
 - Connection Type: Bluetooth
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: (need to check when I get home)

```
**Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!** (need to check when I get home)
```

dlech, 2019-08-13:
Can you share your program to reproduce the problem?

mwheinz, 2019-08-13:
This is my test program:

> #!/usr/bin/env python3
> 
> import os
> import sys
> from ev3dev2.sensor.lego import InfraredSensor, TouchSensor
> from ev3dev2.sensor import INPUT_1, INPUT_4
> from ev3dev2.sound import Sound
> from time import sleep
> 
> def debug_print(*args, **kwargs):
>     '''Print debug messages to stderr.
> 
>     This shows up in the output panel in VS Code.
>     '''
>     print(*args, **kwargs, file=sys.stderr)
> 
> sound = Sound()
> 
> sound.speak("Beginning test")
> 
> touch = TouchSensor(INPUT_1)
> ir = InfraredSensor(INPUT_4)
> ir.mode = ir.MODE_IR_PROX
> 
> sound.speak("Proxmity Test")
> sound.speak("Hold down the touch sensor to stop")
> while not touch.is_pressed:
>     sound.speak("IR Proximity is {}".format(ir.proximity))
> sound.speak("Proximity test completed")
> touch.wait_for_released()
> 
> sound.speak("I R Remote Button Test")
> sound.speak("Set remote to channel 1")
> ir.on_channel1_top_left = lambda state : sound.speak("Top Left state {}".format(state))
> ir.on_channel1_bottom_left = lambda state : sound.speak("Bottom Left state {}".format(state))
> ir.on_channel1_top_right = lambda state : sound.speak("Top Right state {}".format(state))
> ir.on_channel1_bottom_right = lambda state : sound.speak("Bottom Right state {}".format(state))
> sound.speak("Hold down the touch sensor to stop")
> while not touch.is_pressed:
>     ir.process()
> 
> sound.speak("Remote button test completed")
> touch.wait_for_released()
> 
> ir.mode = ir.MODE_IR_SEEK
> sound.speak("IR Seeker Mode Test")
> sound.speak("Place remote in beacon mode")
> while not ir.beacon():
>     sleep(0.25)
> 
> sound.speak("Beginning test")
> sound.speak("Turn off beacon or press touch sensor to stop")
> while ir.beacon() and not touch.is_pressed:
>     msg = "Heading is {}, Proximity is {}, Beacon is {}".format(ir.heading(),ir.distance(), ir.beacon())
>     debug_print(msg)
>     #sound.speak(msg)
> 
> sound.speak("Test complete")

This is the output from a typical run. During this test the IR beacon was never more than 6" from the sensor while it was running and, for the last 5 seconds, the beacon was turned off but ir.beacon() continued to return true. (Apologies, I have no idea why the text is appearing as extra-large and bold...)

> Starting: brickrun --directory="/home/robot/Tank/../TestTools" "/home/robot/Tank/../TestTools/ir.py"
> Started.
> ----------
> Heading is -2, Proximity is 3, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
[... 247 lines cut ...]
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> Heading is 0, Proximity is 100, Beacon is True
> ----------
> Completed successfully.