ev3dev/ev3dev #1302: A way to boot into EV3's main OS without removing SD Card

bean07, 2019-08-06:
**Is your feature request related to a problem? Please describe.**
I've always been fustrated when removing my SD Card.

**Describe the solution you'd like**
I would like a way to boot into EV3's main Operating System without removing my SD Card

**Describe alternatives you've considered**
Removing SD Card

dlech, 2019-08-06:
For anyone interested in taking this on, we would probably need to modify [U-Boot](https://github.com/ev3dev/u-boot) to read one of the buttons via GPIO and skip booting from the SD card if the button is pressed.

Most buttons already have special features (back and center are reset, right activates the firmware flasher, left is not operable in early boot because of default internal pulldown), so that leaves us with either the up or down button.

BertLindeman, 2019-08-06:
Maybe a simple "hardware" tweak: add some adhesive tape to the SD-card. It makes it so much easier to remove the sd-card.
Something like this (Though this blue tape is not so good...)

[image]