ev3dev/ev3dev #1306: Ev3 Python Visual Studio Code Start time

JohnHZheng, 2019-08-24:
I am starting to use Python3 for ev3 development via Visual Studio Code. I am using EV3DEV Device Browser and when I right click a py file in EV3 and click "Run", the ev3 light flashed about 10 seconds and actually run. 
If I run it from EV3 brickman file manager, it will run right away. 

How can I shorten the start time in Visual Studio Code - EV3DEV Device Browser - Run? 

Thanks

dlech, 2019-08-25:
VS Code must be downloading some files that take 10 seconds to send to the EV3 brick. There are some options in the VS Code extension that can be used to exclude files from being downloaded. You could use this to exclude any unnecessary files.

JohnHZheng, 2019-08-25:
It is not the debug and run from VS Code, where all files got downloaded. 

I downloaded all files first (where I only have a few). Then I select the remove Py file from VSCode device browser, right click, and then choose "Run" from the list. The brick blinked 10 times (around 10 seconds), then the code start to run. So there are no files downloaded between the time I click Run and the code is actually running.  

And if I use Brick File Browser directly and navigate the PY and click the center button, the code will run immediately.

WasabiFan, 2019-08-25:
Can you give some details about your code? Is it using normal `python3`, `micropython` with ev3dev-lang-python, or `micropython` with pybricks? If the first one, try [using micropython](https://python-ev3dev.readthedocs.io/en/ev3dev-stretch/micropython.html). If it's already using micropython, I'm not sure what is wrong.

JohnHZheng, 2019-08-25:
I am using Python3. So I changed to microPython SheBang, and now I got error from library. 
Started.
----------
Traceback (most recent call last):
  File "/home/robot/JohnZ/testGameLine.py", line 4, in <module>
  File "ev3dev2/motor.py", line 43, in <module>
ImportError: no module named 'ev3dev2.stopwatch'

WasabiFan, 2019-08-25:
Sounds like you're on a broken version of the library ("beta 4"); try updating:

```
sudo apt-get update
sudo apt-get install --only-upgrade python3-ev3dev2 micropython-ev3dev2
```

Then run ` dpkg-query -l {python3,micropython}-ev3dev*` and confirm you're now running `2.0.0~beta5` for both the `micropython` and non-micropython versions:

```
robot@ev3dev:~$ dpkg-query -l {python3,micropython}-ev3dev*                                                                                                                                                              Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                             Version                       Architecture                  Description
+++-================================================-=============================-=============================-=====================================================================================================
ii  micropython-ev3dev2                              2.0.0~beta5                   all                           Python language bindings for ev3dev for MicroPython
ii  python3-ev3dev                                   1.2.0                         all                           Python language bindings for ev3dev
ii  python3-ev3dev2                                  2.0.0~beta5                   all                           Python language bindings for ev3dev
```

wizard1066, 2019-08-27:
I downloaded MicroPython from the LEGO website and it was this.

robot@fox:~/robot/MicroPython$ sudo dpkg-query -l {python3,micropython}-ev3dev*  
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                Version                Architecture           Description
+++-===================================-======================-======================-===========================================================================
ii  micropython-ev3dev2                 2.0.0~beta3            all                    Python language bindings for ev3dev for MicroPython
ii  python3-ev3dev                      1.2.0                  all                    Python language bindings for ev3dev
ii  python3-ev3dev2                     2.0.0~beta3            all                    Python language bindings for ev3dev

Just updated it!! THANKS super useful POST!!