ev3dev/ev3dev #1310: mindsensors NXTSumoEyes-V2 Driver

lerker100, 2019-09-02:
The NXTSumoEyes-V2 sensor from mindsensors doesn't have any drivers. It shows up as a NXT touch sensor on my ev3. It's a really simple sensor, In Ev3Blocks it returns two booleans.

Noc1a, 2025-03-21:
With what language do you want to use your SumoEyes?

Noc1a, 2025-03-29:
It doesn't have any dedicated Linux driver. But this is an NXT analog sensor. Which means you can easily use it by creating an analog lego port and then instantiating a generic sensor:

```
    port = LegoPort(INPUT_1)
    sleep(0.2)
    port.mode="nxt-analog"
    sumoEyes = Sensor(INPUT_1)
    sumoEyes.value(0)
```
You need then to interpret the values.  

To make it easier, I created a Python driver for it (based on Mindsensors driver) [msSumoEyesSensor.py](https://github.com/Noc1a/ev3dev-python-drivers/blob/9dadd9bd1c4b38aecf12d2a6591705dad55dbb17/msSumoEyesSensor.py) that you can use as follow:
```
    sumoEyes = MsSumoEyesSensor(INPUT_1)
    debug_print("zone", sumoEyes.detectObstactleZone(True))
    # change range
    sumoEyes.setRange(MsSumoEyesSensor.SHORT_RANGE)
    debug_print("zone", sumoEyes.detectObstactleZone(True))
```