ev3dev/ev3dev #1311: Port Splitter for Ev3 motors

DigitalRookie4, 2019-09-02:
I have searched all over the internet for ev3 port splitters for motors. I have got 4 large motors and 2 medium motors and I do not have enough motor ports for them all. I can usually only find sensor port splitters which is not what I am looking for. I am looking for a **MOTOR** one. I have found one but I am not 100% sure if it is what I am looking for or not. Would anybody out there be able to tell me if I would be able to [(http://www.mindsensors.com/ev3-and-nxt/52-port-splitter-for-nxt-digital-sensors) ] I am also not going to buy [http://www.mindsensors.com/ev3-and-nxt/21-multiplexer-for-nxtev3-motors] because it is too expensive. If the one I mentioned above which might be a sensor one is not right can somebody out there give me a suggestion on a cheap and efficient port splitter for lego ev3 motors.

dlech, 2019-09-02:
port-splitter-for-nxt-digital-sensors is only for I2C sensors. It won't work for motors. multiplexer-for-nxtev3-motors is the only one I know of for motors.

DigitalRookie4, 2019-09-03:
@dlech  But aren't they very expensive (and I need 2) so thats about £110!

DigitalRookie4, 2019-09-03:
@dlech  There is something I have seen on the first lego link ((http://www.mindsensors.com/ev3-and-nxt/52-port-splitter-for-nxt-digital-sensors) that I saw. it mentioned this in the documentation guide:
> Using SPLIT-Nx-v2 with NXT Motors
Upto three motors can be attached to SPLIT-Nx-v2. Turning motor port ON
through the NXT program will turn all the attached motors ON at once.
This is useful if you need to run all motors at once, however independent
control of each attached motor is not possible and encoder values of the
motors can not be used. i.e. rotation control feature of motor blocks can not
be used. Also note that the power supplied to the motors will be limited by
the power coming from the NXT brick.
More about the encoder value: The encoder reading on NXT will be Logical
AND operation of encoder values of all attached motors. For simple
programs such value is not of much use.

dlech, 2019-09-03:
This is the important part:

> however independent
> control of each attached motor is not possible and encoder values of the
> motors can not be used.

It all depends on what you want to do with the motors.

DigitalRookie4, 2019-09-04:
@dlech well thats fine then because I would use these so that each side of the tank would move with 2 motors at the same time. 3 things I want to know first before I buy this:

1. How would I define the motors and makes them rotate in my code?
2. How certain are that this adaptor would work?
3. Do you think it is safe to use?
4. So you are saying that the port-splitter-for-nxt-digital-sensors ((http://www.mindsensors.com/ev3-and-nxt/52-port-splitter-for-nxt-digital-sensors)) will work with the motors

DigitalRookie4, 2019-09-06:
@dlech well thats fine then because I would use these so that each side of the tank would move with 2 motors at the same time. 3 things I want to know first before I buy this:

How would I define the motors and makes them rotate in my code?
How certain are that this adaptor would work?
Do you think it is safe to use?
So you are saying that the port-splitter-for-nxt-digital-sensors ((http://www.mindsensors.com/ev3-and-nxt/52-port-splitter-for-nxt-digital-sensors)) will work with the motors

DigitalRookie4, 2019-09-08:
@dlech  any suggestions?

dlech, 2019-09-08:
> How would I define the motors and makes them rotate in my code?

With the DCMotor class https://ev3dev-lang-python-1.readthedocs.io/en/restruct-docs/motors.html#dc-motor

> How certain are that this adaptor would work?

I am confident it will work with the limitations that have been described.

> Do you think it is safe to use?

There are safety precautions in place that should prevent overloading the batteries.

bmegli, 2019-09-08:
@DigitalRookie4 

If I understand correctly:
- 2 motors on the left side
- 2 motors on the right side

There is one more detail to consider. Every motor is a bit different. 
When controlling as DC motor the speed at duty cycle may vary a bit by motor (even identical).
This may or may not be a problem but it's good to know beforehand.

Also when running without speed control for left/right side the robot is less likely to move exactly straight ahead (turning a bit to some side). Again, this may or may not be a problem and may be corrected to some extent in the program running the robot.

DigitalRookie4, 2019-09-14:
@bmegli Your understanding of what I want to do is exactly correct. I will take you advice. Thank you.