ev3dev/ev3dev #1312: Connecting 2 ev3 bricks with bluetooth using latest R2 release

ThadHouse, 2019-09-17:
Trying out the new R2 snapshot release attempting to connect 2 EV3s together, both running the same OS, and it's not working. They see each other and connect, however they immediately disconnect after about 5 seconds. Using `bluetoothctl`, after connecting, it connects, and then gives `org.bluez.Error.NotAvailable`, and then disconnects about 5 seconds later. I can't find anything in any error logs explaining whats going wrong. 

I need the EV3's talking over either Bluetooth serial or Bluetooth sockets, but it does need to be bluetooth.

dlech, 2019-09-17:
"connect" only connects to existing services. There is not a serial port service by default. Have you tried writing a program using Bluetooth sockets? If the EV3s are paired, they should connect automatically when your program runs.

ThadHouse, 2019-09-17:
Oh ok. Any recommended guide on setting up bluetooth sockets? I've never done it before, only ever done serial. But sockets definitely seem like a better idea.

dlech, 2019-09-17:
Unfortunately, there isn't a python3 version of pybluez available in Stretch (there is in Buster though).

But it is still possible to use raw sockets without installing extra software. e.g. https://stackoverflow.com/questions/15828916/bluetooth-server-with-python-3-3

There are also some command line tools that can set up a Bluetooth serial port, but it has been a long time since I used them.

ThadHouse, 2019-09-17:
I'm using C++, I think I found a guide to get bluetooth sockets setup. I'll see if it works, and if so I'll post the results here for future users.

dlech, 2019-09-17:
Great. I think I used this as a reference in the past: https://people.csail.mit.edu/albert/bluez-intro/x502.html

dlech, 2020-04-10:
FYI, ev3dev-stretch R3 has Pybricks 2.0 which includes an easy to use class for creating Bluetooth connections.