ev3dev/ev3dev #1313: Python script needed to run dGPS sensor on EV3 mindstorms.

naynt, 2019-09-23:
**Is your feature request related to a problem? 
No, I wish to obtain or create a python script for the dGPS on the Mindstorm EV3.

**Describe the solution you'd like**
Is it possible to run the following obtained script: [https://github.com/DexterInd/BrickPi3/blob/master/Software/Python/Examples/DI-dGPS.py]
from Dexter Industries created for the BrickPi3 but for EV3 Mindstorms EV3 instead?

**Additional context**
In other words, I was wondering whether anyone has successfully ran a micropython script on the EV3 to get the dGPS to return satellite coordinates.

wizard1066, 2019-09-23:
I am going to write an app to pull information from a iPhone for using in a script, could be GPS co-ordinates.  Reply to this thread if you can spare the time to be a beta tester.

dlech, 2019-09-23:
The Dexter code can't be used directly. However, the constant values are still useful.

To make it work on ev3dev, the following needs to be done:
- Put the port in `other-i2c` mode.
- Use the Python SMBus library to get a direct I2C connection
- Translate BP.transact_i2c() to SMBus methods