ev3dev/ev3dev #1314: Bluetooth still ocasionally not available after reboot, with R2 Release Candidata

JunOllyLi, 2019-09-25:


philltran, 2019-10-02:
I am experiencing the same on R2,. Is rebooting the only work around?

dlech, 2019-10-02:
It depends on where the failure is. One time, I was able to run `sudo systemctl restart bluetooth` to get it working with out reboot, but another time this didn't work.

We could also try experimenting with the sleep time to see if making it longer makes a difference.

    sudo nano /lib/systemd/system/ev3-bluetooth.service

And change the line:

    ExecStartPre=/bin/sleep 10

philltran, 2019-10-02:
Thanks for the tip. Let me know if there is anything I can do to help with the project.

dlech, 2019-10-02:
Try 20 or 30 seconds and report back the results :smile:

philltran, 2019-10-02:
Of course now it's working after I re-imaged the SD card. 

I changed the sleep time and everything seems to still be working. 

I will try it again if the bluetooth goes out again. 

Thanks for the help.

keenanjohnson, 2019-10-28:
I'm having this issue on my ev3 as well.

keenanjohnson, 2019-10-28:
Updating the sleep time to 30 did not resolve the issue for me.

keenanjohnson, 2019-10-28:
Interestingly reimaging the SD card has also resolved this for me. Seems like currently the image process must be done twice?

dlech, 2019-11-07:
Replying to @janderll https://github.com/ev3dev/ev3dev/issues/925#issuecomment-550994759 and https://github.com/ev3dev/ev3dev/issues/925#issuecomment-551030963

I've seen some other people mention that installing/fiddling with `rfkill` seems to make a difference. Theoretically, `connmand` is supposed to take care of this and installing `rfkill` should not be necessary.

Could everyone who is still having problems share what kind of SD card they are using? I am starting to wonder if cheap (and therefore slow) microSD cards are affecting the timing compared to the SanDisk Extreme cards that I am using.

JanderII, 2019-11-07:
I'm using a SanDisk Ultra Class 10 A1 32GB.

It is certainly bizarre that after lots of failures that it's now consistently working. That does seem to imply that something has changed on the SD card now.

keenanjohnson, 2019-11-07:
Yeah I'm also using a fairly high quality SD card. A SanDisk Ultra Class 10.

https://www.amazon.com/SanDisk-microSDHC-Standard-Packaging-SDSQUNC-032G-GN6MA/dp/B010Q57T02

neilmcguire, 2019-11-09:
I was having Bluetooth Not Available issues consistently, then made sure I could connect to the BT in Lego OS so I knew it wasn't a hardware issue.

I am using a Kingston 32 GB Class 10. We followed the `rfkill` suggestions over on ev3dev/ev3dev#925 and it made the Bluetooth available. Because the reflash has fixed it for some people I am trying that right now, and will report back. And if it doesn't work to reflash, at least I have the `rfkill` fiddling I can do.

Here's what it looked like:
```
robot@ev3dev:~$ ev3dev-sysinfo
Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534C6247
BOARD_INFO_TYPE=main
robot@ev3dev:~$ sudo apt-get update && sudo apt-get install rfkill
robot@ev3dev:~$ /usr/bin/hciattach -t 30 /dev/ttyS2 any 2000000 flow nosleep
robot@ev3dev:~$ sudo rfkill list all
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: hci0: Bluetooth
        Soft blocked: yes
        Hard blocked: no
robot@ev3dev:~$ sudo rfkill unblock 2
robot@ev3dev:~$ sudo rfkill list all
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
robot@ev3dev:~$ sudo reboot
```

salamanders, 2019-11-23:
+1 to this issue.
micropython ISO from the Lego site, ran `sudo apt-get dist-upgrade`
Booted with nothing plugged in: no bluetooth.
Booted with just the USB/WiFi dongle in: still no bluetooth.

```
Linux ev3dev 4.14.117-ev3dev-2.3.4-ev3 #1 PREEMPT Thu May 9 15:13:01 CDT 2019 armv5tejl
robot@ev3dev:~$ ev3dev-sysinfo
Image file:         ev3-micropython-v1.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165353FA38
BOARD_INFO_TYPE=main

dlech, 2019-11-23:
@salamanders What do the logs say? `sudo systemctl --no-pager status ev3-bluetooth` ?

CubixSystem, 2019-11-24:
@dlech I think my logs may help

```
robot@ev3dev:~$ ev3dev-sysinfo
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           226.09
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534CDB22
BOARD_INFO_TYPE=main
robot@ev3dev:~$ sudo systemctl --no-pager status ev3-bluetooth
[sudo] password for robot:
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-11-24 19:58:20 UTC; 2min 29s ago
  Process: 438 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)
  Process: 333 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)

Nov 24 19:57:09 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Nov 24 19:57:20 ev3dev hciattach[438]: Found a Texas Instruments' chip!
Nov 24 19:57:20 ev3dev hciattach[438]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Nov 24 19:57:20 ev3dev hciattach[438]: Loaded BTS script version 1
Nov 24 19:57:20 ev3dev hciattach[438]: texas: changing baud rate to 2000000, flow control to 1
Nov 24 19:57:50 ev3dev hciattach[438]: Initialization timed out.
Nov 24 19:58:20 ev3dev systemd[1]: ev3-bluetooth.service: Control process exited, code=exited status=1
Nov 24 19:58:20 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Nov 24 19:58:20 ev3dev systemd[1]: ev3-bluetooth.service: Unit entered failed state.
Nov 24 19:58:20 ev3dev systemd[1]: ev3-bluetooth.service: Failed with result 'exit-code'.
```

dlech, 2019-11-24:
> hciattach[438]: Initialization timed out.

This is the old error from #925, so adjusting the delay as describe in https://github.com/ev3dev/ev3dev/issues/1314#issuecomment-537621073 may help in this case.

dlech, 2019-11-24:
In this issue, we are expecting a different error, such as:

    hciattach[461]: Can't init device hci0: Operation not possible due to RF-kill (132)Can't initialize device: Operation not possible due to RF-kill

CubixSystem, 2019-11-24:
@dlech thanks for pointing this out, tomorrow I try `ev3dev-stretch R2` and report results

I'm pretty new here and thought that run [Option 2: Upgrading all installed software](https://www.ev3dev.org/docs/tutorials/upgrading-ev3dev/) should be enough to be up to date

dlech, 2019-11-24:
> I'm pretty new here and thought that run [Option 2: Upgrading all installed software](https://www.ev3dev.org/docs/tutorials/upgrading-ev3dev/) should be enough to be up to date

Hopefully that is true :smile: but since there have been reports here of re-flashing the SD card making a difference, it is hard to say for sure what else could be going on.

CubixSystem, 2019-11-25:
@dlech I use `ev3dev-stretch R2` and here is the result

1. After first time boot with new image Bluetooth was available in Brickman menu

2. After restart problem appear again
```
Oct 23 16:48:29 ev3dev hciattach[445]: Can't init device hci0: Operation not possible due to RF-kill (132)Can't initialize device: Operation not possible due to RF-kill
```

Full log:
```
robot@ev3dev:~$ ev3dev-sysinfo
Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           226.09
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534CDB22
BOARD_INFO_TYPE=main
robot@ev3dev:~$ sudo systemctl --no-pager status ev3-bluetooth -l
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-10-23 16:48:29 UTC; 4min 18s ago
  Process: 445 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)
  Process: 332 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)

Oct 23 16:48:17 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Oct 23 16:48:28 ev3dev hciattach[445]: Found a Texas Instruments' chip!
Oct 23 16:48:28 ev3dev hciattach[445]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Oct 23 16:48:28 ev3dev hciattach[445]: Loaded BTS script version 1
Oct 23 16:48:28 ev3dev hciattach[445]: texas: changing baud rate to 2000000, flow control to 1
Oct 23 16:48:29 ev3dev hciattach[445]: Can't init device hci0: Operation not possible due to RF-kill (132)Can't initialize device: Operation not possible due to RF-kill
Oct 23 16:48:29 ev3dev systemd[1]: ev3-bluetooth.service: Control process exited, code=exited status=1
Oct 23 16:48:29 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Oct 23 16:48:29 ev3dev systemd[1]: ev3-bluetooth.service: Unit entered failed state.
Oct 23 16:48:29 ev3dev systemd[1]: ev3-bluetooth.service: Failed with result 'exit-code'.
```

3. After next restarts error in logs changed
```
robot@ev3dev:~$ sudo systemctl --no-pager status ev3-bluetooth -l
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-11-25 22:40:49 UTC; 4min 2s ago
  Process: 439 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)
  Process: 327 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)

Nov 25 22:39:38 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Nov 25 22:39:48 ev3dev hciattach[439]: Found a Texas Instruments' chip!
Nov 25 22:39:48 ev3dev hciattach[439]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Nov 25 22:39:48 ev3dev hciattach[439]: Loaded BTS script version 1
Nov 25 22:39:48 ev3dev hciattach[439]: texas: changing baud rate to 2000000, flow control to 1
Nov 25 22:40:18 ev3dev hciattach[439]: Initialization timed out.
Nov 25 22:40:49 ev3dev systemd[1]: ev3-bluetooth.service: Control process exited, code=exited status=1
Nov 25 22:40:49 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Nov 25 22:40:49 ev3dev systemd[1]: ev3-bluetooth.service: Unit entered failed state.
Nov 25 22:40:49 ev3dev systemd[1]: ev3-bluetooth.service: Failed with result 'exit-code'.
```

dlech, 2019-11-25:
Thanks for the detailed info. We know that `Initialization timed out.` is a timing issue and can possibly be improved by adjusting the delay.

If we can get the `Can't init device hci0: Operation not possible due to RF-kill` error again, please run `sudo journalctl --no-pager --boot=0` so we can see when Bluetooth was trying to start relative to other services.

CubixSystem, 2019-11-26:
Catch it <details>
  <summary>Logs</summary>

  ```sh
robot@ev3dev:~$ sudo systemctl --no-pager status ev3-bluetooth -l
[sudo] password for robot:
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-11-25 23:12:53 UTC; 1min 53s ago
  Process: 436 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)
  Process: 359 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)

Nov 25 23:12:42 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Nov 25 23:12:52 ev3dev hciattach[436]: Found a Texas Instruments' chip!
Nov 25 23:12:52 ev3dev hciattach[436]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Nov 25 23:12:52 ev3dev hciattach[436]: Loaded BTS script version 1
Nov 25 23:12:52 ev3dev hciattach[436]: texas: changing baud rate to 2000000, flow control to 1
Nov 25 23:12:53 ev3dev hciattach[436]: Can't init device hci0: Operation not possible due to RF-kill (132)Can't initialize device: Operation not possible due to RF-kill
Nov 25 23:12:53 ev3dev systemd[1]: ev3-bluetooth.service: Control process exited, code=exited status=1
Nov 25 23:12:54 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Nov 25 23:12:54 ev3dev systemd[1]: ev3-bluetooth.service: Unit entered failed state.
Nov 25 23:12:54 ev3dev systemd[1]: ev3-bluetooth.service: Failed with result 'exit-code'.
robot@ev3dev:~$ sudo journalctl --no-pager --boot=0
-- Logs begin at Thu 2016-11-03 17:16:45 UTC, end at Mon 2019-11-25 23:15:05 UTC. --
Nov 25 23:12:15 ev3dev kernel: Booting Linux on physical CPU 0x0
Nov 25 23:12:15 ev3dev kernel: Linux version 4.14.117-ev3dev-2.3.4-ev3 (david@freyr) (gcc version 6.4.1 20171012 (Linaro GCC 6.4-2017.11)) #1 PREEMPT Thu May 9 15:13:01 CDT 2019
Nov 25 23:12:15 ev3dev kernel: CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
Nov 25 23:12:16 ev3dev kernel: CPU: VIVT data cache, VIVT instruction cache
Nov 25 23:12:16 ev3dev kernel: OF: fdt: Machine model: LEGO MINDSTORMS EV3
Nov 25 23:12:16 ev3dev kernel: Memory policy: Data cache writeback
Nov 25 23:12:16 ev3dev kernel: Reserved memory: created DMA memory pool at 0x80000000, size 0 MiB
Nov 25 23:12:16 ev3dev kernel: OF: reserved mem: initialized node serial@80000000, compatible id shared-dma-pool
Nov 25 23:12:16 ev3dev kernel: DaVinci da850/omap-l138/am18x variant 0x1
Nov 25 23:12:16 ev3dev kernel: On node 0 totalpages: 16384
Nov 25 23:12:16 ev3dev kernel: free_area_init_node: node 0, pgdat c07081e0, node_mem_map c3f79000
Nov 25 23:12:16 ev3dev kernel:   DMA zone: 128 pages used for memmap
Nov 25 23:12:16 ev3dev kernel:   DMA zone: 0 pages reserved
Nov 25 23:12:16 ev3dev kernel:   DMA zone: 16384 pages, LIFO batch:3
Nov 25 23:12:16 ev3dev kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Nov 25 23:12:16 ev3dev kernel: pcpu-alloc: [0] 0
Nov 25 23:12:16 ev3dev kernel: Built 1 zonelists, mobility grouping on.  Total pages: 16256
Nov 25 23:12:16 ev3dev kernel: Kernel command line: console=tty1
Nov 25 23:12:16 ev3dev kernel: PID hash table entries: 256 (order: -2, 1024 bytes)
Nov 25 23:12:16 ev3dev kernel: Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Nov 25 23:12:16 ev3dev kernel: Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Nov 25 23:12:16 ev3dev kernel: Memory: 53248K/65536K available (5071K kernel code, 259K rwdata, 1660K rodata, 172K init, 158K bss, 12288K reserved, 0K cma-reserved)
Nov 25 23:12:16 ev3dev kernel: Virtual kernel memory layout:
                                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                   vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
                                   lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
                                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                                     .text : 0xc0008000 - 0xc04fc3c8   (5073 kB)
                                     .init : 0xc069d000 - 0xc06c8000   ( 172 kB)
                                     .data : 0xc06c8000 - 0xc0708ca0   ( 260 kB)
                                      .bss : 0xc070d854 - 0xc073542c   ( 159 kB)
Nov 25 23:12:16 ev3dev kernel: SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Nov 25 23:12:16 ev3dev kernel: Preemptible hierarchical RCU implementation.
Nov 25 23:12:16 ev3dev kernel:         Tasks RCU enabled.
Nov 25 23:12:16 ev3dev kernel: NR_IRQS: 325
Nov 25 23:12:16 ev3dev kernel: clocksource: timer0_1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
Nov 25 23:12:16 ev3dev kernel: sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
Nov 25 23:12:16 ev3dev kernel: Console: colour dummy device 80x30
Nov 25 23:12:16 ev3dev kernel: console [tty1] enabled
Nov 25 23:12:16 ev3dev kernel: Calibrating delay loop... 227.32 BogoMIPS (lpj=1136640)
Nov 25 23:12:16 ev3dev kernel: pid_max: default: 32768 minimum: 301
Nov 25 23:12:16 ev3dev kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Nov 25 23:12:16 ev3dev kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Nov 25 23:12:16 ev3dev kernel: CPU: Testing write buffer coherency: ok
[... cut ...]
Nov 25 23:14:48 ev3dev sudo[557]: pam_unix(sudo:session): session closed for user root
Nov 25 23:14:57 ev3dev systemd-resolved[450]: Switching to system DNS server ::1.
Nov 25 23:15:05 ev3dev sudo[567]:    robot : TTY=pts/0 ; PWD=/home/robot ; USER=root ; COMMAND=/bin/journalctl --no-pager --boot=0
Nov 25 23:15:05 ev3dev sudo[567]: pam_unix(sudo:session): session opened for user root by robot(uid=0)
Nov 25 23:15:06 ev3dev ntpd[486]: error resolving pool 3.debian.pool.ntp.org: Name or service not known (-2)
  ```
  
</details>

dlech, 2019-11-26:
Thanks again for the logs.

I don't know why I never noticed this before, but there is a `systemd-rfkill` service. We don't need this since ConnMan will take care of this already. So we should be able to do the following:

    sudo systemctl mask systemd-rfkill.service
    sudo systemctl mask systemd-rfkill.socket

This should make sure that systemd doesn't try to mess with with the Bluetooth adapter.

ylukin, 2019-12-15:
I was also seeing the same error (Operation not possible due to RF-Kill) on a brand new SD card with ev3dev-stretch-ev3-generic-2019-10-23.img. Masking systemd rfkill as suggested in previous comment worked for me. After several reboots, the brick connects to my Alexa device over bluetooth reliably now.

laurensvalk, 2019-12-19:
Seeing similar issues, so reporting logs for reference in case this is helpful:

```
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-ev3-generic-2019-10-23 *
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165351FD02
BOARD_INFO_TYPE=main

```

> If we can get the Can't init device hci0: Operation not possible due to RF-kill error again, please run sudo journalctl --no-pager --boot=0 so we can see when Bluetooth was trying to start relative to other services.
```
sudo systemctl --no-pager status ev3-bluetooth -l
[sudo] password for robot: 
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-12-14 17:53:55 UTC; 4 days ago
  Process: 470 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)
  Process: 393 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)

Dec 14 17:53:43 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Dec 14 17:53:53 ev3dev hciattach[470]: Found a Texas Instruments' chip!
Dec 14 17:53:53 ev3dev hciattach[470]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Dec 14 17:53:53 ev3dev hciattach[470]: Loaded BTS script version 1
Dec 14 17:53:53 ev3dev hciattach[470]: texas: changing baud rate to 2000000, flow control to 1
Dec 14 17:53:55 ev3dev hciattach[470]: Can't init device hci0: Operation not possible due to RF-kill (132)Can't initialize device: Operation not possible due to RF-kill
Dec 14 17:53:55 ev3dev systemd[1]: ev3-bluetooth.service: Control process exited, code=exited status=1
Dec 14 17:53:55 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Dec 14 17:53:55 ev3dev systemd[1]: ev3-bluetooth.service: Unit entered failed state.
Dec 14 17:53:55 ev3dev systemd[1]: ev3-bluetooth.service: Failed with result 'exit-code'.
```

> So we should be able to do the following:
```
sudo systemctl mask systemd-rfkill.service
sudo systemctl mask systemd-rfkill.socket
```

I will try this and report back.

mbr4477, 2019-12-21:
@laurensvalk Can confirm that 
```
sudo systemctl mask systemd-rfkill.service
sudo systemctl mask systemd-rfkill.socket
```
plus a reboot worked for me. Thanks!

laurensvalk, 2019-12-28:
Just coming back to say that Bluetooth hasn't failed a single time since :tada: Awesome work!

CubixSystem, 2019-12-28:
@dlech Big thanks for fix!
I can confirm https://github.com/ev3dev/ev3dev/issues/1314#issuecomment-568186729 too

dlech, 2020-04-10:
ev3dev-stretch R3 has been released with this fix. Please open a new issue if there are problems with the R3 release.

closed 2020-04-10 by dlech