ev3dev/ev3dev #1315: Remove ev3dev-fixed libopenmpt0 from Buster

JakubVanek, 2019-09-25:
**Describe the bug**
Currently it is impossible to install OpenCV 3.2 on Debian Buster without forcing the libopenmpt0 version to `0.4.3-1`. This is probably due to the presence of the older bugfixed version `0.2.7386~beta20.3-3+deb9u3ev3dev1` (https://github.com/ev3dev/ev3dev/issues/1204) that overrides the newer version from Debian repositories. I think that the updated upstream version should be OK (i.e. only Stretch should need the bugfix).

**To Reproduce**
```sh
sudo apt-get install libopencv3.2-java
```

**Expected behavior**
OpenCV 3.2 gets installed.

**Screenshots**
```
robot@ev3dev:~$ sudo apt install libopencv3.2-java libopencv3.2-jni libopencv-calib3d3.2  libopencv-features2d3.2 libopencv-highgui3.2 libopencv-videoio3.2 libopencv-objdetect3.2 libopencv-contrib3.2 libopencv-viz3.2 libvtk6.3 libavformat58 libopenmpt0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libavformat58 : Depends: libopenmpt0 (>= 0.3.0) but 0.2.7386~beta20.3-3+deb9u3ev3dev1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

```
robot@ev3dev:~$ apt-cache policy libopenmpt0
libopenmpt0:
  Installed: (none)
  Candidate: 0.2.7386~beta20.3-3+deb9u3ev3dev1
  Version table:
     0.4.3-1 500
        500 http://httpredir.debian.org/debian buster/main armel Packages
     0.2.7386~beta20.3-3+deb9u3ev3dev1 990
        990 http://archive.ev3dev.org/debian buster/main armel Packages
```

dlech, 2019-09-25:
Should be removed from the archive now.

JakubVanek, 2019-09-25:
Yup, `apt-cache policy` now doesn't show the older version. Thanks!
```
robot@ev3dev:~$ apt-cache policy libopenmpt0
libopenmpt0:
  Installed: 0.4.3-1
  Candidate: 0.4.3-1
  Version table:
 *** 0.4.3-1 500
        500 http://httpredir.debian.org/debian buster/main armel Packages
        100 /var/lib/dpkg/status
```

closed 2019-09-25 by JakubVanek