ev3dev/ev3dev #1316: Motor left on after program exit by the back button

JunOllyLi, 2019-09-26:
**Describe the bug**
When the program is running and motor is running, use the back button to exit program. It did go back to the file manager, but the motor was still running. I had to unplug the motors and re-plug them in to fix the issue.
This happens very rarely, though

**To Reproduce**
Steps to reproduce the behavior:
1. Go to file manager and select a program to run
2. The program is a simple line following code which call MoveSteering.on() in an endless loop
3. Use the back button to exit the program
4. The UI went back to file manager, but the motor is still running.

```
while True:
    r = (c.reflected_light_intensity - black) * 100 / (white - black) #normalizing the refelctive light
    diff = (r - 50) * gain  #calculating difference

    #limiting difference
    if diff > 100:
        diff = 100
    if diff < -100:
        diff = -100

    sp = speed
    m.on(diff, sp) #running
```

**Expected behavior**
Motor stops when program interrupted by user

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Window 10, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Other software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: snapshot-ev3dev-stretch-ev3-generic-2019-09-08.img.xz

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

**Additional context**
Add any other context about the problem here.

WasabiFan, 2019-09-28:
@dlech this is probably a brickrun issue.

dlech, 2019-10-23:
It looks like part of the program is missing. Can we get the full program to reproduce the problem?