ev3dev/ev3dev #1318: How do you access the tilt angle of the EV3 gyro sensor from the MicroPython environment?

LeeJenkins, 2019-09-27:
**Question**
How do you access the tilt angle of the EV3 gyro sensor from the MicroPython environment?

**Background Information**
Similar to [How to Get Value of 'TILT-ANGLE' in Gyro Sensor](https://github.com/ev3dev/ev3dev-lang-python/issues/616), but for the MicroPython environment.

**Screenshots**
N/A

**Desktop (please complete the following information):**
N/A

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: EV3 MicroPython
 - The code stamped on the gyro sensor ends with "N6" and we can see the "TILT_ANGLE" option in the Brickman interface.

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

dlech, 2019-09-27:
Regular MicroPython or Pybricks MicroPython?

LeeJenkins, 2019-09-27:
Pybricks. This is for a FIRST LEGO League team. I forgot about the other one.

dlech, 2019-09-27:
You can call `self._mode('TILT-ANGLE')` to set the mode and `self._value(0)` to read the value.

But this mode isn't really useful for most applications because switching modes causes the angle to be reset back to 0. 

The only cases I can think of where it would be useful are if using the second axis (and only the second axis) makes mounting the gyro sensor easier or if the robot does not need to uses both axes at the same time, e.g. the robot uses 1 axis to drive straight or make a precise turn, then when it arrives at its destination, it uses the second axis to make sure that it is not tipping over while lifting something heavy.

LeeJenkins, 2019-09-27:
Thanks for the answer and the extra info. One of the FLL missions this year involves driving up a ramp and stopping at the top. The kids will use the tilt axis to determine when the robot has reached the landing at the top of the ramp. I'll let them know not to switch modes.

LeeJenkins, 2019-09-28:
FYI for those who might find this issue/answer in the future, the proper mode setting is `self._mode('TILT-ANG')`, without the letters L-E at the end.

Thanks again, David. Today my son got his robot to drive up the ramp and then stop using the tilt angle information you posted here. Much appreciated!

closed 2019-09-28 by LeeJenkins