ev3dev/ev3dev #1319: NXT SoundSensor not detected.

naynt, 2019-09-29:
**What are you trying to do?**

I have been using the ev3.dev2 stretch, python for Mindstorms EV3. In general the program has been working extremely well for me and is brilliant to work with.
However I have an issue getting the program to detect my sound sensor. I was wondering whether you would be able to tell me how to get it working.

**What did you expect to happen**
For the brick to detect the SoundSensor.

The error I get is: ev3dev2.DeviceNotFound: SoundSensor(ev3-ports:in1) is not connected.

**Additional information**

**Software Versions**
Windows 10
Visual Studio Code with ev3.dev device browser extension (August 2019 version 1.38)
Image file: ev3dev-stretch-ev3-generic-2019-03-03
Kernel version: 4.14.96-ev3dev-2.3.2-ev3
Brickman v0.10.0
The SoundSensor works fine with lego mindstorms EV3 official software so can confirm the sensor works fine!
I have the Ultrasonic and Infrared official sensors and they work fine with ev3.dev2.

**Source Code**

Here is the program I am trying to run.

```
#!/usr/bin/env/python3

from ev3dev2.sensor.lego import SoundSensor
from ev3dev2.sensor import INPUT_1

ss = SoundSensor(INPUT_1)

```

**Background Information**

I am trying to ... because I want to build a robot that ...

dlech, 2019-09-29:
Analog NXT sensors cannot be completely automatically detected. So you have to get a port object for the port the sensor is connected to. Then use `set_device` to change the sensor to `lego-nxt-sound`. Then you probably have to wait a short time for the kernel driver to be set up (i.e. add a time delay in your program), then you can create the `SoundSensor` instance.

Alternately, it might be possible to set up the sensor using the `Device Browser` on the EV3 before starting your program.

There is a chart of sensor types [here](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html).

naynt, 2019-09-29:
Hi David,

Thank you very much for your quick response, it is extremely appreciated!

Kind Regards,

Simon

naynt, 2019-09-29:
Hi David,

Could you please clarify for me what you mean by port object?

Thank you!

Simon

dlech, 2019-09-29:
Sure.

https://python-ev3dev.readthedocs.io/en/ev3dev-stretch/ports.html

naynt, 2019-09-30:
Thank you,

Would you recommend the following code:

from ev3dev2.sensor.lego import SoundSensor

from ev3dev2.sensor import INPUT_1

from ev3dev2.port import LegoPort

from time import sleep

p = LegoPort(INPUT_1)

p.set_device = 'lego-nxt-sound'

sleep(1)

s = SoundSensor()

Thank you for your time!

Kind Regards,

Simon

https://python-ev3dev.readthedocs.io/en/ev3dev-stretch/ports.html

dlech, 2019-09-30:
looks reasonable to me

closed 2019-10-08 by naynt