ev3dev/ev3dev #1320: Split_NXT-v2 port splitter not detected by ev3dev2 MIndstorms EV3.

naynt, 2019-09-29:
**What are you trying to do?**

I have plugged in two sensors: Ultrasonic Sensor and Infrared Sensor (official lego), into
Mindsensors Split_NXT-v2 port splitter and the port splitter itself into INPUT_1 of the brickman.

**What did you expect to happen**
For the brickman to detect both these sensors. I am aware that some 3rd party sensors have
specific addresses and if they happen to have the same one like 0x08, you have to manually change them. However these EV3 mindstorm sensors should not conflict.

**What actually happened**
Both sensors where not detected

**Source Code**

Here is the program I am trying to run.

```
from ev3dev2.sensors.lego import UltrasonicSensor, InfraredSensor
from ev3dev.sensor import INPUT_1

us = UltrasonicSensor(INPUT_1)
ir = InfraredSensor(INPUT_1)
```

**Additional information:**

Visual Studio Code with ev3.dev device browser extension (August 2019 version 1.38)
Windows 10  
Image file: ev3dev-stretch-ev3-generic-2019-03-03
Kernel version: 4.14.96-ev3dev-2.3.2-ev3
Brickman v0.10.0

dlech, 2019-09-29:
SPLIT-Nx-v2 only works with NXT I2C sensors (and possibly motors). It won't won't work with EV3 sensors because they use UART instead of I2C. See #1311 for a recent discussion.

Mindsensors has a [different product](http://www.mindsensors.com/ev3-and-nxt/23-ev3-sensor-multiplexer-for-ev3-or-nxt) for multiplexing EV3 sensors.

naynt, 2019-09-29:
Hi David,

Thank you for that information!

Much appreciated!

Regards,

Simon

closed 2019-10-08 by naynt