ev3dev/ev3dev #1321: docker exec --tty brickman_armel make error in MACOS

zhanwen, 2019-10-02:
**Question**
docker exec --tty brickman_armel make

[  6%] Built target brickman_data
[  8%] Linking C executable brickman
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabi/8/../../../libev3devkit-0.5.so: undefined reference to `pow10'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/brickman.dir/build.make:1366: brickman] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/brickman.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

**Background Information**

i already install [ev3devKit](https://github.com/ev3dev/ev3devKit), brickman step have been success. ./docker/stepup.sh armel 

**Screenshots**
[image]

zhanwen, 2019-10-02:
./docker/setup.sh armel 

[image]

dlech, 2019-10-02:
`pow10(...)` probably needs to be replaced with `pow(10,...)`

closed 2019-10-02 by zhanwen

jankow86, 2020-08-02:
I am expecting exactly the same issue. Could you please guide me how to solve this problem?