ev3dev/ev3dev #1323: how modify ev3dev brickman src code

zhanwen, 2019-10-02:
**Question**
I want to modify the screen message, for example 'File Browser' to ‘文件’.

dlech, 2019-10-02:
It is hard-coded, so you will have to build your own copy from source. https://github.com/ev3dev/brickman

zhanwen, 2019-10-02:
yeah,  i build brickman have meet above problem.

`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev build`
[image]

i have execute 
`sudo apt-get update`
`sudo apt-get install ubuntu-dev-tools pbuilder-ev3dev`
`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev base`

dlech, 2019-10-02:
What OS/version is the virtual machine running?

zhanwen, 2019-10-02:
yeah, yeah. virtual machine not running ?

zhanwen, 2019-10-02:
`uname -a`
`Linux xxx-VirtualBox 4.4.0-142-generic #168~14.04.1-Ubuntu SMP Sat Jan 19 11:26:28 UTC 2019 x86_64 _86_64 GNU/Linux`

dlech, 2019-10-02:
That is the kernel version. What is the OS version. Ubuntu 18.04?

dlech, 2019-10-02:
I see it now. 14.04. Please use 18.04.

zhanwen, 2019-10-02:
yeah, i have use 18.04
`uname -a`
`Linux xyz-pc 5.0.0-29-generic #31~18.04.1-Ubuntu SMP Thu Sep 12 18:29:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
`
[image]

zhanwen, 2019-10-02:
i have add pakcage repository to apt by running
`sudo wget -P /etc/apt/sources.list.d/ http://archive.ev3dev.org/ubuntu/ev3dev-xenial.list`

dlech, 2019-10-02:
If you are running ` OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev base`, are you also using the `ev3dev-stretch` branch of brickman?

> sudo wget -P /etc/apt/sources.list.d/ http://archive.ev3dev.org/ubuntu/ev3dev-xenial.list

This will not get the right tools. Please do this instead

    sudo apt-add-repository ppa:ev3dev/tools

zhanwen, 2019-10-02:
i fork brickman to my github, and git clone my brickman, then 
`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev base`
`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev build`

dlech, 2019-10-02:
The default branch is `ev3dev-buster` so this is what you get when it is cloned. In the `brickman` directory, `git checkout ev3dev-stretch` will checkout the version for ev3dev stretch.

zhanwen, 2019-10-02:
thanks, i check branch found `ev3dev-buster`, thanks,thanks
and i have checkout the version for `ev3dev stretch`. then
`sudo apt-add-repository ppa:ev3dev/tools`
`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev base`
`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev build`
[image]

dlech, 2019-10-02:
I don't see what the actual error is in the screenshot, but one thing I forgot to say is that you probably need to run `git submodule update` after changing the branch.

zhanwen, 2019-10-02:
i have been try, thanks. then
`OS=debian DIST=stretch ARCH=armel pbuilder-ev3dev build`
[image]
[image]

dlech, 2019-10-02:
This is what it looks like when I build it:

```
david@freyr:~/work/ev3dev-dpkg/brickman$ git checkout ev3dev-stretch
Switched to branch 'ev3dev-stretch'
Your branch is up to date with 'origin/ev3dev-stretch'.
david@freyr:~/work/ev3dev-dpkg/brickman$ git submodule update 
david@freyr:~/work/ev3dev-dpkg/brickman$ OS=debian ARCH=armel DIST=stretch pbuilder-ev3dev dev-build
gbp:info: Building with pbuilder for stretch:armel
gbp:info: Performing the build
Building with pbuilder for distribution stretch, architecture armel
I: using pbuilder as pbuilder
dpkg-checkbuilddeps: error: Unmet build dependencies: libev3devkit-dev
W: Unmet build-dependency in source
dpkg-source: info: using options from brickman/debian/source/options: --tar-ignore=.* --tar-ignore=build* --tar-ignore=doc/sphinx/_build --tar-ignore=doc/sphinx/_html_extra
dh clean --with systemd
   dh_clean
dpkg-source: info: using options from brickman/debian/source/options: --tar-ignore=.* --tar-ignore=build* --tar-ignore=doc/sphinx/_build --tar-ignore=doc/sphinx/_html_extra
dpkg-source: info: using source format '3.0 (native)'
dpkg-source: info: building brickman in brickman_0.10.2.tar.xz
dpkg-source: info: building brickman in brickman_0.10.2.dsc
I: Generating source changes file for original dsc
dpkg-genchanges: info: including full source code in upload
dpkg-source: info: using options from brickman/debian/source/options: --tar-ignore=.* --tar-ignore=build* --tar-ignore=doc/sphinx/_build --tar-ignore=doc/sphinx/_html_extra
[sudo] password for david: 
I: pbuilder: network access will be disabled during build
I: Current time: Wed Oct  2 09:59:28 CDT 2019
I: pbuilder-time-stamp: 1570028368
I: Building the build Environment
I: extracting base tarball [/home/david/pbuilder-ev3dev/debian/base-stretch-armel.tgz]
I: copying local configuration
I: mounting /proc filesystem
I: mounting /sys filesystem
I: creating /{dev,run}/shm
I: mounting /dev/pts filesystem
I: redirecting /dev/ptmx to /dev/pts/ptmx
I: policy-rc.d already exists
I: Obtaining the cached apt archive contents
I: Copying source file
I: copying [../brickman_0.10.2.dsc]
I: copying [../brickman_0.10.2.tar.xz]
I: Extracting source
dpkg-source: warning: extracting unsigned source package (brickman_0.10.2.dsc)
dpkg-source: info: extracting brickman in brickman-0.10.2
dpkg-source: info: unpacking brickman_0.10.2.tar.xz
I: using fakeroot in build.
I: Installing the build-deps
 -> Attempting to parse the build-deps 
 -> Considering build-dep debhelper (>= 9)
   -> Trying debhelper
 -> Considering build-dep dh-systemd
   -> Trying dh-systemd
 -> Considering build-dep cmake
   -> Trying cmake
 -> Considering build-dep valac (>= 0.24)
   -> Trying valac
 -> Considering build-dep libgirepository1.0-dev
   -> Trying libgirepository1.0-dev
 -> Considering build-dep libgudev-1.0-dev
   -> Trying libgudev-1.0-dev
 -> Considering build-dep libev3devkit-dev
   -> Trying libev3devkit-dev
 -> Considering build-dep netpbm
   -> Trying netpbm
 -> Installing  debhelper dh-systemd cmake valac libgirepository1.0-dev libgudev-1.0-dev libev3devkit-dev netpbm
Reading package lists...
Building dependency tree...
Reading state information...
debhelper is already the newest version (10.2.5).
The following additional packages will be installed:
[... 561 lines cut ...]
--   Found gmodule-2.0, version 2.50.3
--   Found gio-2.0, version 2.50.3
--   Found gio-unix-2.0, version 2.50.3
--   Found grx-3.0, version 3.1.1
--   Found gudev-1.0, version 230
-- Found Vala: /usr/bin/valac  
-- checking for a minimum Vala version of 0.24
--   found Vala, version 0.34.7
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_INSTALL_LOCALSTATEDIR
    CMAKE_INSTALL_SYSCONFDIR

-- Build files have been written to: /build/brickman-0.10.2/obj-arm-linux-gnueabi
   dh_auto_build
        make -j1
make[1]: Entering directory '/build/brickman-0.10.2/obj-arm-linux-gnueabi'
/usr/bin/cmake -H/build/brickman-0.10.2 -B/build/brickman-0.10.2/obj-arm-linux-gnueabi --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /build/brickman-0.10.2/obj-arm-linux-gnueabi/CMakeFiles /build/brickman-0.10.2/obj-arm-linux-gnueabi/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all
make[2]: Entering directory '/build/brickman-0.10.2/obj-arm-linux-gnueabi'
```

zhanwen, 2019-10-02:
i have build success, very very very thanks. and i ask question
[image]

`brickman_0.10.2_armel.deb` 
`brickman-dbgsym_0.10.2_armel.deb`
the two file have  difference?

dlech, 2019-10-02:
`dbgsym` is short for debug symbols. It is only needed for debugging.

`brickman_0.10.2_armel.deb` is the main package.

zhanwen, 2019-10-02:
> `dbgsym` is short for debug symbols. It is only needed for debugging.
> 
> `brickman_0.10.2_armel.deb` is the main package.

i have very appreciated. very thanks.

zhanwen, 2019-10-02:
i have one question, now, i build `brickman_0.10.2_armel.deb`, then upload `robot@ev3dev`.
then execute 
`sudo dpkg -i brickman_0.10.2_armel.deb`
i have found previous `brickman` version is `0.10.0`, `ev3dev` is `ev3dev-stretch-ev3-generic-2019-03-03.zip`, now i have use `0.10.2` replace `0.10.0` will have problem?

dlech, 2019-10-02:
It may require a newer ev3devkit library. You can run `sudo apt install -f` after this to fix any dependency problems.

zhanwen, 2019-10-02:
i have to do build brickman, fix controller.vala Label context, for example 'File Browser' to '文件浏览'.
i upload `robot@ev3dev`,then 
`sudo dpkg -i brickman_0.10.2_armel.deb`
i have execute `sudo apt install -f`
[image]
but found no package install. then `ev3dev` start timeout.
[image]
i restart ev3dev still timeout.

zhanwen, 2019-10-02:
my goal is to replace the english content on brickman's screen. 
for example:
`File Browser` to `文件浏览`
`Device Browser` to `设备浏览`
and so on.

dlech, 2019-10-02:
If you run `sudo systemctl status brickman` what is the output?

zhanwen, 2019-10-02:
> sudo systemctl status brickman

[image]

dlech, 2019-10-02:
try running `sudo sh -c 'G_MESSAGES_DEBUG=all brickman'`

zhanwen, 2019-10-02:
> sudo sh -c 'G_MESSAGES_DEBUG=all brickman'

[image]

dlech, 2019-10-02:
I guess `sudo apt install -f` didn't do its job. Try `sudo apt install libgrx-3.0-2-plugin-linuxfb`

zhanwen, 2019-10-02:
now, have execute success, install one package, 
[image]
but found ev3dev no show chinese. 
[image]

i have been install 
`en_US ISO-8859-1`
`zh_CN GB2312`
`zh_CN.GBK GBK`
`zh_CN.UTF-8 UTF-8`
`zh_TW BIG5`
`zh_TW.UTF-8 UTF-8`
in `ssh terminal` can show chinses. 
[image]

dlech, 2019-10-02:
brickman does not have any translations or support for translations. All text is hard-coded. It does not use console fonts either it only supports bitmap fonts. So the program will have to be modified to use a different font and all the text will have to be changed using a character encoding that is supported by the font.

zhanwen, 2019-10-02:
yeah, i install font, found restart ev3dev show english font have been change. now i want modified font, i should modified which files? or i must modified bitmap fonts?

zhanwen, 2019-10-02:
> brickman does not have any translations or support for translations. All text is hard-coded. It does not use console fonts either it only supports bitmap fonts. So the program will have to be [...]

through ev3dev tools（brickman or other tools） can complete font modified? thanks.

dlech, 2019-10-02:
I think the default font is defined in ev3devkit instead of brickman.

https://github.com/ev3dev/ev3devKit/blob/a487854c943b2946001e2c1327f3209d3928256c/src/ui/Fonts.vala

zhanwen, 2019-10-02:
i think `Font.load ("fixed sans", 8);` code.
this assign to font size, but not found Font type. I may not understand vala language,  Font type is server defualt?

dlech, 2019-10-02:
It uses [fontconfig](https://www.freedesktop.org/wiki/Software/fontconfig/) to find the best matching font for the name given.

zhanwen, 2019-10-02:
> It uses [fontconfig](https://www.freedesktop.org/wiki/Software/fontconfig/) to find the best matching font for the name given.
[image]
 The closest matching font is selected. now, i can assign to font type ?
Finally, I would like to ask, is it possible to modified font, for example
[image]
`File Browser` to `文件浏览`.
thanks.

dlech, 2019-10-02:
It is possible, but it is very much work which I have already explained in https://github.com/ev3dev/ev3dev/issues/1323#issuecomment-537577898.

zhanwen, 2019-10-02:
> It is possible, but it is very much work which I have already explained in [#1323 (comment)](https://github.com/ev3dev/ev3dev/issues/1323#issuecomment-537577898).

Could you please list the steps, I will try hough it is hard. Finally I would appreciate it.

dlech, 2019-10-02:
1. Find some fonts that you want to try. For example, [xfonts-unifont](https://packages.debian.org/sid/xfonts-unifont) package has every possible unicode character. Or there may be nicer looking fonts in other packages. Any debian package with `xfonts-...` should work.
2. Install the `xfonts-...` packages on the EV3.
3. Change the fonts in ev3devkit. For example "unifont", 32. (unifont only has one size of Chinese character, so it is probably not the best choice) https://github.com/ev3dev/ev3devKit/blob/ev3dev-stretch/src/ui/Fonts.vala
4. Compile, the ev3devkit packages and install them on the EV3
5. Change the text in brickman. This might need to use a character encoding that is compatible with the font that is selected (I am not sure). For example: https://github.com/ev3dev/brickman/blob/ev3dev-stretch/src/controller/FileBrowserController.vala#L43
6. Compile brickman and install the package on the EV3.

zhanwen, 2019-10-02:
I very appreciate, I'll tell you the result of my attempt in tomorrow.
Finally, thank you very much for your patient reply.

zhanwen, 2019-10-03:
> 3\. Change the fonts in ev3devkit. For example "unifont", 32. (unifont only has one size of Chinese character, so it is probably not the best choice)

change the fonts in ev3devkit. For example "unifont", 32
`default_font = Font.load ("fixed sans", 8);` to `default_font = Font.load ("fixed sans", 32);`?
or
`Font.load ("fixed sans", 8);` , `Font` to `unifont`?
this change size ok ?

dlech, 2019-10-03:
> `Font.load ("fixed sans", 8);` , `Font` to `unifont`?

Change "fixed sans" to "unifont" or name of some other font.

>     3\. nstall ev3devkit for ev3
>         `sudo dpkg -i libev3devkit-0.5-0_0.5.3.1_armel.deb`
>         but show error

When you build ev3devkit, it creates more than 1 package. The data package and the linuxfb pluging package that are created have to be copied to the EV3 and installed with `sudo dpkg -i` also.

> why ev3 screen always show `File Browser`, this context i not config, ev3 screen show message from ?

Maybe the text is coming from somewhere else? Or maybe the modified brickman program did not get installed correctly?

zhanwen, 2019-10-03:
For example "unifont", 32. (unifont only has one size of Chinese character, so it is probably not the best choice)

Could you please explain 32 means, if 32 one size of Chinese character, 64 two size of Chinese character?

dlech, 2019-10-03:
32 is just a recommendation of the font size you want (in points, not pixels) The actual size will depend on what is available in the font that is chosen.

zhanwen, 2019-10-03:
yeah, i understand ahead minutes.
now, i found my mistake =), ev3 screen always show english message, because my modifies file in stretch-buster branch, and my build in the stretch branch, this content not change, :(

zhanwen, 2019-10-03:
now, i can show ev3devkit version or status?
for example, brickman have
 `sudo systemctl status brickman`
 `sudo sh -c 'G_MESSAGES_DEBUG=all brickman'`
i want show ev3devkit version, see if they are change.

dlech, 2019-10-03:
If you changed the version in the CMakeLists.txt file, then the version should be on the "About" page on the EV3 screen.

dlech, 2019-10-03:
I don't think there is a way to show the version unless you add one.

zhanwen, 2019-10-04:
@dlech Thank you very much for your help in these two days. I have solved my problem. Thank you very much again.

zhanwen, 2019-10-04:
Hi, dlech. 
Now i have new problem, I want vala programing, could you recommend IDEA(eg: Java IDEA), vala language have programming tools? or can you recommend a book about vala (eg: java vs Java thinking program).

dlech, 2019-10-04:
I don't know of any books about Vala. This may be helpful if you already know Java: https://wiki.gnome.org/Projects/Vala/ValaForJavaProgrammers

Programming tools: https://wiki.gnome.org/Projects/Vala/Tools

zhanwen, 2019-10-04:
Happy new day，thank you very much again, i go to study ^_^.