ev3dev/ev3dev #1325: Pybricks is already running on this device. Exiting...

DigitalRookie4, 2019-10-09:
I am trying to run my code in my lego Ev3 robot and Once I execute it from my robot's file on my computer the program gets executed but after about 5 seconds the program stops and gives off the message `Pybricks is already running on this device. Exiting...`.

dlech, 2019-10-09:
FYI, Pybricks is different from ev3dev-lang-python.

If you reboot the EV3, does the problem go away?

Can you share the program?

DigitalRookie4, 2019-10-12:
No the issue is still happening

dlech, 2019-10-12:
Ah, there is a bug in the Pybricks sound module that if something else is using the sound driver, then it thinks that `pybricks-micropython` is already running. `pyglet` is probably using the sound card as well.

For now, I'm afraid that the only workaround is probably to not do `from pybricks import ev3brick as brick` and use `pyglet` for sound and display.

r1chter1, 2019-11-16:
`File "c:/Users/Administrator/Documents/WRO2020/Bissmillah/main.py", line 2, in <module>
    from pybricks import ev3brick as brick
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\pybricks\ev3brick.py", line 43, in <module>
    class display:
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\pybricks\ev3brick.py", line 51, in display
    def image(cls, file_name: str, alignment: Align = Align.CENTER, coordinate: Tuple[int, int] = None, clear: bool = True):
AttributeError: type object 'Align' has no attribute 'CENTER'`
I have this issue please help. Does 'pybricks' work at all?. This error is in library itself. wth?
@drewwhis

dlech, 2019-11-17:
@r1chter1, it looks like you are running your program on your Windows computer. That doesn't work. You can only run the computer on the EV3.

drewwhis, 2019-11-18:
@r1chter1 are you referring to the pybricks-stubs package? There's no "real" code there; the package only exists to make it easier to **write** the code on a computer; the code will still only **run** on an EV3 brick. If you get execution errors on the brick, it's possible there is a difference between what's in the API provided by LEGO and what's in the stubs package.