ev3dev/ev3dev #1326: compiler linaro

cho934, 2019-10-10:
Hello,

What are the versions of gcc_linaro which are compatible with ev3dev ?

I'm trying to migrate from "sourcery codebench arm-2013.11-33" to the last one "gcc-linaro-7.4.1-2019.02".

And I saw also this available repo in ev3dev:
https://github.com/ev3dev/gcc-linaro-arm-linux-gnueabihf-6.4

What are your advices ?

cho934, 2019-10-29:
Until now, I succeed to build with Linaro but not to execute the code on the EV3
It remains some tests to do with the following parameters with gcc:
-march=armv5te -mtune=arm926ej-s -msoft-float -mfloat-abi=soft

First test failed too, impossible to really build with the soft float...

cho934, 2019-11-28:
Here is the answer from Linaro

```
As an Open Source consortium, Linaro does not have any commercial product set; rather, we are member funded for member directed projects and efforts for Linux on Arm. Linaro Developer Technical Support (LDTS) gives priority to tickets from our members and our services customers. Those tickets are reviewed within 48 hours by an LDTS Support Engineer and will receive close attention from our support team. So if you are from a member company (http://www.linaro.org/members) and made this request with your personal email account, please resend your request using your business email.

LDTS will work on community (non-Member) tickets on a best effort basis. If you are not from a Member Company, please be aware that our free-of-charge service does not provide comprehensive technical support. The scope is much less broad than that offered to members and customers but if we have information to hand that can help you, we will share it.

Regarding your query, the Linaro toolchain (and its successor provided by Arm) is designed for Armv7 architecture systems (all Cortex-A processors are Armv7 or higher). The toolset is not appropriate for use with Armv5. Whilst the compiler can generate code for Armv5 architecture processors such as ARM926 the libraries bundled with the toolchain will contain Armv7 instructions that cannot be executed on your platform.

You should look for an alternative toolset supplier that provides a compiler that supports your system, alternatively if your company is interested in contracting our Services group (https://www.linaro.org/services/ ) to help you more directly then please let me know I will make the appropriate introductions. Thank you for your interest in Linaro.

Daniel
Linaro Support and Solutions Engineering.
```

cho934, 2019-12-09:
Hi, I continued to look for an arm v5 builder and I found this one:
https://toolchains.bootlin.com/releases_armv5-eabi.html

I tested sucessfully all versions eabi for glibc version 2.24.

With the  following arguments of g++ :
```
-std=c++0x -DGLIBCXX_USE_NANOSLEEP -O0 -g3 -Wall -c -fmessage-length=0 -marm -march=armv4t -mfloat-abi=soft
```

Build and execution with threads are both ok. I felt that the execution is a little bit slower using my read/write performance test in c++ (more than minimum 1ms with POSIX functions instead of minimum 300us)... strange...

If someone is available to test it too.

bmegli, 2019-12-10:
Compiler is one thing.

If you want to use libraries that are available on EV3 (e.g. installed from packages), in a sane manner, you will need sysroot (headers and libraries).

You  may copy sysroot from EV3. This can be done manually but there are tools for that, for example [here](https://gnutoolchains.com/raspberry/tutorial/sysroot). It doesn't matter that tutorial is for RaspberryPI, the tool only synchronizes library and header directories. You will also have to install *-dev packages on EV3 for libraries that you want to use (so that you have headers). 

Note that this means doing the thing somewhat backwards. Normally you would generate SDK (toolchain, sysroot) with the same tool that is used for building the image for device (Buildroot, Yocto, ...). 

If you are not using EV3 system libraries you may skip that (synchronizing sysroot).

dlech, 2019-12-10:
This is why https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/ is the recommended solution for cross-compiling. Everything is already set up in a docker image with an ev3dev-compatible sysroot.