ev3dev/ev3dev #1327: LEGO GyroSensor not connected

KWSmit, 2019-10-11:
**Describe the bug**
Autodetection of LEGO GyroSensor does not work on the latest snapshot (ev3dev-stretch-ev3-generic-2019-09-24). On `stretch` autodetect goes wrong and the EV3-brick does not connect the GyroSensor. You have to manually set the portmode to `ev3-uart` (using Brickman's Device Browser), then the EV3 brick does detect and connect the GysroSensor succesfully.

On `buster` (same snapshot release) even manually setting the port does not result in connecting the GyroSensor.

Works fine with older releases (tested it succesfully with `ev3dev-stretch-ev3-generic-2019-03-03`)

**To Reproduce**
Steps to reproduce the behavior:
1. Connect LEGO GyroSenor to LEGO EV3-brick.
2. Start EV3-brick.
3. Go to Brickman's Device Browser and see that GyroSensor is not detected.

**Expected behavior**
Succefull connection of GyroSensor

**Robot (please complete the following information):**
Device: LEGO EV3-brick and LEGO GyroSensor

```
Image file:         ev3dev-stretch-ev3-generic-2019-09-24
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165335AB3A
BOARD_INFO_TYPE=main
```

dlech, 2019-10-11:
Working for me with a 19N3 and 10N4 sensor. What else could be different about your setup? Any information in the output of `dmesg`?

[image]

KWSmit, 2019-10-11:
Weird, I had this problem for a couple of days, but cannot reproduce it anymore on `stretch`. All I did today was disassembling my prototyping setup (nothing fancy, just an EV3-brick, TouchSensor and GyroSensor and a bunch of Technic beams) and rebuild it again. I even tried a fresh new `stretch` image, but now it keeps working fine (on `stretch`). Could it be something electrical (bad cable or connector or whatsoever)?

As for `buster`, the Gyro still won't connect...

BTW, I have an 19N3 and 51N4 sensor.

I'll do some more testing the next days. When `stretch` keeps working fine, I'll close this issue and open a new one for `buster`, ok?

dlech, 2019-10-11:
If it worked after manually setting the mode of the input port, then it _probably_ isn't a hardware problem. I did notice recently that a color sensor would not work on boot sometimes, but I don't remember what OS I was running on the EV3.

This could also potentially be related to the Bluetooth issues in #1314.

dlech, 2019-11-23:
Found the bug in ev3dev-buster and ev3dev-stretch seems to be working now, so closing.

closed 2019-11-23 by dlech

KWSmit, 2019-11-24:
👍

xav-developer, 2024-11-22:
> Found the bug in ev3dev-buster and ev3dev-stretch seems to be working now, so closing.

Hi ! I have the same problem, build ev3dev-stretch-ev3-generic-2020-04-10 The gyro sensor does not connect , everything works if connected via the mindstorm app ...