ev3dev/ev3dev #1330: Automake/Make not working

Gideon357, 2019-10-14:
I was trying to download jhbuild and could not use the make command. I tried:

`sudo automake ./autogen.sh`
`sudo: automake: command not found`

`sudo make ./autogen.sh`
`sudo: make: command not found`

To install it i tried:
[image]
and 
[image]

I am trying to do this so I can use GTK and ev3dev kit UI for a menu loader to run python scripts faster in FLL.

dlech, 2019-10-14:
You need to run `sudo apt update` first.

You can start programs from the File Browser on the brick already, so I don't see how this would make things faster though.

dlech, 2019-10-14:
Also, I just looked up what jhbuild is. I don't think it will be useful in this case since we don't use it in for anything ev3dev.

dlech, 2019-10-14:
Although I suppose you could use ev3devKit in python to make something that could load programs faster if it doesn't start a new python process, but in that case you don't need any build tools to write python program.

Gideon357, 2019-10-14:
I made a menu that runs functions on a button press so everything loads on startup. I use jhbuild to install GTK. ev3devKit includes a UI tool that needs GTK. Thanks for the input I will try this.

closed 2019-10-15 by Gideon357