ev3dev/ev3dev #1332: ev3dev2 UI

Gideon357, 2019-10-15:
Other than EV3devKit, are there any ways to make UIs in ev3dev. If not, is there a way to use pure Python to make a UI. Also, is there a way to use EV3devKit UI without PyGObject. If I were to go about making a UI toolkit in pure Python (probably not, I would have to use C/C++).

WasabiFan, 2019-10-16:
If you're using `ev3dev-lang-python`, there are both drawing-based `Display` interfaces and text-mode `Console` interfaces. The latter has a demo script for making menus, so that might be your best bet for simple operations, but doesn't support anything more advanced.

Gideon357, 2019-10-16:
Cool, that's actually what I am looking to do. Could you link it? Also, do you have any recommendations for the more advanced UIs and a library? Could I use Tkinter with ev3dev? Thanks for responding.

WasabiFan, 2019-10-16:
Console: https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/ev3dev-stretch/console.html

Display: https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/ev3dev-stretch/display.html

I don't know about using other libraries but if one supports operating on arbitrary frame data it should be workable.

Gideon357, 2019-10-16:
Hi wasabi fan,
I looked at the docs and didn’t see the menu example. I am looking on mobile so I may have missed it. If you could link that it would be helpful. Thanks a lot.

WasabiFan, 2019-10-17:
Apologies; it looks like we didn't get the demo migrated to our proper places. You can see it here: https://github.com/ev3dev/ev3dev-lang-python/blob/2b8aec238a3de79079fda7a1f1e88ce833be8642/utils/console_menu.py

Gideon357, 2019-10-17:
@wasabifan thanks so much for your help.

closed 2019-10-19 by Gideon357