ev3dev/ev3dev #1333: Pip not working

Gideon357, 2019-10-19:
Hi,
I have been trying to `pip install` a package and I get an error that says `pip not found`. To fix this I have used the `pip3` command which generates the same error. I have try `sudo apt-get install pip` and `sudo apt-get install pip3` and `sudo apt-get python3-pip` as well as the get-pip.py file from PyPA. All of these either finish running and still don't work or disconnect my SSH session and give a broken pipe error. I have tried this on the en3 and bridge100 interfaces.

Gideon357, 2019-10-19:
I used both Mac and Windows OS

dlech, 2019-10-19:
See #1273, #873, etc.

meglai, 2019-10-22:
> I am having the same issue running `pip3`. In #1273 and #873, the problem apparently was resolved by re-installing pip3 multiple times. I installed and uninstalled many times already, but when run `pip3 --version`, nothing happens after I waited for a long time. Please help. I am trying to install 'pip3 install paho-mqtt' to complete my ioT project.

Gideon357, 2019-10-22:
Try `sudo apt-get install python3 paho-mqtt`. Hope I can help

wiseOsprey1, 2021-05-10:
pip3 install always fails for me

robot@ev3dev:~$ sudo pip3 install paho-mqtt
Collecting paho-mqtt
  Could not find a version that satisfies the requirement paho-mqtt (from versions: )
No matching distribution found for paho-mqtt

I tried the following but it failed too:

robot@ev3dev:~$ sudo apt install python3 paho-mqtt
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package paho-mqtt

dlech, 2021-05-10:
PyPi no longer supports Python 3.5. Ev3dev-stretch uses Python 3.5. So `pip3` is probably no longer usable for any package.

wiseOsprey1, 2021-05-10:
Thanks.  What's the best way to install paho-mqtt?  Is there an image archive that I can download and install?  If so, can you provide instructions?

wiseOsprey1, 2021-05-10:
Thanks.  I was able to import the module into my program.  My program acts like it's working properly, but the messages I'm posting to broker.hivemq.com from my EV3 aren't being seen by my subscriber on Windows 10.  That subscriber does see messages from my Windows 10 mosquitto_sub test application.  I'll look for discussions addressing that issue.  If you know of any, please advise.

gideongrinberg, 2021-05-10:
You can also always use the [files section](https://pypi.org/project/paho-mqtt/#files) on PyPi to download a wheel or source distribution. Seeing as `pip` is no longer usable with ev3dev, I'm going to close the issue.

closed 2021-05-10 by gideongrinberg