ev3dev/ev3dev #1334: I am trying to use EV3 sensor multiplexer from mindsensors on EV3Micropython

Sunrise98, 2019-10-20:
**Question**

How do you ...?

**Background Information**

I am trying to ... because I want to build a robot that ...

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

daniel974, 2019-10-20:
> Reopened #1334 <https://github.com/ev3dev/ev3dev/issues/1334> .
>
>
> L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
> https://www.avast.com/antivirus

Please

how to unsubscribe from this?

thank's

dlech, 2019-10-20:
@daniel974 you are currently "watching" this project, so you can log into GitHub and "unwatch" it. You can find more info about GitHub notifications at https://help.github.com/en/articles/about-notifications.

@Sunrise98  You can find some information here: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/muxs.html#ev3-sensor-multiplexer-for-ev3-or-nxt

In Python you will need to create a `LegoPort` object for each port that you want to use then `port.set_device = 'lego-ev3-color'` for each sensor connected to the mux. Then add a delay (usually 0.5 seconds is enough) before creating the sensor objects.