ev3dev/ev3dev #1336: Brickpi3 and I2C device

giga, 2019-10-22:
I am trying to use an I2C device (pixycam2) on a brickpi3 using ev3dev stretch.

I am seeing a weird bevahior: when I set an input port mode to nxt-i2c, the motor (medium motor) does not run anymore. When I switch back the port mode to none, motor starts running. It does not matter if there is an actual sensor plugged in the port or not.

To reproduce: run the following program.

```
#!/usr/bin/env python3

from ev3dev2.sensor import INPUT_1, INPUT_2, INPUT_3, INPUT_4
from ev3dev2.motor import LargeMotor, OUTPUT_A, OUTPUT_B, OUTPUT_C, OUTPUT_D
from ev3dev2.port import LegoPort
from time import sleep

smotor = LargeMotor(OUTPUT_C)

lego_port = LegoPort(INPUT_2)
lego_port.mode = 'nxt-i2c'
sleep(0.5)

print('run motor')
smotor.on(-100)
while True:
    pass
```

Motor does not run. Run 
```
echo none > /sys/class/lego-port/port1/mode
```
in an other shell. Motor starts running.

dlech, 2019-11-25:
Thanks for the bug report and simple test case. I was able to reproduce this, so there appears to be a bug in the kernel driver.

dlech, 2019-11-25:
After some more testing, I got it working by setting up the sensor on the port.

```python
#!/usr/bin/env python3

from ev3dev2.sensor import INPUT_1, INPUT_2, INPUT_3, INPUT_4
from ev3dev2.motor import LargeMotor, OUTPUT_A, OUTPUT_B, OUTPUT_C, OUTPUT_D
from ev3dev2.port import LegoPort
from time import sleep

smotor = LargeMotor(OUTPUT_C)

lego_port = LegoPort(INPUT_2)
lego_port.mode = 'nxt-i2c'
lego_port.set_device = 'ms-absolute-imu 0x11'  # this fixes it
sleep(0.5)

print('run motor')
smotor.on(-100)
while True:
    pass
```

So this tells me that it is a BrickPi3 firmware issue. This should be tested on Raspbian for Robots and reported to Dexter Industries.