ev3dev/ev3dev #1337: EV3 Sensor Mux splitter not detecting EV3 lego sensors

naynt, 2019-10-23:
This is a continuation of Issue #1335 

**What are you trying to do?**
From the previous post #1335, I discovered after that my EV3 Sensor Multiplexor (Mindsensors) seems to be the main problem as it does not seem to detect any of the EV3 sensors.

According to ev3dev documentation, ev3dev2 should be able to automatically detect the EV3 Sensor MUX and is created for NXT/I2C. It has three I2C addresses, 0x50,0x51 and 0x52. It contains 4 ports: Host, C1, C2 and C3. The host port is connected to port 1 of the Brick. I ran the following example script and I get the following error"

`#!/usr/bin/env python3 

#---------------------------------------------------------------------------------------------------------------

# Test: EV3 Sensor MUX

#----------------------------------------------------------------------------------------------------------------

from ev3dev2.sensor.lego import UltrasonicSensor
from sys import stderr
from time import sleep
from ev3dev2.sensor import INPUT_1

# port 1 = EV3SensorMUX by Mindsensors
# port 1 = Ultrasonic sensor

us = UltrasonicSensor(INPUT_1)
us_value = us.value(0)
us_dist = us.distance_centimeters() # takes multiple measurements
us_other = us.other_sensor_present() # detects if other Ultrasensor sensors present nearby
us_numvalues = us.num_values
us_address = us.address
us_units = us.units
print("Ultrasonic value: ",us_value, file=stderr)
print("Distance CM: ",us_dist,file=stderr)
print("other us sensors present: ",us_other,file=stderr)
print("Distance number values: ",us_numvalues,file=stderr)
print("us address: ",us_address,file=stderr)
print("Distance units: ",us_units,file=stderr)`

Traceback (most recent call last):
  File "/home/robot/Main Scripts/sensor.py", line 18, in <module>
    us = UltrasonicSensor(INPUT_1)
  File "/usr/lib/python3/dist-packages/ev3dev2/sensor/lego.py", line 447, in __init__
    super(UltrasonicSensor, self).__init__(address, name_pattern, name_exact, driver_name=['lego-ev3-us', 'lego-nxt-us'], **kwargs)
  File "/usr/lib/python3/dist-packages/ev3dev2/sensor/__init__.py", line 94, in __init__
    super(Sensor, self).__init__(self.SYSTEM_CLASS_NAME, name_pattern, name_exact, **kwargs)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 210, in __init__
    chain_exception(DeviceNotFound("%s is not connected." % self), None)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 38, in chain_exception
    raise exception from cause
ev3dev2.DeviceNotFound: UltrasonicSensor(ev3-ports:in1) is not connected.
----------
Exited with error code 1.

**Question:**
Do I need to add anything additional to the script?

Sorry for all the posts.

----------------------------------------------------------------------------------------------------------

Windows 10
Visual Studio Code with ev3.dev device browser extension (August 2019 version 1.38)
Image file: ev3dev-stretch-ev3-generic-2019-03-03
Kernel version: 4.14.96-ev3dev-2.3.2-ev3
Brickman v0.10.0

dlech, 2019-10-23:
That is expected. See notes at http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/muxs.html#ev3-sensor-multiplexer-for-ev3-or-nxt

Also, as stated in the previous issue, you cannot use `INPUT_1` to mean a port on the multiplexer. It should be something like `'ev3-ports:in1:i2c50:mux1'` instead. The program above is expecting the ultrasonic sensor to be directly connected to the EV3.

naynt, 2019-10-23:
Thank you for the clarification and providing me an example. I got confused, due to the split-NX2-v2  port not requiring this as it uses I2C address system.

closed 2019-10-23 by naynt

naynt, 2019-10-24:
**Would the following script be correct, 3 EV3 sensors to the EV3-sensor-MUX?

`us_mux = LegoPort(address='in2:i2c50:mux1') 
ir_mux = LegoPort(address='in2:i2c51:mux2') 
gyro_mux = LegoPort(address='in2:i2c52:mux3') 

us_mux.set_device = 'lego-ev3-us'
ir_mux.set_device = 'lego-ev3-ir'
gyro_mux.set_device = 'lego-ev3-gyro'`