ev3dev/ev3dev #1338: EV3-Sensor-MUX and sensor detection (the struggle continues)

naynt, 2019-10-24:
**Problems with my following script in getting the EV3 brick to detect EV3 sensors via EV3-Sensor-MUX**

`#!/usr/bin/env python3 

#---------------------------------------------------------------------------------------------------------------

# Test: EV3 Sensor MUX

#----------------------------------------------------------------------------------------------------------------

from ev3dev2.sensor.lego import UltrasonicSensor,InfraredSensor,GyroSensor
from ev3dev2.port import LegoPort
from sys import stderr
from time import sleep
from ev3dev2.sensor import INPUT_2

# port 2 = EV3-Sensor-MUX, Ultrasonic sensor, IR sensor, Gyro sensor

us_port = LegoPort(address='in2:i2c50:mux1') # port 1 of MUX address i2c '50' connected to port 2 EV3 Brick
ir_port = LegoPort(address='in2:i2c51:mux2') # port 2 of MUX address i2c '51' connected to port 2 EV3 Brick
gyro_port = LegoPort(address='in2:i2c52:mux3') # port 3 of MUX address i2c '52' connected to port 2 EV3 Brick

us_port.set_device = 'lego-ev3-us' 
ir_port.set_device = 'lego-ev3-ir'
gyro_port.set_device = 'lego-ev3-gyro'

us = UltrasonicSensor()
ir = InfraredSensor()
gyro = GyroSensor()

us_value = us.value(0)
us_dist = us.distance_centimeters() # takes multiple measurements
us_other = us.other_sensor_present() # detects if other Ultrasensor sensors present nearby
us_numvalues = us.num_values
us_address = us.address
us_units = us.units
print("Ultrasonic value: ",us_value, file=stderr)
print("Distance CM: ",us_dist,file=stderr)
print("other us sensors present: ",us_other,file=stderr)
print("Distance number values: ",us_numvalues,file=stderr)
print("us address: ",us_address,file=stderr)
print("Distance units: ",us_units,file=stderr)

ir_value = ir.value(0)
ir_proximity = ir.proximity # an estimate of the distance between the sensor and the objects in front of it
ir_beacon= ir.distance(channel=1) # returns distance (0,100) to the beacon on the given channel
ir_heading = ir.heading(channel=1) # returns heading (-25,25) to the beacon on the given channel
ir_address = ir.address
print("Infrared value: ",ir_value,file=stderr)
print("Distance proximity: ",ir_proximity,file=stderr)
print("Distance from beacon ",ir_beacon,file=stderr)
print("Heading or direction: ",ir_heading,file=stderr)
print("IR address: ",us_address,file=stderr)

gyro_value = gyro.value(0)
gyro_angle = gyro.angle
gyro_address = gyro.address
gyro_units = gyro.units

print("Gyro value: ",gyro_value,file=stderr)
print("Gyro angle: ",gyro_angle,file=stderr)
print("Gyro address: ",gyro_address,file=stderr)
print("Gyro units: ",gyro_units,file=stderr)
`

Starting: brickrun --directory="/home/robot/Main Scripts" "/home/robot/Main Scripts/sensor.py"
Started.
----------
Traceback (most recent call last):
  File "/home/robot/Main Scripts/sensor.py", line 20, in <module>
    us_port = LegoPort(address='in2:i2c50:mux1') # port 1 of MUX address i2c '50' connected to port 2 EV3 Brick
  File "/usr/lib/python3/dist-packages/ev3dev2/port.py", line 77, in __init__
    super(LegoPort, self).__init__(self.SYSTEM_CLASS_NAME, name_pattern, name_exact, **kwargs)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 210, in __init__
    chain_exception(DeviceNotFound("%s is not connected." % self), None)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 38, in chain_exception
    raise exception from cause
ev3dev2.DeviceNotFound: LegoPort(in2:i2c50:mux1) is not connected.
----------
Exited with error code 1.

dlech, 2019-10-24:
Follow this advice to find the right "address". https://github.com/ev3dev/ev3dev/issues/1335#issuecomment-545072439

naynt, 2019-10-25:
**You can use the Device Browser on the EV3 to find the "address" of each sensor and port** 
How exactly? Do you mean:
`a=legoPort(INPUT_1)
print (a.address, file=stderr)`
output for ev3 Sensor MUX = ev3-ports:in1:i2c81:mux2'

**The INPUT_1 constants only work if a sensor is connected directly to the port, otherwise, you need to use a string that contains the full address.**
Yes, I partially understand, could you please give me an example of a full address?

This is what I assume I think it may mean:
The address of the EV3 sensor MUX  connected to input 1 of the Ev3 Brick would be:
`LegoPort(address = 'ev3-ports:in1:i2c81:mux2')`
The address of the Ultrasonic sensor connected to C1 port of the EV3 sensor MUX would be:
`LegoPort(address = 'ev3-ports:in1:US-DIST-CM)`

Either way, please explain like I am five :-)

dlech, 2019-10-25:
Address of the port is `ev3-ports:in1:i2c81`, address of the sensor is `ev3-ports:in1:i2c81:mux2`

This is the Device Browser. It tells you the address so you don't have to guess.

[image]
[image]
[image]
[image]

naynt, 2019-10-25:
Thank you for clarifying. I got it confused with the  EV3DEV device browser shown in Visual studio Code extension. As always, thank you for your patience!

closed 2019-10-25 by naynt