ev3dev/ev3dev #1340: Error message when trying to use set_device to change default driver:'lego-ev3-color' to 'lego-ev3-us')

naynt, 2019-10-26:
`#!/usr/bin/env python3 

from ev3dev2.sensor.lego import UltrasonicSensor
from ev3dev2.port import LegoPort
from sys import stderr
from time import sleep

port = LegoPort(address ='ev3-ports:in1:i2c80:mux') # address of Ultrasonic sensor (port1) attached to EV3SensorMux
port.set_device('lego-ev3-us')
port.mode('US-DIST-CM')

sleep(0.5)

print(port.address,file=stderr)`

Traceback (most recent call last):
  File "/home/robot/Main Scripts/sensor_test.py", line 11, in <module>
    port.set_device('lego-ev3-us')
  File "/usr/lib/python3/dist-packages/ev3dev2/port.py", line 136, in set_device
    raise Exception("set_device is a write-only property!")
Exception: set_device is a write-only property!
----------
Exited with error code 1.

dlech, 2019-10-26:
https://github.com/ev3dev/ev3dev/issues/1161#issuecomment-417742377

naynt, 2019-10-26:
So I replicated what the person did in this thread, except using the Ultrasonic sensor but I still get the same error.

`#!/usr/bin/env python3 

from ev3dev2.sensor.lego import UltrasonicSensor
from ev3dev2.port import LegoPort
from sys import stderr
from time import sleep

mux1 = LegoPort(address ='ev3-ports:in1:i2c80:mux1') 
mux1.mode = 'uart'
mux1.set_device('lego-ev3-us')
sleep(1)
us = UltrasonicSensor('ev3-ports:in1:i2c82:mux1')
print(us.value(0))`

Traceback (most recent call last):
  File "/home/robot/Main Scripts/sensor_test.py", line 11, in <module>
    mux1.set_device('lego-ev3-us')
  File "/usr/lib/python3/dist-packages/ev3dev2/port.py", line 136, in set_device
    raise Exception("set_device is a write-only property!")
Exception: set_device is a write-only property!
----------
Exited with error code 1.

dlech, 2019-10-26:
That is because that person made the same mistake you did. Please re-read the comment that I linked (and only the comment that I linked).

Also, please see this page on how to format your posts correctly so that I don't have to keep editing them: https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks

naynt, 2019-10-27:
I see, mux1.set_device = 'lego-ev3-us' instead of mux1.set_device('lego-ev3-us'). Excellent, thanks for all your help. I am good from here on. Do you accept donations?

closed 2019-10-27 by naynt

dlech, 2019-10-27:
:smile: Thanks for the offer, but we don't have a need for donations at this time.