ev3dev/ev3dev #1341: How to access faster to data (read and write) ?

cho934, 2019-10-28:
Hello,
I'm using 2 Ev3 motors and 2 encoders with 2 NXT angle sensors in C++.

The read/write accesses are very long (15ms), even after optimising the C++ (10ms). 
Is there a way to better optimize the writings and the readings less than 1ms each ? is it linked to the kernel with some delays ?

I'm thinking also to create 4 threads for each sensors to parallelize all the requests. What do you think ?

dlech, 2019-10-28:
Can you share your code? 10-15ms sounds long. It should be more like 1ms.

The CPU only has one core, so multiple threads will only make things slower.

cho934, 2019-10-28:
- About writing for motors:

```
void AsservDriver::setMotorLeftPower(int value)
{
    if (_motor_left_.connected()) {
            limit(value, MAXVALUE_duty_cycle_sp);
            _motor_left_.set_duty_cycle_sp_run_direct(value);
    }
}
```

And I already optimised c++ code to write faster and open ofstream only once.

```
LegoMotor::LegoMotor(address_type address) :
        large_motor(address)
{
    if (this->connected()) {
        //if both motors are connected, then initialize each motor.
        this->reset();
        this->set_polarity(motor::polarity_inversed);
        this->set_ramp_down_sp(0);
        this->set_ramp_up_sp(0);
        this->set_stop_action(motor::stop_action_brake);

    }
    if (_path.empty()) {
        printf("ERROR - LegoMotor::LegoMotor() path empty !!!!\n");
        return;
    }
    _os_duty_cycle_sp = new std::ofstream();
    _os_duty_cycle_sp->rdbuf()->pubsetbuf(NULL, 0);
    _os_duty_cycle_sp->open(_path + "duty_cycle_sp");
    _os_command = new std::ofstream();
    _os_command->rdbuf()->pubsetbuf(NULL, 0);
    _os_command->open(_path + "command");
}
void LegoMotor::set_duty_cycle_sp_run_direct(int value)
{
    //this->set_duty_cycle_sp(value).run_direct();
    this->set_duty_cycle_sp_optimised(value);
    this->run_direct_optimised();
}
void LegoMotor::set_duty_cycle_sp_optimised(int value)
{
    //_os_duty_cycle_sp->clear();
    *_os_duty_cycle_sp << value;
}
void LegoMotor::run_direct_optimised()
{
    //_os_command->clear();
    *_os_command << "run-direct";
}
```

- About reading with encoders, I'm using the fastest (and redifined the function get_attr_int into get_attr_int_optimized 

```
long LegoAngleSensor::getValueDegrees()
{
    if (this->connected()) {

        //methode 1 => slow
        //long ticks = this->value(0);
        //methode 2 => slow
        //long ticks = this->get_attr_int("value0");

        //methode3
        long ticks = this->get_attr_int_optimized();
        //logger().info() << address_ << " " << ticks << logs::end;
        return ticks;
    } else
        return -999999;
}
```

Opening the value0 in the constructor to do it once as well

```
constexpr char LegoAngleSensor::ht_angle[];

LegoAngleSensor::LegoAngleSensor(address_type address) :
        i2c_sensor(address, { ht_angle })
{
    address_ = address;
    if (this->connected()) {
        this->set_mode("ANGLE-ACC");

        //Mise à jour du polling pour mettre a jour en i2c toutes les 20ms
        this->set_poll_ms(20);
    }
    usleep(10000);
    _ifs_value = NULL;
    this->openIfstreamFile("value0");
}

void LegoAngleSensor::reset()
{
    if (this->connected())
        this->set_command("RESET");
}

long LegoAngleSensor::getValueDegrees()
{
    if (this->connected()) {

        //methode 1
        //long ticks = this->value(0);
        //methode 2
        //long ticks = this->get_attr_int("value0");

        //methode3
        long ticks = this->get_attr_int_optimized();
        //logger().info() << address_ << " " << ticks << logs::end;
        return ticks;
    } else
        return -999999;
}

void LegoAngleSensor::openIfstreamFile(const std::string &name) const
{
    if (_path.empty()) {
        logger().error() << "ERROR - LegoAngleSensor::openIfstreamFile() path empty !!!!\n" << logs::end;
        //printf("ERROR - LegoAngleSensor::openIfstreamFile() path empty !!!!\n");
        return;
    }

    _ifs_value = new std::ifstream();
    _ifs_value->open(_path + name);
    _ifs_value->clear();

}
int LegoAngleSensor::get_attr_int_optimized() const
{
    for (int attempt = 0; attempt < 2; ++attempt) {
        try {
            _ifs_value->seekg(0, std::ios::beg);
            int result = -999999;
            *_ifs_value >> result;

            if (result == -999999)
            {
                logger().error() << "ERROR - get_attr_int_optimized - BAD RESULT!! " << address_ << logs::end;
                _ifs_value->seekg(0, std::ios::beg);
                *_ifs_value >> result;

                if (result == -999999)
                            {
                    logger().error() << "!!!!!!!!!!!!!!!!!!!!!!!!ERROR - get_attr_int_optimized - BAD RESULT!! " << address_ << logs::end;
                    exit(0);
                            }
            }
            return result;
        } catch (...) {
            // This could mean the sysfs attribute was recreated and the
            // corresponding file handle got stale. Lets close the file and try
            // again (once):
            if (attempt != 0) {
                logger().error() << "ERROR - get_attr_int_optimized !!!! attempt=" << attempt << logs::end;
                throw;
            }
            _ifs_value->clear();
        }
    }
}
```

I don't have more ideas for optimizing in C++. These optimisations allow to loop every 15/20ms for the motion system and PID position, but I really need to be faster (under 5/10ms), and the bottleneck is not the cosinus/sinus calculation for odometry but the access to data (read then write)
I can log the time into a sample program, let me time to do it.

Are there some other delays behind ? in the kernel ? in the I2c request behind ? Could you check please ?

;)
Cho

dlech, 2019-10-28:
I don't really know C++ very well, but the code looks OK to me. Doing the same calls in Python is much faster. The kernel just returns the latest value, so there is no delay for any I/O to the sensor.

This should do basically the same thing... open the file and disable buffering of the file, then do the same 3 system calls, seek, read and write. It is less than 1ms for all 3 calls, so I don't think there is any kernel problem.

```python
#!/usr/bin/env python3

import os
import sys
import timeit

SENSOR_PATH = '/sys/class/lego-sensor/sensor0/'
MOTOR_PATH = '/sys/class/tacho-motor/motor0/'

sensor = open(SENSOR_PATH + 'value0', 'rb', buffering=0)
motor = open(MOTOR_PATH + 'duty_cycle_sp', 'wb', buffering=0)

def test():
    sensor.seek(0, os.SEEK_SET)
    _ = sensor.read().decode()
    motor.write(str(100).encode())

COUNT = 1000

t = timeit.timeit(stmt='test()', number=COUNT, globals=globals())

print(t / COUNT * 1000, 'ms per loop', file=sys.stderr)
```

```
Starting: brickrun --directory="/home/robot/issue1341" "/home/robot/issue1341/main.py"
Started.
----------
0.8855197071097791 ms per loop
----------
Completed successfully.
```

gantzm, 2019-10-29:
I'm working through some of these same issues in Java land, though I also have a bunch of C++ experience.  Multiple layers of abstractions in these higher level language is most certainly not our friend.  If you look at the docs for [ifstream](http://www.cplusplus.com/reference/fstream/ifstream/) you'll see several layers of abstractions, including a bunch of stuff for format control and other such stuff.  Then on top of that, ifstream also maintains an internal [filebuf](http://www.cplusplus.com/reference/fstream/filebuf/) object that adds yet more abstractions.

I suspect that if you try something very basic like [fopen](http://www.cplusplus.com/reference/cstdio/fopen/), [fseek](http://www.cplusplus.com/reference/cstdio/fseek/), and [fread](http://www.cplusplus.com/reference/cstdio/fread/) you'll see a significant performance increase.  ( Yeah, I know not very c++'ish, but worth the experimenting.)  Now, you'll have to do data conversion on your own from the raw bytes, but there are quick ways of doing that.

As a point of reference I'm getting some very good numbers from Java, but only after the JIT compiler has done it's magic.  I basically warm up the JIT compiler by calling critical methods repeatedly before starting the actual code.

bmegli, 2019-10-29:
@cho934 

Note - I think your problem is in the user space. Try fread, fseek, frite as suggested by @gantzm 
or even read, write, lseek to check if this is your problem.

> Are there some other delays behind ? 

> in the I2c request behind ?

As @dlech mentioned you are reading cached values from sysfs (last written by the kernel). 
I2C read is on kernel side. Same for motors. This should not be affecting your program.

> in the kernel?

Sysfs by design uses textual data. So you have:

binary value in kernel -> conversion to string -> user space read -> conversion to binary

This adds some delay (but we are speaking microseconds level, not milliseconds).

In the past, having same concerns as you, I tested [binary version](https://github.com/bmegli/lego-linux-drivers/compare/718bcf30aeb4f396f89db38afa870cda4945b7dd...bmegli:tacho-motor-binary-attributes) of motor attributes (no string convertions).

With read/lseek (or fread/fseek, I don't remeber), 2 motors and speed/position readout I was well below 1 ms loop time.

The gain wasn't worth the complexity.

## Side notes

### Gyroscope enhanced odometry

I was solving similar problem like you - reading positions of 2 motors + I2C readout from gyroscope.

I got better system performance (in terms of odometry precission) by:
- disabling polling on I2C sensor
- reading the sensor value from user space with direct attribute

Note that this is somewhat against what your are trying to do, I2C read will [take around 5 ms](https://github.com/ev3dev/ev3dev/issues/445#issuecomment-228300750) of your loop time. On the other hand you will be able to get the gyro readouts with more than 100 Hz (10 ms) and will be better synchronized with motor position readings.

The code that did that is [here](https://github.com/bmegli/ev3dev-mapping-modules/blob/master/ev3dead-reconning/main.cpp). Loop time was purposefully kept at around 100 Hz (10 ms) with poll_ms variable (put to sleep). The reason was the gyroscope I used worked at 100 Hz internally, so there was little reason to read more often. You may see the result [here](https://www.youtube.com/watch?v=9o_Fi8bHdvs) and [here](https://www.youtube.com/watch?v=vzND_ISdhEs).

### Other things that may be affecting your loop time

Is your system under high load?

### Electrical and mechanical time constant of the motor

How fast motor can react to your setpoint on the physical level is probably far worse than 10 ms. There may be little to gain by changing the setpoint that often. See for example [this](https://forums.basicmicro.com/viewtopic.php?f=3&t=724&p=2497&hilit=mechanical+response+time#p2497) post, from "Motors are slow to respond.".

bmegli, 2019-10-29:
Another thing:

``` C++
void LegoMotor::set_duty_cycle_sp_run_direct(int value)
{
    //this->set_duty_cycle_sp(value).run_direct();
    this->set_duty_cycle_sp_optimised(value);
    this->run_direct_optimised();
}
```

I don't think you should be using `run_direct` every time.
Use `run_direct` once and then manipulate only `duty_cycle_sp` later.

cho934, 2019-10-29:
> Another thing:
>
> ```c++
> void LegoMotor::set_duty_cycle_sp_run_direct(int value)
> {
>     //this->set_duty_cycle_sp(value).run_direct();
>     this->set_duty_cycle_sp_optimised(value); [...]

Good catch!

cho934, 2019-10-29:
> Note - I think your problem is in the user space. Try fread, fseek, frite as suggested by @gantzm
> or even read, write, lseek to check if this is your problem.
>
>
>
> As @dlech mentioned you are reading [...]

Ok I will test that method too.

> 
> ## Side notes
> ### Gyroscope enhanced odometry
> I was solving similar problem like you - reading positions of 2 motors + I2C readout from gyroscope.
> 
> I got better system performance (in terms of odometry precission) by:
> 
> * disabling polling on I2C sensor

Could you explain the goal of the polling and how it works? if no polling, is the data of the sensor going on to be refreshed ?

> * reading the sensor value from user space with direct attribute

"with direct attribute", what does it mean ?

> Note that this is somewhat against what your are trying to do, I2C read will [take around 5 ms](https://github.com/ev3dev/ev3dev/issues/445#issuecomment-228300750) of your loop time. On the other [...]

Intesresting, I will glance at it.

> ### Other things that may be affecting your loop time
> Is your system under high load?

Currently with the loop of 20ms, I'm sure not. in case of high load, i received error messages for each overload.

> ### Electrical and mechanical time constant of the motor
> How fast motor can react to your setpoint on the physical level is probably far worse than 10 ms. There may be little to gain by changing the [...]

Yes, I know that. This is the goal of the motion system which use a PID position calculation. Moreover using duty_cycle_sp, I suppose the internal speed PID is activated.

bmegli, 2019-10-29:
> Could you explain the goal of the polling and how it works? if no polling, is the data of the sensor going on to be refreshed ?

Polling in general is the process of cyclically reading some value (e.g. each 20 ms, at 50 Hz). 

In the context of ev3dev NXT/I2C sensor (like [NXT angle sensor](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#hitechnic-nxt-angle-sensor)) this means that kernel cyclically reads sensor value using I2C. This value is stored until next such I2C read. When you read value from the user space program you normally get this cached value. The polling frequency (how often to read) can be controlled with `poll_ms` attribute, however there are limits how low you can get, mostly determined by slow EV3 I2C baudrate and other factors.

If you disable polling the data will not be refreshed but you may read it yourself (more on that later).

> > * reading the sensor value from user space with direct attribute
> 
> "with direct attribute", what does it mean ?

The alternative to polling done by kernel (the default) for NXT/I2C sensor is to disable polling and "read on demand" from user space or force readout by setting mode.

One way to do it is to set poll_ms to 0 and set mode, then read value as usual. It may have some side effects on some sensors.
Another way is to set poll_ms to 0 and use binary read on direct attribute of I2C device (this will give raw sensor data). Here you wait for the read to finish, not the kernel.

I thought you were doing something similar to me (that angle sensor is gyroscope, but it turns out it's encoder).

> > ### Electrical and mechanical time constant of the motor
> > How fast motor can react to your setpoint on the physical level is probably far worse than 10 ms. There may be little to gain by changing the setpoint that often. See for example [this](https://forums.basicmicro.com/viewtopic.php?f=3&t=724&p=2497&hilit=mechanical+response+time#p2497) post, from "Motors are slow to respond.".
> 
> Yes, I know that. This is the goal of the motion system which use a PID position calculation. Moreover using duty_cycle_sp, there is no regulation control. I suppose the internal speed PID is activated if I use run_forever with speed_sp, right ?

Yes, I was confused, thinking that angle sensor is gyroscope, not encoder.

cho934, 2019-11-02:
I tested different way to read/write.

The POSIX functions seems to be the fastest :
- more than 10/15 ms with ev3dev-lang-cpp
- 1,3 ms for each read and write on a file 
- 600 us for both (write then read) when you add priority 99 on the main program
- less than 400 us (see below) if you open/close files only once

Here is the fastest function:

```
void test::ReadWriteTest::do_io_benchmark_posix_readw_optimised(std::string filename, int wdata, int times)
{
    std::vector<char> inBuffer(1024);
    int fd = testPosixIOOpenFile(filename.c_str()); //open file
    for (int i = 0; i < times; ++i) {
        auto time1 = system_clock::now();
        testPosixIOWInt(fd, wdata + i);
        auto time2 = system_clock::now();
        testPosixIOR(fd, inBuffer);
        auto time3 = system_clock::now();
        std::string s(inBuffer.begin(), inBuffer.end()); //fast for string
        int data = std::stoi(s, nullptr, 0);
        auto time4 = system_clock::now();
        std::cout << i << " POSIX Write: " << " Wtime: " << duration_cast<microseconds>(time2 - time1).count() << " us"
                << " Read: " << data << " Rtime: " << duration_cast<microseconds>(time3 - time2).count() << " us"
                << " Convert time: " << duration_cast<microseconds>(time4 - time3).count() << " us" << " Total time: "
                << duration_cast<microseconds>(time4 - time1).count() << " us" << std::endl;
    }
    ::close(fd);
}

void test::ReadWriteTest::testPosixIOR(int fd, std::vector<char> &inBuffer)
{
    size_t inFileSize = ::lseek(fd, 0, SEEK_END);
    ::lseek(fd, 0, SEEK_SET);
    for (size_t bytesLeft = inFileSize, chunk = inBuffer.size(); bytesLeft > 0; bytesLeft -= chunk) {
        if (bytesLeft < chunk) {
            chunk = bytesLeft;
        }
        ::read(fd, &inBuffer[0], chunk);
    }
}

int test::ReadWriteTest::testPosixIOOpenFile(const char *outFile)
{
    int out = ::open(outFile, O_CREAT | O_RDWR, 0666);
    if (out < 0) {
        std::cout << "Can't open output file: " << outFile << std::endl;
        return -1;
    }
    return out;
}

void test::ReadWriteTest::testPosixIOWInt(int fd, int x)
{
    char tmp[9] = { 0x0 };
    sprintf(tmp, "%d", x);
    ::lseek(fd, 0, SEEK_SET);
    ::write(fd, tmp, strlen(tmp));
}

void test::ReadWriteTest::testPosixIOWString(int fd, std::string data)
{
    ::write(fd, data.c_str(), sizeof(data));
}
```

And the results:

```
0 POSIX Write:  Wtime: 584 us Read: -500 Rtime: 137 us Convert time: 246 us Total time: 968 us
1 POSIX Write:  Wtime: 108 us Read: -499 Rtime: 172 us Convert time: 61 us Total time: 342 us
2 POSIX Write:  Wtime: 94 us Read: -498 Rtime: 95 us Convert time: 49 us Total time: 238 us
3 POSIX Write:  Wtime: 163 us Read: -497 Rtime: 231 us Convert time: 60 us Total time: 455 us
4 POSIX Write:  Wtime: 100 us Read: -496 Rtime: 95 us Convert time: 49 us Total time: 245 us
5 POSIX Write:  Wtime: 96 us Read: -495 Rtime: 93 us Convert time: 49 us Total time: 240 us
6 POSIX Write:  Wtime: 96 us Read: -494 Rtime: 94 us Convert time: 50 us Total time: 241 us
7 POSIX Write:  Wtime: 96 us Read: -493 Rtime: 169 us Convert time: 49 us Total time: 316 us
8 POSIX Write:  Wtime: 93 us Read: -492 Rtime: 164 us Convert time: 49 us Total time: 307 us
9 POSIX Write:  Wtime: 93 us Read: -491 Rtime: 91 us Convert time: 46 us Total time: 231 us
10 POSIX Write:  Wtime: 94 us Read: -490 Rtime: 96 us Convert time: 48 us Total time: 239 us
11 POSIX Write:  Wtime: 85 us Read: -489 Rtime: 92 us Convert time: 46 us Total time: 225 us
12 POSIX Write:  Wtime: 108 us Read: -488 Rtime: 100 us Convert time: 58 us Total time: 267 us
13 POSIX Write:  Wtime: 89 us Read: -487 Rtime: 95 us Convert time: 50 us Total time: 234 us
14 POSIX Write:  Wtime: 88 us Read: -486 Rtime: 90 us Convert time: 47 us Total time: 226 us
15 POSIX Write:  Wtime: 95 us Read: -485 Rtime: 91 us Convert time: 50 us Total time: 237 us
16 POSIX Write:  Wtime: 89 us Read: -484 Rtime: 93 us Convert time: 47 us Total time: 230 us
17 POSIX Write:  Wtime: 93 us Read: -483 Rtime: 97 us Convert time: 49 us Total time: 240 us
18 POSIX Write:  Wtime: 90 us Read: -482 Rtime: 94 us Convert time: 46 us Total time: 231 us
19 POSIX Write:  Wtime: 93 us Read: -481 Rtime: 94 us Convert time: 49 us Total time: 236 us
```

bmegli, 2019-11-03:
Ok!

> The POSIX functions seems to be the fastest

This is somehow expected. The `read` function is just interface to system `read` call. The other ones (`fread`, `ifstream`) end up either calling `read` function or `read` system call themselves. `read`is as fast as you can get. Same goes for `write` and seek functions.

> more than 10/15 ms with ev3dev-lang-cpp

This seems surprisingly long. What is the exact scenario? Same read/write you do with POSIX `read`/`write`, nothing else?

@ddemidov

If that is the case there may be room for improvement for the C++ binding (e.g. using `read` under the hood)

## Side notes for real time programming

Huh, if you are already paranoid... Unless it is interesting you don't bother.

```C++
 std::string s(inBuffer.begin(), inBuffer.end()); //fast for string
```

The `std::string` constructor ends up in heap (new/delete) memory allocation. Heap memory allocation may occasionally lead to pagefaults, which take some time (looking for free memory block).

You may also lock your memory with `mlockall` to make sure your memory is never swapped out (always kept in RAM).

There is nice write-up about that in [ROS2 Design: Introduction to Real-time Systems](https://design.ros2.org/articles/realtime_background.html)

```C++
int data = std::stoi(s, nullptr, 0);
```

The funny thing about string conversions is that time it takes depends  on the value.
For example motor position can be as large as 2,147,483,647 and it takes more time to parse then value close to 0.

## Benchmarks

This is a matter of personal preference, but I would rather measure time of multiple iterations (e.g 10 000) and then average the value or also compute the standard deviation.

When you are measuring time as low as 50 us (string conversions), getting the time may also take some time.

50 us is both very little and a lot. I have a system (on microcontroller) that reads 2 encoders + IMU at 500 Hz and Lidar data, sends through USB, which keeps loop time below 100 us. The whole idea is to have data synchronized with this 100 us precision.

If you are concerned with times at this range (microseconds) you are reaching the limits of non real-time operating systems.

cho934, 2019-11-07:
> Ok!
>
>
> This is somehow expected. The `read` function is just interface to system `read` call. The other ones (`fread`, `ifstream`) end up either calling `read` function or `read` system call [...]

Agree.

> > more than 10/15 ms with ev3dev-lang-cpp
> 
> This seems surprisingly long. What is the exact scenario? Same read/write you do with POSIX `read`/`write`, nothing else?
> 
> @ddemidov
> 
> If that is the case there may be room for improvement for the C++ binding (e.g. using `read` under the hood)

Yes, this the goal. I'm going to post the code on ddemidov C++ bindings and ask for how to add it.

> ## Side notes for real time programming
> Huh, if you are already paranoid... Unless it is interesting you don't bother.
> 
> ```c++
>  std::string s(inBuffer.begin(), inBuffer.end()); //fast for string
> ```
I kept using string and I succeeded to be under 400us... it's already a good improvement.

> The `std::string` constructor ends up in heap (new/delete) memory allocation. Heap memory allocation may occasionally lead to pagefaults, which take some time (looking for free memory block).
>
> You [...]

Why not.

> ```c++
> int data = std::stoi(s, nullptr, 0);
> ```
> 
> The funny thing about string conversions is that time it takes depends on the value.
> For example motor position can be as large as 2,147,483,647 and it takes more time to parse then value close to 0.

Interesting.

> ## Benchmarks
> This is a matter of personal preference, but I would rather measure time of multiple iterations (e.g 10 000) and then average the value or also compute the standard deviation.

Also. But my goal is also to have good result in one or 2 calls too.

> When you are measuring time as low as 50 us (string conversions), getting the time may also take some time.

Of course, I tried to use 2 technics, with gettimeofday and the other with system_clock (in chrono.h). (no difference, and same average results)

> 50 us is both very little and a lot. I have a system (on microcontroller) that reads 2 encoders + IMU at 500 Hz and Lidar data, sends through USB, which keeps loop time below 100 us. The whole idea is to have data synchronized with this 100 us precision.

Of course, I have better results with other linux cards ;)

> If you are concerned with times at this range (microseconds) you are reaching the limits of non real-time operating systems.

In summary, using thread priority 99 and posix functions, I divided by 40 the single request of read/write. So it's indeed a very good improvement for the ev3.

cho934, 2019-11-07:
Here are the extend class to use posix functions.

You can use the functions to open/read/close each time or use the splitted functions to optimize calls in your own. It increases of course a little bit more the performance.

**ev3dev_extend.h**

```
#ifndef DRIVER_EV3_EV3DEV_EXTEND_HPP_
#define DRIVER_EV3_EV3DEV_EXTEND_HPP_
#include <string>
class device_extend
{
public:
    void set_attr_int_posix(int fd, int value);
    int get_attr_int_posix(int fd);
    void set_attr_string_posix(int fd, std::string name);
    std::string get_attr_string_posix(int fd);
    int open_posix_in(std::string pathfile);
    int open_posix_out(std::string pathfile);
    void close_posix(int fd);

    int open_get_attr_int_close_posix(std::string name);
    std::string open_get_attr_string_close_posix(std::string pathfile);
};
#endif /* DRIVER_EV3_EV3DEV_EXTEND_HPP_ */
```

**ev3dev_extend.cpp**
```
#include "ev3dev_extend.hpp"
#include <fcntl.h>
#include <stddef.h>
#include <unistd.h>
#include <cstdio>
#include <cstring>
#include <iostream>
#include <vector>

int device_extend::open_get_attr_int_close_posix(std::string pathfile)
{
    int in = open_posix_in(pathfile);
    int data = get_attr_int_posix(in);
    close_posix(in);
    return data;
}

std::string device_extend::open_get_attr_string_close_posix(std::string pathfile)
{
    int in = open_posix_in(pathfile);
    try {
        std::string data = get_attr_string_posix(in);
        close_posix(in);
        return data;
    } catch (...) {
        close_posix(in);
        throw;
    }
}

void device_extend::set_attr_int_posix(int fd, int value)
{
    try {
        char tmp[10] = { 0x0 }; //maximum digits
        sprintf(tmp, "%d", value);
        ::lseek(fd, 0, SEEK_SET);
        ::write(fd, tmp, strlen(tmp));
        if (errno == ENODEV) {
            printf("[set_attr_int_posix] ENODEV !!!\n");
        }
    } catch (...) {
        throw;
    }
}

void device_extend::set_attr_string_posix(int fd, std::string name)
{
    try {
        const char *tmp = name.c_str();
        ::lseek(fd, 0, SEEK_SET);
        ::write(fd, tmp, strlen(tmp));
        if (errno == ENODEV) {
            printf("[set_attr_string_posix] ENODEV !!!\n");
        }
    } catch (...) {
        throw;
    }
}

std::string device_extend::get_attr_string_posix(int fd)
{
    try {
        std::vector<char> inBuffer(1024);
        size_t inFileSize = ::lseek(fd, 0, SEEK_END);
        ::lseek(fd, 0, SEEK_SET);

        for (size_t bytesLeft = inFileSize, chunk = inBuffer.size(); bytesLeft > 0; bytesLeft -= chunk) {
            if (bytesLeft < chunk) {
                chunk = bytesLeft;
            }
            ::read(fd, &inBuffer[0], chunk);
        }
        std::string s(inBuffer.begin(), inBuffer.end());
        return s;
    } catch (...) {
        throw;
    }
}

int device_extend::get_attr_int_posix(int fd)
{
    std::string s = get_attr_string_posix(fd);
    int data = std::stoi(s, nullptr, 0); //convert to int
    return data;
}

int device_extend::open_posix_in(std::string pathfile)
{
    const char *inFile = pathfile.c_str();
    int in = ::open(inFile, O_RDONLY, 0666);
    if (in < 0) {
        std::cout << "open_posix_in :: Can't open output file: " << inFile << std::endl;
        return -1;
    }
    return in;
}

int device_extend::open_posix_out(std::string pathfile)
{
    const char *outFile = pathfile.c_str();
    int out = ::open(outFile, O_WRONLY, 0666);
    if (out < 0) {
        std::cout << "open_posix_out :: Can't open output file: " << outFile << std::endl;
        return -1;
    }
    return out;
}

void device_extend::close_posix(int fd)
{
    ::close(fd);
}
```

cho934, 2019-11-07:
And to increase thread priority, I used that
http://www.yonch.com/tech/82-linux-thread-priority