ev3dev/ev3dev #1348: ev3 not connecting to linux ubuntu 19 via ssh

yulikdaniel, 2019-11-17:
**What are you trying to do?**

I am trying to connect to my ev3 with ev3dev-stretch installed on it from linux ubuntu 19.04
I connected the both computers to the same wifi network (I used a tplink usb modem for the ev3)

**What did you expect to happen**

I excpected to be able to write code from the ev3 terminal on my PC.

**What actually happened**

The program did not do anything for several minutes and then gave the output "ssh: connect to host ev3dev.local port 22: Connection timed out"

**Source Code**

I tried three different lines, but neither of them worked.

```
ssh robot@ev3dev
ssh robot@ev3dev.local
ssh robot@<IP adress>
```

**Screenshots**
[image]

**Desktop (please complete the following information):**
Linux ubuntu 19.04

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: EV3-stretch

BertLindeman, 2019-11-17:
Hi,

My first reaction would be to ping the addresses you tried to reach with  `ssh`

Do they succeed? E.g.
```
ssh -c 1 ev3dev.local
```

Gr, Bert

yulikdaniel, 2019-11-30:
Dear Bert,
Thank you for your help.
Sorry for the delay - it's just that I rarely have access to the ev3. As I am not quite advanced in ssh-connecting, I just pasted "ssh -c 1 ev3dev.local" and got the answer "Unknown cipher type '1' ". I tried to google that, but didn't find any easy-to-understand answer, so I write back.

Yulik

BertLindeman, 2019-11-30:
Arggggggg 
I MEANT to write PING instead of SSH:
```ping -c 1 ev3dev.local```

So you can verify the connectivity...

Sorry

yulikdaniel, 2019-12-08:
It does not connect. Gives out 
`1 packets transmitted, 0 received, 100% packet loss, time 0ms`
See the screenshot
[image]

dlech, 2019-12-08:
It looks like you have configured a static IP address on the EV3 that is on a different subnet from your computer. Make sure DHCP is enabled on the EV3 when connecting to Wi-Fi.

jessephillips, 2024-03-25:
Hey @yulikdaniel I'm having the same issue. Did you ever find a solution? I recently was trying to install RPyC on the ev3, and now having connection issues. I'm wondering if these are related. Any ideas?

yulikdaniel, 2024-03-25:
I'm sorry, I have no idea...
I think I was asking this for a student, but since then a lot of time has
passed and I barely remember what an ev3 is.
Good luck solving your problem!